evcc-io/evcc #8178: HUAWEI S-DONGLE A05 Fehler beim Auslesen des Modbus

twingo667, 2023-05-26:
### Describe the bug

Egal ob ich das Template huawei-dongle oder huawei-dongle-powersensor nehme, erhalte ich folgenden Fehler:
read failed: modbus: exception '2' (illegal data address, function '3')
Den Fehler habe ich auf PV, GRID und BATTERIE 

### Steps to reproduce

.

### Configuration details

```yaml
Huawei S-Dongle A05
```

### Log details

```text
read failed: modbus: exception '2' (illegal data address, function '3')
```

### What type of operating system are you running?

Linux

### Version

0.117.3

jheinitz, 2023-05-26:
Moin!

Ich habe auch den Huawei SDongleA-05 und bei mir läuft das mit dem Template `huawei-dongle-powersensor`. Ist denn Dein Modbus so eingestellt, dass Du auch von Deiner `evcc` IP drauf zugreifen kannst? Ich erinnere mich, dass man ModbusTCP freischalten muss und dann wählen kann zwischen `eingeschränkt` und `offen` oder so ähnlich.

Meine Konfig sieht ( oder sah vor Umstellung auf MQTT) so aus:
```
meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: wechselrichter.example.com
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: wechselrichter.example.com
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: wechselrichter.example.com
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
  capacity: 10
```

Vielleicht kannst Du noch mal Deine Konfiguration hier posten.

Viele Grüße

twingo667, 2023-05-26:
Vielen Dank! Ich habe am Dingle die Modbus Kommunikation auf uneingeschränkt stehen und kann auch über andere Tools auf den Modbus zugreifen. Deshalb dachte ich, dass es ein evcc Thema ist.

Meine Meter-Konfiguration ist identisch, habe sogar die IP- Adressen statt hostnames drin.

viele Grüße 
Marcus

jheinitz, 2023-05-26:
Versuchst Du denn gleichzeitig mit `evcc `**UND** den anderen Tools gleichzeitig auf den Modbus zuzugreifen? Das geht nämlich nicht. Dafür brauchst Du dann den `modbusproxy `von `evcc`. Dadurch erreichst Du, dass immer nur einer auf den Modbus zugreift.

twingo667, 2023-05-26:
Natürlich nicht. Das kann der Huawei nicht ab 🤷‍♂️

jheinitz, 2023-05-26:
Dann fällt mir auch nichts mehr ein. Sorry .....

RTTTC, 2023-05-26:
What is the sdongle firmware?
Inverter firmware?

twingo667, 2023-05-26:
Sdongle 125
Inverter 153

RTTTC, 2023-05-26:
This could potentially be the problem, Your inverter should be good but latest sdongle firmware is 134. Not sure since what version it works OK with evcc, maybe 126?
Here you can find 133 https://photomate.zendesk.com/hc/en-gb/articles/4486389816861-Smart-Dongles , or ask Huawei tech support for the latest version, or to update it remotely for you.

jheinitz, 2023-05-26:
I also have 125 and 153 and it works on my side.

RTTTC, 2023-05-27:
https://github.com/evcc-io/evcc/discussions/7178

twingo667, 2023-05-27:
the Modbus id is 100 for the Dongle and I used it

RTTTC, 2023-05-27:
Yes, thats precisely your problem :)
Use inverter modbus id.

twingo667, 2023-05-27:
Ok, only the Master inverter? What about the slave inverters?