evcc-io/evcc #8179: Keba-Modbus: Session not updated correctly in UI

Duell10111, 2023-05-26:
### Describe the bug

Somehow the charging session is not updated in the UI, whereas the debug commands shows that evcc knows the correctly charged kWh.
Therefore the reading of the value from the WB seems to work.

UI:

[image]

evcc charger command:

```
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 01 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 01 00 00 00 07 ff 03 04 00 04 ca a5
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 02 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 03 00 00 00 06 ff 03 03 fc 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 03 00 00 00 07 ff 03 04 00 53 2e 3d
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 04 00 00 00 06 ff 03 04 0c 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 04 00 00 00 07 ff 03 04 02 3c ec 23
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 05 00 00 00 06 ff 03 03 f0 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 05 00 00 00 07 ff 03 04 00 00 1f 65
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 06 00 00 00 06 ff 03 03 f2 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 1f 8b
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 07 00 00 00 06 ff 03 03 f4 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 07 00 00 00 07 ff 03 04 00 00 1f 75
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 08 00 00 00 06 ff 03 03 ec 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 08 00 00 00 07 ff 03 04 00 00 00 07
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 09 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 09 00 00 00 07 ff 03 04 00 00 00 03
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 0a 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 0a 00 00 00 07 ff 03 04 00 00 00 03
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 0b 00 00 00 06 ff 03 05 de 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 0b 00 00 00 07 ff 03 04 00 00 72 c1
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 0c 00 00 00 06 ff 03 05 dc 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 0c 00 00 00 07 ff 03 04 00 00 00 00
Power:          5451W
Energy:         3754.7kWh
Current L1..L3: 8.04A 8.07A 8.05A
Charge status:  C
Enabled:        true
Charged:        2.9kWh
Identifier:     <none>
Diagnostic dump:
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 0d 00 00 00 06 ff 03 03 f6 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 0d 00 00 00 07 ff 03 04 01 66 2e 57
        Serial: 23473751
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 0e 00 00 00 06 ff 03 03 fa 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 0e 00 00 00 07 ff 03 04 03 0a 2a 00
        Firmware:       3.10.42
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 0f 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 0f 00 00 00 07 ff 03 04 00 04 ca a5
        Product:        314021
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 10 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 10 00 00 00 07 ff 03 04 00 00 00 00
        Phases source:  0
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 11 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 11 00 00 00 07 ff 03 04 00 00 00 00
        Phases state:   0
[keba  ] TRACE 2023/05/26 13:26:46 modbus: send 00 12 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2023/05/26 13:26:46 modbus: recv 00 12 00 00 00 07 ff 03 04 00 00 00 00
        Failsafe timeout:       0s
```

WB UI:

[image]

### Steps to reproduce

[... 395 lines cut ...]
[lp-1  ] DEBUG 2023/05/26 12:54:08 charge power: 4195W
[site  ] DEBUG 2023/05/26 12:54:08 pv power: 9131W
[site  ] DEBUG 2023/05/26 12:54:08 battery soc: 100%
[site  ] DEBUG 2023/05/26 12:54:08 battery power: 0W
[site  ] DEBUG 2023/05/26 12:54:08 grid power: -3620W
[site  ] DEBUG 2023/05/26 12:54:08 site power: -3570W
[lp-1  ] DEBUG 2023/05/26 12:54:08 charge currents: [6.1 6.13 6.1]A
[lp-1  ] DEBUG 2023/05/26 12:54:08 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/26 12:54:08 charge meter total import: 3751.826kWh
[lp-1  ] DEBUG 2023/05/26 12:54:08 charger status: C
[lp-1  ] DEBUG 2023/05/26 12:54:08 vehicle soc: 65%
[lp-1  ] DEBUG 2023/05/26 12:54:08 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/05/26 12:54:08 vehicle range: 318km
[lp-1  ] DEBUG 2023/05/26 12:54:08 pv charge current: 13.3A = 8.1A + 5.17A (-3570W @ 3p)
[lp-1  ] DEBUG 2023/05/26 12:54:08 max charge current: 13.3A
[site  ] DEBUG 2023/05/26 12:54:18 ----
```

### What type of operating system are you running?

Linux

### Version

evcc 0.117.4 (29b68abc)

andig, 2023-05-27:
> Somehow the charging session is not updated in the UI

This needs a modbus trace log of evcc running and `evcc charger` both at the time the error occurs. Its hard to imagine that there is a difference over modbus.

closed 2023-05-27 by andig

mucki12, 2023-06-06:
Die Änderung aus #8093 müssten ja in der nightly implementiert sein. Ich habe ich auch über die angeblich geladenen kWh gewundert und daher evcc gerade mit download der aktuellen nightly komplett neu aufgesetzt (Datenbank gelöscht, evcc Image gelöscht, neu geladen und mit vorhandener evcc.yaml neu gestartet).

Obwohl zu keiner Zeit ein Fahrzeug angeschlossen war und mit der neuen Installation nie ein Ladevorgang durchgeführt wurde, zeigt er mir komischerweise eine Ladung mit 10,4 kWh an:
[evcc-evccnightly.csv](https://github.com/evcc-io/evcc/files/11665733/evcc-evccnightly.csv)

[image]

mucki12, 2023-06-06:
Ja genau. Am 5.6. - habe mich nur gefragt warum die nightly Version mit Modbus Template bei Neuinstallation am 6.6. diese noch anzeigt? Test mit Neuinstallation ebenfalls heute und der Release Version und udp zeigt diese nicht an. 

Aber nach deiner Rückmeldung kann ich davon ausgehen, dass dort auch alte Ladevorgänge angezeigt werden und diese beim nächsten Ladevorgang zurückgesetzt werden, richtig?

andig, 2023-06-06:
Offensichtlich- ansonsten geht die Frage an den Keba Support. Wir zeigen an was über Modbus kommt.

Duell10111, 2023-06-09:
Was ich auch komisch finde ist, das bei den Ladevorgängen die Geladene Energie falsch dargestellt wird, aber die Zählerstände stimmen. 😅

[image]

andig, 2023-06-09:
> Was ich auch komisch finde ist, das bei den Ladevorgängen die Geladene Energie falsch dargestellt wird, aber die Zählerstände stimmen. 😅

Kannst du das konkret machen? Faktor falsch? 10 zu groß? Leider sind auch die Ladestände nur deshalb richtig weil wir es anders machen als es in der Doku steht- da ist also Experiment angesagt.

Duell10111, 2023-06-09:
> Kannst du das konkret machen? Faktor falsch? 10 zu groß? Leider sind auch die Ladestände nur deshalb richtig weil wir es anders machen als es in der Doku steht- da ist also Experiment angesagt.

Irgendwie nicht Faktor falsch, sondern manchmal übernimmt er einfach den Wert vom vorherigen Ladevorgang. 🤨

[image]

Hier ein weiteres Beispiel:

[image]

Ich bin mir auch nicht ganz sicher woran das genau liegen kann.
Die Modbus Register sahen eigentlich richtig aus. 🤔

Hier wird es korrekt zurückgegeben nur steht es in der UI nicht richtig. :/

[image]

mucki12, 2023-06-09:
> Offensichtlich- ansonsten geht die Frage an den Keba Support. Wir zeigen an was über Modbus kommt.

Du hattest natürlich (wie immer) recht - die Anzeige wurde mit einem neuen Ladevorgang korrekt zurückgesetzt und zeigt mir jetzt zumindest mit den letzten zwei Ladungen alles korrekt an. Vielen Dank für die letzten Updates.

[image]
[image]
[image]

Duell10111, 2023-06-12:
> > > Was ich auch komisch finde ist, das bei den Ladevorgängen die Geladene Energie falsch dargestellt wird, aber die Zählerstände stimmen. 😅
> > 
> > 
> > Kannst du das konkret machen? Faktor falsch? 10 zu groß? Leider sind auch die Ladestände nur deshalb richtig weil wir es anders machen als es in der Doku steht- da ist also Experiment angesagt.
> 
> Irgendwie nicht Faktor falsch, sondern manchmal übernimmt er einfach den Wert vom vorherigen Ladevorgang. 🤨
> 
> [image]
> 
> Hier ein weiteres Beispiel:
> 
> [image]
> 
> Ich bin mir auch nicht ganz sicher woran das genau liegen kann. Die Modbus Register sahen eigentlich richtig aus. 🤔
> 
> Hier wird es korrekt zurückgegeben nur steht es in der UI nicht richtig. :/
> 
> [image]
> 
> Was mir noch aufgefallen ist, das wenn ich das Auto abstecke und dann EVCC neu starte, fängt er wieder korrekt von 0 kWh an. 😅

Ich habe vermutlich das Problem gefunden und zwar beinhaltet das ChargedEnergy Register den Wert des vorherigen Ladevorgangs solange der nächste Ladevorgang noch nicht gestartet wurde.

Vlt. kann das dass Problem bei EVCC verursachen. 🤔

Duell10111, 2023-06-22:
Ich habe mal dazu einen Workarround gebaut und der funktioniert bei mir auch gut mit meiner Keba :).
Was haltet ihr davon?

https://github.com/evcc-io/evcc/pull/8577

premultiply, 2023-06-22:
Zu kompliziert.
Einfach `chargedEnergy()` entfernen.
Das kommt dann automatisch richtig via `totalEnergy()`.