evcc-io/evcc #8196: "eebus not found" hassio addon mit elli connect

tstr137, 2023-05-27:
### Describe the bug

Ist sicherlich eher eine spezielle Konfiguration, aber ich komme beim Einrichten der Wallbox nicht weiter. 

EVCC soll im Homeassistant laufen, das bekomme ich auch soweit gestartet, aber dann erscheint Fehlermeldung:

cannot create charger 'wallbox3'
cannot create charger 'template'
cannot create charger 'eebus'
i/o timeout

### Steps to reproduce

1. at startup
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
  
network:
  
  schema: http
  
  host: 192.168.100.xxx:xxxx # .local suffix announces the hostname on MDNS
  
  port: 
  

  
log: debug
  
levels:
  
  cache: error
  

  
# unique installation id
  
plant: *****  

  
interval: 10s # control cycle interval
  

  
meters:
  
- type: template
  
  template: fronius-solarapi-v1 
  
  usage: grid  
  
  host: 192.168.0.xxx 
  
  name: grid1
  
- type: template
  
  template: fronius-solarapi-v1 
  
  usage: pv  
  
  host: 192.168.0.xxx 
  
  name: pv2
  

  
chargers:
  
- type: template
  
  template: elliconnect 
  
  ski: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx  
  
  ip: 192.168.0.xxx  
  
  name: wallbox3
  

  
loadpoints:
  
- title: Stellplatz
  
  charger: wallbox3
  
  mode: pv
  
  phases: 3
  
  mincurrent: 5
  
  maxcurrent: 16
  
  resetOnDisconnect: true
  

  
site:
  
  title: Zuhause
  
  meters:
  
    grid: grid1
  
    pv:
  
    - pv2
  

  
eebus:
  
  certificate:
  
    public: |
  
      -----BEGIN CERTIFICATE-----
  
     xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
  
      -----END CERTIFICATE-----
  

     private: |
  
      -----BEGIN EC PRIVATE KEY-----
  
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
  
      -----END EC PRIVATE KEY-----
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/05/27 16:08:03 evcc 0.117.4
[main  ] INFO 2023/05/27 16:08:03 using config file: /config/evcc.yaml
[main  ] INFO 2023/05/27 16:08:03 starting ui and api at :7070
[db    ] INFO 2023/05/27 16:08:03 using sqlite database: /data/evcc.db
[eebus ] INFO 2023/05/27 16:08:03 Local SKI:  yyyyyyyyyyyyyyyyyyyyyyyyyyyyyyy
[eebus ] DEBUG 2023/05/27 16:08:03 mdns: announce
[eebus ] DEBUG 2023/05/27 16:08:03 mdns: using zeroconf
[eebus ] DEBUG 2023/05/27 16:08:03 starting websocket server on :4712
[eebus ] DEBUG 2023/05/27 16:08:03 mdns: start search
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.117.4

StefanSchoof, 2023-05-27:
Aus dem Bug Formular:

> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.

tstr137, 2023-05-27:
Stimmt. Da kommt aber das gleiche raus. 

[main  ] INFO 2023/05/27 16:20:24 evcc 0.117.4
[main  ] INFO 2023/05/27 16:20:24 using config file: C:\xxxxxxxxxxxxx\evcc_0.117.4_windows-amd64\evcc.yaml
[main  ] INFO 2023/05/27 16:20:24 starting ui and api at :7070
[db    ] INFO 2023/05/27 16:20:24 using sqlite database: C:\Users\xxxxxxxxx\.evcc\evcc.db
[eebus ] INFO 2023/05/27 16:20:24 Local SKI:  c8ff5d18213481b9fb06a92d10ad01a68911b4dc
[eebus ] DEBUG 2023/05/27 16:20:26 mdns: announce
[eebus ] DEBUG 2023/05/27 16:20:26 mdns: using zeroconf
[eebus ] DEBUG 2023/05/27 16:20:26 starting websocket server on :4712
[eebus ] DEBUG 2023/05/27 16:20:26 mdns: start search
[main  ] FATAL 2023/05/27 16:21:56 cannot create charger 'wallbox3': cannot create charger 'template': cannot create charger 'eebus': i/o timeout
[main  ] FATAL 2023/05/27 16:21:56 will attempt restart in: 5m0s

StefanSchoof, 2023-05-27:
Hast du auf der Wallbox das Pairing vorgenommen.

(Ich denke das eignet sich gerade ehr als Diskussion, da gerade noch nicht genug logs und ähnliches da sind und vermutlich bei dir im Setup noch etwas nicht ganz stimmt)

tstr137, 2023-05-27:
Danke nochmal für den Hinweis, es lag am Netzwerk

StefanSchoof, 2023-05-27:
Wenn du das gelöst hast, schließe bitte dieses Issue.

closed 2023-05-27 by tstr137