evcc-io/evcc #8205: pvPower in InfluxDB gibt nur die Power einer Erzeugungsanlage aus

StevieC121176, 2023-05-28:
### Describe the bug

Ich habe 2 Erzeugungsanlagen in EVCC konfiguriert. 
Der Parameter pvpower in influxdb gibt aber nur den Wert der ersten PV Anlage aus. 
Richtigerweise müssten hier beide zusammengerechnet werden oder es müsste noch einen weiteren Datenpunkt für die 2. Anlage geben.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
pi@raspberrypi:~ $ evcc meter
[main  ] INFO 2023/05/28 12:25:38 evcc 0.117.4 (29b68abc)
[main  ] INFO 2023/05/28 12:25:38 using config file: /home/pi/evcc.yaml
[db    ] INFO 2023/05/28 12:25:39 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/05/28 12:25:39 connecting evcc-1198243471 at tcp://192.168.178.80:1883
pv1
---
Power: 3817W

pv2
---
Power: 7287W

battery2
--------
Power:    73W
Soc:      100%
Capacity: 12.8kWh

grid3
-----
Power:          -9630W
Energy:         306.4kWh
Current L1..L3: 12.4A 13.8A 14.7A
Power L1..L3:   -2.87e+03W -3.3e+03W -3.47e+03W
```

### What type of operating system are you running?

Linux

### Version

0.117.4 (29b68abc)

StevieC121176, 2023-05-28:
[image]
[image]

andig, 2023-05-28:
Schaue ich mir an. Könntest du mal bitte den ganzen pv Zweig löschen und neu starten? Auch die NaN Werte gehören dort eigentlich nicht hin.

andig, 2023-05-28:
Update: schau bitten erstmal, ob Du die Tags berücksichtigt hast ;I

closed 2023-05-28 by andig

StevieC121176, 2023-05-28:
> Schaue ich mir an. Könntest du mal bitte den ganzen pv Zweig löschen und neu starten? Auch die NaN Werte gehören dort eigentlich nicht hin. Ist das Verhalten bei `battery` identisch oder klappt es da besser?

ich habe nochmal alle zweige gelöscht und neu anlegen lassen. (in iobroker und mqtt explorer. Selbes Ergebnis)

[image]
[image]

StevieC121176, 2023-05-28:
> 

ich steh grad auf dem Schlauch. Was meinst du mit Tags?

andig, 2023-05-28:
Influxdb Tags! Ich denke es werden slle Werte gespeichert, eben mittels Tags.

StevieC121176, 2023-05-28:
Ich habe influx db wie hier [https://github.com/evcc-io/evcc/discussions/4213#discussioncomment-4243763](url) beschrieben installiert und den User evcc angelegt. Ich habe seitdem keine updates oder ähnliches in influxdb gemacht. mir ist aber aufgefallen das dort neue Datenpunkte von evcc dazu gekommen sind wie `auxpower` oder `tariffco2`

Die 2. Erzeugeranlage ist bei mir später dazu gekommen. Nächste Woche kommt mein neuer RaspberryPi, da möchte ich EVCC und Solectrus zusammen aufspielen, mal schauen ob dann beide Anlagen in der InfluxDB zum Anzeigen bekomme, ansonsten wüsste ich nicht wie ich das bewerkstelligen sollte.