evcc-io/evcc #8206: Ladevorgang stoppt nicht bei erreichen des targetSoc-Werts

thumm, 2023-05-28:
### Describe the bug

Ich verwende evcc zur Steuerung einer Pool-Wärmepumpe. Der Ladestatus (soc) repräsentiert dabei die prozentuale Erreichung der in der Poolsteuerung (openhab) eingestellten Zieltemperatur.

In der vehicle-Konfiguration ist unter onIdentify ein Wert für targetSoc von 100 konfiguriert.

Der Ladevorgang stoppt bei Erreichen des Zielwertes allerdings nicht. Das Verhalten lässt sich im source code nachvollziehen:

`// targetSocReached checks if target is configured and reached.`
`// If vehicle is not configured this will always return false`
`func (lp *Loadpoint) targetSocReached() bool {`
`	return lp.vehicle != nil &&`
`		lp.Soc.target > 0 &&`
`		lp.Soc.target < 100 &&`
`		lp.vehicleSoc >= float64(lp.Soc.target)`
`}`

Müsste hier nicht

`
		lp.Soc.target <= 100 &&
`

verwendet werden (kleiner gleich)?

### Steps to reproduce

1. Ladevorgang starten
2. Per MQTT einen SOC-Wert von 100 senden
`mosquitto_pub -h 192.168.19.2 -t house/rooms/garden/waermepumpe/soc -m "100.0"`

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: f3482d4babaa413742e97e3dffcaf239481096250cc33008bfe7724c653cd338

interval: 10s # control cycle interval

mqtt:
  broker: 192.168.19.2
  topic: evcc

meters:
- type: template
  template: solarwatt
  usage: grid
  host: pvmanager.zuhause.xx
  name: grid1
- type: template
  template: solarwatt
  usage: pv
  host: pvmanager.zuhause.xx
  name: pv2
- type: template
  template: solarwatt
  usage: battery
  host: pvmanager.zuhause.xx
  name: battery3

chargers:
- type: template
  template: shelly
  host: 192.168.19.137
  channel: 0
  standbypower: 1
  integrateddevice: true
  icon: waterheater
  name: shelly1pm68C63AFB3A4B

vehicles:
  - name: waermepumpe
    type: custom
    title: Wärmepumpe
    icon: waterheater
    phases: 1
    soc:
      source: mqtt
      topic: house/rooms/garden/waermepumpe/soc
    onIdentify:
      mode: pv
      minSoc: 20
      targetSoc: 99

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  prioritySoc: 90
  bufferSoc: 95
  bufferStartSoc: 96
  residualPower: 200

loadpoints:
- title: Pool
  charger: shelly1pm68C63AFB3A4B
  vehicle: waermepumpe
  mode: off
  phases: 1
  mincurrent: 5 # ca. 750W Waermepumpe + 240W Pumpe = 990W / 2340V = 4,3
  maxcurrent: 6
  enable:
    threshold: 0
    delay: 0m
  disable:
    threshold: 0
    delay: 0m
  guardduration: 0m
  resetOnDisconnect: false
```

### Log details

```text
May 28 20:34:26 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:26 set charge mode: now
May 28 20:34:26 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:26 ----
May 28 20:34:26 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:26 charge power: 0W
May 28 20:34:26 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:26 pv power: 151W
May 28 20:34:27 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:27 battery soc: 94%
May 28 20:34:27 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:27 battery power: 300W
May 28 20:34:28 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:28 grid power: 9W
May 28 20:34:28 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:28 site power: 509W
May 28 20:34:28 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:28 charge meter total import: 11.258kWh
May 28 20:34:28 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:28 charger status: B
May 28 20:34:28 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:28 max charge current: 6A
May 28 20:34:28 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:28 charger enable
May 28 20:34:28 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:28 wake-up timer: start
May 28 20:34:29 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:29 ----
May 28 20:34:30 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:30 charge power: 0W
May 28 20:34:30 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:30 pv power: 151W
May 28 20:34:30 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:30 battery soc: 94%
May 28 20:34:30 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:30 battery power: 300W
May 28 20:34:31 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:31 grid power: 9W
May 28 20:34:31 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:31 site power: 509W
May 28 20:34:31 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:31 charge meter total import: 11.258kWh
May 28 20:34:31 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:31 charger status: B
May 28 20:34:39 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:39 ----
May 28 20:34:40 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:40 charge power: 4W
May 28 20:34:40 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:40 pv power: 150W
May 28 20:34:40 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:40 battery soc: 94%
May 28 20:34:40 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:40 battery power: 320W
May 28 20:34:41 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:41 grid power: -1W
May 28 20:34:41 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:41 site power: 519W
May 28 20:34:41 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:41 charge meter total import: 11.258kWh
May 28 20:34:41 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:41 charger status: C
May 28 20:34:41 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:41 charge start energy: 11.3kWh
May 28 20:34:41 server evcc[4143]: [lp-1  ] INFO 2023/05/28 20:34:41 start charging ->
May 28 20:34:41 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:41 wake-up timer: stop
May 28 20:34:41 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:41 vehicle soc: 95%
May 28 20:34:49 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:49 ----
May 28 20:34:50 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:50 charge power: 5W
May 28 20:34:50 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:50 pv power: 148W
May 28 20:34:50 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:50 battery soc: 95%
May 28 20:34:50 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:50 battery power: 558W
May 28 20:34:51 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:51 grid power: -1W
May 28 20:34:51 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:51 site power: 757W
May 28 20:34:51 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:51 charge meter total import: 11.258kWh
May 28 20:34:51 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:51 charger status: C
May 28 20:34:51 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:51 vehicle soc: 100%
May 28 20:34:59 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:34:59 ----
May 28 20:34:59 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:34:59 charge power: 5W
May 28 20:35:00 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:35:00 pv power: 148W
May 28 20:35:00 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:35:00 battery soc: 95%
May 28 20:35:00 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:35:00 battery power: 558W
May 28 20:35:01 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:35:01 grid power: -4W
May 28 20:35:03 server evcc[4143]: [site  ] DEBUG 2023/05/28 20:35:03 site power: 754W
May 28 20:35:03 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:35:03 charge meter total import: 11.258kWh
May 28 20:35:03 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:35:03 charger status: C
May 28 20:35:03 server evcc[4143]: [lp-1  ] DEBUG 2023/05/28 20:35:03 vehicle soc: 100%
```

### What type of operating system are you running?

Docker container

### Version

0.117.4

premultiply, 2023-05-28:
Bei 100% wird bewusst nicht abgeschaltet.
Bei Batteriespeichern findet da regelmäßig das Balancing statt. Die Ladung wird vom Fahrzeug beendet.

closed 2023-05-28 by premultiply

thumm, 2023-05-29:
> Bei 100% wird bewusst nicht abgeschaltet. Bei Batteriespeichern findet da regelmäßig das Balancing statt. Die Ladung wird vom Fahrzeug beendet.

Danke für die Erklärung.

andig, 2023-05-29:
Du kannst dir in dem Fall helfen, indem du 99 als Limit einstellst. Später mit Wärmekomponenten werden wir das direkt über die Temperatur machen.

thumm, 2023-05-30:
> Du kannst dir in dem Fall helfen, indem du 99 als Limit einstellst. Später mit Wärmekomponenten werden wir das direkt über die Temperatur machen.

Danke für den Tipp. So habe ich es jetzt konfiguriert und es klappt sehr gut.