evcc-io/evcc #8208: Add Schneider Electric EVlink Pro AC

premultiply, 2023-05-29:
OK - neue details gefunden: hier ist das Modus Protokoll:
https://download.schneider-electric.com/files?p_enDocType=Other+technical+guide&p_File_Name=GEX1969300-03.pdf&p_Doc_Ref=GEX1969300

Seite 9: EVlink Pro AC Modbus Interface
EVlink Pro AC Modbus interface can be used to limit the charging power of the
charging station.
Das sieht mir gut aus:
Set Point 4004 RW [0 32] uint16 A Remote energy management setpoint in Amps (99 = No charge in progress)

Gibt es eine Chance mit der Dokumentation das in evcc zu integrieren?
Gruß
Thorsten

_Originally posted by @thorsten-gehrig in https://github.com/evcc-io/evcc/discussions/8147#discussioncomment-6024279_

thorsten-gehrig, 2023-07-25:
Aktualisierter Download-Link ( version nun 04): [https://download.schneider-electric.com/files?p_enDocType=Other+technical+guide&p_File_Name=GEX1969300-04.pdf&p_Doc_Ref=GEX1969300](https://download.schneider-electric.com/files?p_enDocType=Other+technical+guide&p_File_Name=GEX1969300-04.pdf&p_Doc_Ref=GEX1969300
)

thorsten-gehrig, 2023-08-10:
Hi guys
I've received my Wallbox and will install it the next days.
Is there any guidance how I may try & test a custom config for Schneider Electric EVlink Pro AC (via Modbus TCP)?

Here is what i found yet (and may make sense for EVCC):

Power Apparent TOTAL 3059 (or Phase 1/2/3 at 3053/3055/3057) - float32 in KW
Set Point 4004 (0-32 Amps) - uint16 in Ampere

Set command 4001: (3/7/8/9/40) - uint16
Commands list:
• 3 = STOP_CHARGE = command that stops an ongoing charging session.
• 7 = SUSPEND_CHARGE = command that suspends an ongoing charging session.
• 8 = RESTART_CHARGE = command that restarts a suspended charging session.
• 9 = REBOOT = command that restarts the charging station.
• 40 = SET_AVAILABLE = command that switches charging station to available status.
• 41 = SET_UNAVAILABLE = command that switches charging station to unavailable status.

Based on https://www.loxforum.com/forum/german/software-konfiguration-programm-und-visualisierung/276918-schneider-electric-evlink-wallbox-%C3%BCber-modbus-tcp it might be required to define 4005/4006 to the minimal power (in A) to enable dynamic power - but according to documentation this must be set via eSetup as Modus is RO.
Degraded Set Point Mono 4005 for single phase charging (uint16)
Degraded Set Point Mono 4006 for triphase charging (uint16)

I guess it´s Modbus TCP Port: 502
UINT16 is defined as:
A 16-bit unsigned value in 2-1 (or big ENDIAN) format. The high-order byte is first, the low-order byte is second.
Float32 might be "A 32-bit IEEE floating point value returned in two consecutive 16-bit registers. The high-order word is returned in the first register and the low order word in the second. The 4 bytes are returned in 4-3-2-1 format."

Is there anything else required?
I may start with an Modbus-tool to read register to confirm them, 
then set register 4001 and 4004 to see if control does work. 
Any tools recommended to test modbus from Win or Mac?

Any advice how to transfer this into an official EVCC configuration and integration?

Best
Thorsten

thorsten-gehrig, 2023-08-12:
@andig : kannst du irgendwelche Hinweise geben?

andig, 2023-08-13:
Was meinst du? Sieht vollständig aus und müsste implementiert werden.

andig, 2023-08-13:
@thorsten-gehrig  könntest Du Portweiterleitung für info@evcc.io einrichten?

thorsten-gehrig, 2023-08-13:
Kriegen wir hin. 
Die Wallbox wird allerdings erst nächsten Freitag im Betrieb genommen (aus Gründen... ist schon angeschlossen aber nicht eingeschaltet). 
Ich hab Evcc derzeit in docker - kann aber alles möglich machen (zb dedizierte VM)
Weitere Absprache gerne direkt per email (thorsten@gehrig.info) oder auch telefonisch. 

Gruß
Thorsten

closed 2023-08-19 by andig