evcc-io/evcc #8224: oberste UI-Zeile fixieren?

VolkerK62, 2023-05-30:
Könnte man die oberste Zeile im UI (mit dem Namen und dem Einstellungs-Button fixieren?
Das würde mMn kaum Platz kosten und hätte den Vorteil, dass Fehlerhinweise immer sichtbar sind, egal wo man gerade rumscrollt.

naltatis, 2023-05-30:
Mir ist die Problemstellung noch nicht ganz klar. Auf welche Geräten, in welchen Situationen tritt das auf? Fehlerhinweise sollten wir mMn. lieber probieren abzustellen als sie einfacher zugänglich zu machen :D Den Namen halte ich auch für eine eher unwichtige Information. Der ändert sich ja nie.

VolkerK62, 2023-05-30:
Es ist kein wirkliches Problem. War nur ein spontaner Gedanke. Mittlerweile kenne ich ja meine Stolpersteine.
Aufgefallen ist es mir mehrmals, wenn die Fahrzeugerkennung (Bluelink) mal wieder länger dauert. Wenn ich dann hochscrolle, sehe ich den Fehler-Hinweis, dass das Auto nicht erreichbar ist.
Oder mein Garagen-Shelly hat nur mäßigen WLAN Empfang. Wenn der dann nicht reagiert, sehe ich den Fehler-Hinweis auch erst, wenn ich nach oben scrolle.
Liegt evtl. auch daran, dass ich die Details immer eingeblendet habe.
Auf Smartphone ist es weniger das Problem, da die Loadpoints (ich hab 3) nebeneinander liegen. Am Rechner liegt der dritte halt in der zweiten Reihe.

naltatis, 2023-05-31:
Ok das verstehe ich. Die obere Zeile einfach zu fixieren wird ohne Änderungen am Design aber vermutlich nicht gut funktionieren. Ich würd der oberen Zeile jetzt ungern eine eigene Hintergrundfarbe oder einen unteren Rahmen/Schatten geben. Ohne das siehts sonst im runtergerollten Zustand komisch aus.

Ich nehm den Wunsch nach "Fehler nicht übersehen" aber mit auf meine private Todo Liste und schließe den Task hier. Spätestens wenn die Config UI da ist müssen wir mMn sowieso noch mal an das Navigationskonzept ran. Das haben wir beim Ladelog ja erstmal nur sehr rudimentär gelöst.

closed 2023-05-31 by naltatis

VolkerK62, 2023-05-31:
>Ich nehm den Wunsch nach "Fehler nicht übersehen" aber mit auf meine private Todo Liste

ok. Danke