evcc-io/evcc #8254: Twc: wakeup not triggered

andig, 2023-06-01:
Interessant. In Deinem Logfile fehlt jegliche Spur von `wake-up timer: start`. Die Ursache ist wohl hier:

```
[lp-2  ] INFO 2023/04/17 18:49:31 car connected
[lp-2  ] DEBUG 2023/04/17 18:49:31 pv timer elapse
[lp-2  ] DEBUG 2023/04/17 18:49:31 pv timer inactive
[lp-2  ] DEBUG 2023/04/17 18:49:31 charger: guard elapse
[lp-2  ] DEBUG 2023/04/17 18:49:31 guard timer inactive
[lp-2  ] ERROR 2023/04/17 18:49:32 vehicle soc: 408 Request Timeout
[lp-2  ] ERROR 2023/04/17 18:49:32 max charge current 14A: 408 Request Timeout
```

Da Deine Config fehlt kann ich nur raten (bitte immer die Config dazu...). Du hast den Tesla am TWC stecken? Der bekommt jetzt beim Strom setzen einen Fehler vom Auto (evcc denkt aber vom Charger, also dem TWC). Deshalb gehts dann aus der Routine raus und wakeup wird niemals gestartet...

@premultiply in dem Fall ne gute Frage, inwiefern der TWC Fehler vom Auto zurück geben sollte?

Wäre mal spannend, ob das nicht am anderen Charger funktionieren würde...

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/7497#discussioncomment-5640526_

premultiply, 2023-06-01:
Vielleicht den TWC einfach rauswerfen?
Das Ding macht ja eigentlich in jeder Hinsicht nur Ärger.

andig, 2023-06-01:
Es gäb verschiedene Optionen das zu fixen:
- im twc Fehler bei MaxCurrent für Wakeup verwenden -> betrifft nur twc
- im twc Fehler bei MaxCurrent- falls Auto schläft- als api.ErrNotAvailable oder api.ErrMustRetry zurück geben und den im Loadpoint handlen. zusätzlich im twc beim Enable auf schlafendes Auto mit api.ErrMustRetry reagieren. Der Loadpoint würde das dann zum Anlass für wakeup nehmen.
- Analog vor aber das Wakeup direkt irgendwo im twc machen

closed 2023-06-10 by andig