evcc-io/evcc #8288: Für Kostal Piko werden die DC-Werte statt der AC-Werte verwendet

FC4712, 2023-06-03:
### Describe the bug

Ich habe konfiguriert:
- Kostal Smart-Meter
- Kostal Piko 12
- SMA Sunny Boy 1.5
- EVBOX Elvi
Bei den erzeugten Werten wird für Kostal Piko 12 der DC-Wert der beiden Strings als erzeugte Leistung verwendet. Für SMA Sunny Boy 1.5 wird der AC-Wert korrekt verwendet.

Das führt insgesamt dazu, dass zwar der eingespeiste Wert durch den Smart-Meter immer noch korrekt ist, der vermeintliche Hausverbrauch aber zu hoch berechnet wird.

Die Abfrage Kostal Piko 12 erfolgt mit 
http://192.168.178.117/api/dxs.json?dxsEntries=33556736
statt
http://192.168.178.117/api/dxs.json?dxsEntries=67109120

Ich habe den Code dafür in evcc\evcc-master\templates\definition\meter\kostal-piko.yaml gefunden. Vermutlich muss dort geändert werden.

### Steps to reproduce

[image]

Siehe oberer erzeugter Wert mit 5,8 kW, dass müssten 5,6kW sein. Daher ist der Eigenverbrauch 0,8kW statt der realistischen 0,6kW.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: cfac1570328af9a7c662521add04952cc4d8ea357c83fbd2f54114e5214f0331

interval: 10s # control cycle interval

meters:
- type: template
  template: kostal-ksem 
  id: 1  
  host: 192.168.178.175  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: gridkostal
- type: template
  template: kostal-piko 
  usage: pv  
  host: 192.168.178.117  
  name: pvkostal
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.178.204  
  password: xxx  
  name: pvsma

chargers:
  type: template
  template: elvi 
  stationid: EVB-Pxxxx 
  idTag: 'xxx'
  connecttimeout: 5m
  timeout: 2m  
  name: wallbox_elvi

vehicles:
- type: template
  template: renault 
  title: Megane E-Tech  
  user: xxx 
  password: xxx  
  vin: xxx  
  capacity: 60  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv  
  targetSoc: 100  
  minCurrent: 16  
  name: ev_renault

loadpoints:
- title: Wall E
  charger: wallbox_elvi
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Home
  meters:
    grid: gridkostal
    pv:
    - pvkostal
    - pvsma
```

### Log details

```text
[http  ] TRACE 2023/06/03 18:24:34 GET http://192.168.178.117/api/dxs.json?dxsEntries=33556736
[http  ] TRACE 2023/06/03 18:24:35 {"dxsEntries":[{"dxsId":33556736,"value":6102.255371}],"session":{"sessionId":0,"roleId":0},"status":{"code":0}}
```

### What type of operating system are you running?

Windows

### Version

0.117.4

premultiply, 2023-06-15:
Geht nicht so einfach, da Piko (auch) Hybrid-WR sind bzw. sein können.

andig, 2023-06-23:
Wäre die Lösung hier den Piko in Normal und hybrid zu splitten? Möchte das jemand versuchen? Sonst würde ich zu machen.

FC4712, 2023-06-24:
Eine konfigurierbare Unterscheidung würde ich begrüßen. An die Implementierung traue ich mich nicht ran.

andig, 2023-06-24:
Das wäre ganz einfach: templates/definition/meter/kostal-piko.yaml kopieren und in -hybrid umbennen. Dann im Original auf die AC-Werte umstellen und hinterher einmal `make docs` ausführen :)

andig, 2023-06-24:
PR ist da, in dem Gerüst müssten jetzt die richtigen Änderungen gemacht werden. Kann der normale Pico denn eine Batterie haben? Falls nein müsste die raus und wäre ein breaking change, dann sollten wir das rückwärtskompatibel machen.

/cc @premultiply

closed 2023-06-25 by andig