evcc-io/evcc #8290: Easee Wallbox Fehler 429: Too many requests

kaimundo, 2023-06-03:
### Describe the bug

Auto war heute den ganzen Tag vollgeladen angesteckt. Keiner daheim. Eben gesehen das eine Fehlermeldung 429 in evcc auftaucht. Evcc neu gestartet, jedoch nun startet es gar nicht mehr. Kann mir nicht erklären woher die vielen anfragen an easee kommen. Auto war en ganzen Tag angesteckt und niemand war zuhause. Ist da Problem bekannt? Wie lange dauert es bis die easee server mich wieder Abfragen lassen?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: info  
levels:  
  cache: error  
  
interval: 10s # control cycle interval  
  
sponsortoken: *****  
  
meters:  
- type: template  
  template: solaredge-inverter   
  id: 3    
  host: 192.168.178.99    
  port: 1502    
  usage: grid    
  modbus: tcpip    
  timeout: 5s    
  name: grid1  
- type: template  
  template: solaredge-inverter   
  id: 3    
  host: 192.168.178.99    
  port: 1502    
  usage: pv    
  modbus: tcpip    
  timeout: 5s    
  name: pv2  
- type: template  
  template: senec-home   
  usage: battery    
  host: 192.168.178.112    
  name: battery3  
  
chargers:  
- type: template  
  template: easee   
  user: *****  
  password: *****  
  charger: EH8KWDB8    
  name: wallbox5  
- type: template  
  template: easee   
  user: *****  
  password: *****  
  charger: EH3FQPSH    
  name: wallbox6  
  
vehicles:  
- type: template  
  template: tesla   
  title: Roadrunner    
  accessToken: *****  
  refreshToken: *****  
  vin: *****  
  capacity: 82    
  name: ev4  
  
loadpoints:  
- title: Easee-Wallbox Schwarz  
  charger: wallbox5  
  vehicles:  
  - ev4  
  mode: pv  
  phases: 0  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: false  
  disable:  
   delay: 2m  
- title: Easee-Wallbox Weiss  
  charger: wallbox6  
  vehicles:  
  - ev4  
  mode: pv  
  phases: 0  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: false  
  disable:  
   delay: 2m  
  
site:  
  title: Zuhause  
  meters:  
    grid: grid1  
    pvs:  
    - pv2  
    batteries:  
    - battery3  
  residualPower: 100  
  # bufferSoC: 100  
  # prioritySoC: 100  
  
tariffs:  
  currency: EUR  
  grid:  
    type: fixed  
    price: 0.3482 # [currency]/kWh  
  
  feedin:  
    # rate for feeding excess (pv) energy to the grid  
    type: fixed  
    price: 0.0625 # [currency]/kWh
```

### Log details

```text
32 charger status: A
[site  ] DEBUG 2023/06/03 21:53:42 ----
[lp-1  ] DEBUG 2023/06/03 21:53:42 charge power: 0W
[lp-2  ] DEBUG 2023/06/03 21:53:42 charge power: 0W
[site  ] DEBUG 2023/06/03 21:53:42 pv power: 0W
[site  ] DEBUG 2023/06/03 21:53:42 battery soc: 70%
[site  ] DEBUG 2023/06/03 21:53:42 battery power: 1650W
[site  ] DEBUG 2023/06/03 21:53:42 grid power: -121W
[site  ] DEBUG 2023/06/03 21:53:42 site power: 1629W
[lp-1  ] DEBUG 2023/06/03 21:53:42 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/03 21:53:42 charge meter total import: 2374.800kWh
[lp-1  ] DEBUG 2023/06/03 21:53:42 charger status: A
[site  ] DEBUG 2023/06/03 21:53:52 ----
[lp-1  ] DEBUG 2023/06/03 21:53:52 charge power: 0W
[lp-2  ] DEBUG 2023/06/03 21:53:52 charge power: 0W
[site  ] DEBUG 2023/06/03 21:53:52 pv power: 0W
[site  ] DEBUG 2023/06/03 21:53:52 battery soc: 70%
[site  ] DEBUG 2023/06/03 21:53:52 battery power: 1598W
[site  ] DEBUG 2023/06/03 21:53:52 grid power: -80W
[site  ] DEBUG 2023/06/03 21:53:52 site power: 1618W
[lp-2  ] DEBUG 2023/06/03 21:53:52 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/06/03 21:53:52 charge meter total import: 11.221kWh
[lp-2  ] DEBUG 2023/06/03 21:53:52 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.117.4

andig, 2023-06-03:
Da gibts nichts zu tun. Wir machen schlicht nicht „too many“ Requests.

closed 2023-06-03 by andig

kaimundo, 2023-06-03:
Ok und woher kommt,dann ie Fehlermeldung? Bzw wie kann ich das verhindern? Woher soll es denn sonst kommen als von Evcc? Vor allem wenn keiner daheim ist? Kann mir da keiner helfen oder nen Tipp geben?

andig, 2023-06-03:
Dir wird nur Easee sagen können was sie da gesehen haben wollen oder du schaust im trace log selbst wieviele Requests es waren. Ich wage zu behaupten sehr wenige, Fehlinterpretation bei Easee.

kaimundo, 2023-06-03:
Ist bisher noch nie vorgekommen dass ich das hatte. Nimmt keine gute Wendung bei easee wenn man jetzt schon willkürlich ausgesperrt wird.

andig, 2023-06-04:
Wir haken auch nochmal beim Partnersupport nach...

BionicThomas, 2023-06-04:
Hi Kaimundo.
Ich habe das selbe Problem.
Per Web: "Bei der Authentifizierung ist ein Fehler aufgetreten!"

Homebridge 
[4.6.2023, 13:20:57] [Easee] Error getting observations Request failed with status code 429
[4.6.2023, 13:20:57] [Easee] "Too many requests"
[4.6.2023, 13:20:57] [Easee] Error authenticating. Request failed with status code 429
[4.6.2023, 13:20:57] [Easee] "Too many requests"
[4.6.2023, 13:20:57] [Easee] Failed to get devices. Retry attempt 1 of 3 in 60 seconds...

Die Easee App scheint nicht davon betroffen. 
Und der Easee Cloud status gibt auch keine Probleme an
https://easee.statuspage.io

Ich habe heute beim Easee Support ein Ticket eröffnet.