evcc-io/evcc #8298: Sync charger: don't disable if charging needs to start anyway

andig, 2023-06-04:
Hier anderes Setup (Easee), aber gleiches Problem. @DerAndereAndi hat es bereits auf den Punkt gebracht. Boxen die sofort beim Anstecken selbstständig den Ladevorgang starten (und die EVCC dann erstmal wieder abschaltet) werden diese Meldung immer erzeugen. Es ist weder EEBUS noch Elli-Box spezifisch.

_Originally posted by @GrimmiMeloni in https://github.com/evcc-io/evcc/discussions/8223#discussioncomment-6077345_

dc-redeclipse, 2023-06-25:
Die Elli-Box beginnt das Laden mit 4kW morgens um 6, obwohl nicht genug PV Strom da ist im Modus PV, wenn man den Wagen ID.3 schon am Vorabend angesteckt hat. Beste Lösung ist die Verbindung von dem ID über die API von VW kurz zu trennen und neu zu verbinden.
@andig Plant ihr einen so etwas einzubauen um das Problem zu lösen?

andig, 2023-06-25:
Welches Problem? Was hat das mit dem Thema hier- Ladestart beim Anstecken- zu tun?

dc-redeclipse, 2023-06-25:
Ja genau.
Der ID 3 wird eingesteckt am vor Abend, beginnt normal mit dem Laden, falls noch etwas Sonne da ist. Am nächsten Morgen beginnt schon um 6 Uhr bevor die mindest PV Leistung erreicht ist, das Laden mit (6A) 4.1kW, wenn gerade mal von der PV nur 1kW kommt. Aber evcc zeigt es nicht mehr an, dass der ID 3 noch immer verbunden ist und geladen wird. Lösung ist über die Volkswagen API den ID 3 einmal zu trennen und wieder zu verbinden, dann geht alles wieder.
Ich vermute da einen bug.

andig, 2023-06-25:
Mach bitte neues Issue mit Logfile. Das hat hier überhaupt nichts zu suchen und ohne Logfile bringt die Diskussion nicht.

GrimmiMeloni, 2023-07-16:
Hab hier mal ein wenig drüber nachgedacht. Berücksichtigen müssen wir.

1. LP.Mode != OFF
2. When LP.Mode == PV muß TargetCurrent > 0 sein
3. Ist Vehicle Identification in-flight? (insbesondere bei Erkennung über State e.g. Enyaq)

Robbe64, 2023-07-18:
Den Fall wie in #8909 beschrieben konnte ich so bisher nicht mehr reproduzieren. Allerdings hat heute wieder das Laden nicht gestartet nachdem das Fahrzeug beim Anstecken erst scheinbar angefangen hatte zu laden und dann gestoppt wurde. Anbei nun das Log dazu. [log.txt](https://github.com/evcc-io/evcc/files/12076650/log.txt)

andig, 2023-07-21:
Idee: was würde passieren, wenn wir den Check ans Ende von Update() verschieben würden?
Oder der Charger uns den Hinweis gäbe dass er ein Eigenleben hat und deshalb anders behandelt werden muss?

GrimmiMeloni, 2023-07-21:
> Idee: was würde passieren, wenn wir den Check ans Ende von Update() verschieben würden?

So auf den ersten Blick scheint mir das eine ziemlich einfache Lösung für dieses Problem! #nice
Ich kann das hier morgen mal testen.

GrimmiMeloni, 2023-07-22:
Heute morgen nochmal drüber nachgedacht. Es gibt hier zwei Anwendungsfälle.

1. Fall: Default Mode am Loadpoint ist != OFF
Hierbei reicht vermutlich tatsächlich das Verschieben von chargerSync an das Ende von Update().

2. Fall: Default Mode am Loadpoint == OFF, Mode am Vehicle jedoch != OFF
Hier wird es jetzt komplizierter. Das Verschieben an das Ende von Update() hilft zumindest nicht bei Autos die über Ihren State asynchron identifiziert werden. Was es bräuchte ist dass der Ladevorgang nicht sofort stoppt, sondern erst wenn das Auto auch nach einer gewissen Zeit (2-3 Minuten?) noch nicht erkannt wurde. 
Ich habe den Code dazu nur kurz überflogen und auf die schnelle nichts passendes gesehen. Gibt es bereits einen Timer/Timeout für die asynchrone Fahrzeugerkennung den wir nutzen könnten? Die UI animiert ja auch das Icon während dieser Phase, daher vermute ich es gibt die Info irgendwo im System.

Macht es Sinn hierfür ein separates issue "sync charger: don't disable charging while vehicle identification is in progress" zu öffnen ?

Annex:
Zum besseren Verständnis hier nochmal der Anwendungsfall den ich (und einige andere Anwender) nutzen.
Ich betreibe meine außen zugängliche Wallbox offen, d.h. ohne Authentifizierung.
Damit nicht jeder Laden kann, habe ich als Default Mode am Loadpoint "OFF" definiert.
Für meinen Enyaq habe ich als Default Mode PV gesetzt.
Bis das Fahrzeug erkannt ist, handelt es sich aus evcc Sicht um ein Gastfahrzeug und folglich stoppt evcc den Ladevorgang der von der Wallbox direkt nach dem Anstecken gestartet wird. Dies ist eigentlich auch so gewünscht, denn sonst könnte jeder Laden.

andig, 2023-07-22:
> Macht es Sinn hierfür ein separates issue "sync charger: don't disable charging while vehicle identification is in progress" zu öffnen ?

Nein. Die Fahrzeugerkennung hat nichts mit dem "enabled" Zustand der Wallbox zu tun.

GrimmiMeloni, 2023-07-22:
> Nein. Die Fahrzeugerkennung hat nichts mit dem "enabled" Zustand der Wallbox zu tun.

Naja, ich würde sagen dass die Fahrzeugerkennung über den Mode vom Vehicle durchaus den Zustand der Wallbox beeinflusst - wenn auch indirekt.

Aber ich wäre dafür hier erstmal nur das Szenario ohne Vehicle spezifischen Modus zu betrachten.

andig, 2023-07-22:
> Naja, ich würde sagen dass die Fahrzeugerkennung über den Mode vom Vehicle durchaus den Zustand der Wallbox beeinflusst - wenn auch indirekt.

Ich weiß ja nicht warum du das sagst, aber syncCharger schaut nur auf den Charger Status. Nichts anderes.

GrimmiMeloni, 2023-07-22:
> Ich weiß ja nicht warum du das sagst, aber syncCharger schaut nur auf den Charger Status. Nichts anderes.

Ja, das ist auch korrekt. Aber es geht um das Zusammenspiel von chargerSync() und dem darauf folgenden Code in Loadpoint.Update(). Ich versuch es nochmal anders zu erklären. Danke für die Geduld. 😉 

Der Ziel dieses Issues ist es zu verhindern das durch den Charger Sync ein `charger.Enable(false)` aufgerufen wird, wenn danach in Loadpoint.Update() ein Ladevorgang gestartet wird, der zu `charger.Enable(true)` führt.

Ob Loadpoint.Update() einen Ladevorgang startet, hängt vom eingestellten Modus (Off, PV, ...) ab.

Dieser Modus wiederum kann aber auch vom Vehicle abhängen, wenn ein Mode speziell am Vehicle konfiguriert ist.
Die Fahrzeug Erkennung kann also u.a. auch zwischen OFF und PV, PV+min, ... umschalten.
Die Kette der Abhängigkeiten ist aus meiner Sicht

`Loadpoint -> Modus -> Vehicle > Vehicle detection
`

Wenn ich den Satz oben umschreibe kann ich also auch sagen:

Ob Loadpoint.Update() einen Ladevorgang startet, hängt _vom erkannten Fahrzeug und dem daraus resultierenden_ eingestellten Modus (Off, PV, ...) ab.

In der Praxis sehe ich häufig, daß entweder in zwei aufeinander folgenden Intervallen oder sogar im gleichen Intervall der Ladevorgang zunächst gestoppt und danach gestartet wird.

Wenn es in einem Intervall passiert ist der vorgeschlagene Change wahrscheinlich ausreichend. Wenn die Fahrzeugerkennung länger dauert (oder das Timing nicht gut ist), haben wir weiterhin das ungewünschte "abschalten dann einschalten".

andig, 2023-07-22:
> Dieser Modus wiederum kann aber auch vom Vehicle abhängen, wenn ein Mode speziell am Vehicle konfiguriert ist.

Puh, ja aber das ist auch die absolute Ausnahmesituation denn charger out of sync kann ständig passieren, Fahrzeugerkennung aber nur nach Anschließen. Insofern würde ich das erstmal ausklammern.

loeffelpan, 2023-07-22:
Finde ich ehrlich gesagt nicht. Ich denke es gibt viele Box-Fahrzeug-Kombination, die auf die asynchrone Erkennung angewiesen ist. Und die dauert tatsächlich manchmal weiter länger als der Intervall.
Und damit hätte man wieder ständig das Verhalten, dass es erst stoppt (zurecht weil noch Gastfahrzeug) und später iwann startet (wenn das Fahrzeug erkannt ist).

Fänds auch gut, wenn "abgewartet" würde bis alle Bedingungen klar sind bevor EVCC final auswertet und etwas tut.

andig, 2023-07-22:
> Und damit hätte man wieder ständig das Verhalten, dass es erst stoppt (zurecht weil noch Gastfahrzeug) und später iwann startet (wenn das Fahrzeug erkannt ist).

Warum hätte man das ständig? Gibts dafür Belege?

Anders gesagt: gerne neues Thema wenn das hier erledigt ist. Es hilft nicht ein schwieriges Problem schwieriger zu machen.

loeffelpan, 2023-07-22:
Ich sehe regelmäßig bei meiner Konstellation (Skoda, asynchrone Erkennung und EEBus Elli Charger) dass die Erkennung lange dauert und es zunächst stoppt und wieder startet.

Aber evtl. wäre mit dem einfachen Vorschlag hier der Großteil gelöst.
Daher find ich es, das in zwei Schritten anzugehen.

andig, 2023-07-22:
Dann bitte :). Aus meiner Sicht weder sinnvoll noch machbar.

andig, 2023-07-22:
Nochmal eine Idee. Wir haben ja bisher folgende Optionen:

- in `syncCharger` überlegen was alles passieren könnte und eine komplexe Entscheidung treffen -> schwierig
- `syncCharger` nach hinten schieben und schauen ob das Problem noch besteht -> möglich

Eine weitere Variante die mir gut gefiele wäre:

- `syncCharger` nur dann *handeln* lassen wenn der Fehler schon länger besteht, entweder mit Timeout oder einfach einen Durchlauf später

Die letzte Variante wäre sehr einfach- sollen wir das mal probieren? Irgendwas mit Fahrzeugerkennung wäre dann analog als Timeout umsetzbar.

andig, 2023-07-22:
PR dafür ist sehr einfach- wäre Klasse wenn das jemand mit einem Problemcharger (Easee?) oder häufigen Out of Sync mal ausprobieren könnte.

/cc @premultiply @DerAndereAndi bzgl. EEBus

DerAndereAndi, 2023-07-22:
Bzgl. EEBUS: Wenn das Auto am Anfang eines 10s Intervalls angesteckt wird, ist es möglich (nicht sicher) dass alle Informationen innerhalb des Intervals zur Verfügung stehen und damit kein `Out of Sync` kommt und auch das Fahrzeug erkannt ist. Ansonsten eher unwahrscheinlich. Alles hängt also vom Intervall selbst ab und wann innerhalb des Intervalls angesteckt wird.

GrimmiMeloni, 2023-07-23:
> PR dafür ist sehr einfach- wäre Klasse wenn das jemand mit einem Problemcharger (Easee?) oder häufigen Out of Sync mal ausprobieren könnte.

Ich baue mir eine Version und teste das heute Mittag mit der Easee.

andig, 2023-07-23:
> Ich baue mir eine Version und teste das heute Mittag mit der Easee.

Das wäre Klasse.

Allerdings hat diese Version einen deutlichen Nachteil. So wie der PR jetzt ist geht evcc im Intervall weiter vom alten Status aus. Wenn z.B. intern "disabled", tatsächlich aber "enabled" dann würde im Modbus "OFF" (der ja eindeutig ist) auch nicht abgeschaltet.

Das bringt mich auf eine weitere Idee- oben ergänzt- die mir noch viel besser gefällt: wir beschränken `syncCharger` auf das Erzeugen einer Warnung, schalten nicht *aber* aktualisieren den internen Status. Danach greift einfach die normale Logik des Ladepunktes. Dann tatsächlich sollte das vorgezogene Handeln gar nicht notwendig sein wenn danach ohnehin der Status neu beurteilt wird.

Weiterer PR für Test Variante 4 folgt.

GrimmiMeloni, 2023-07-23:
Ich habe einen build mit Variante 4 getestet. Das sieht für default mode == vehicle mode gut aus.
Wenn default mode != vehicle mode ist, kommt das System in einen merkwürdigen Zustand.
Es wird dann disabled, und nicht wieder enabled, obwohl der mode geändert wurde.

Ich habe einen fix (als PR auf den Variante 4 PR) gebaut der zumindest hier mit Easee jetzt einen Ladestart ohne stop/start ermöglicht, und das auch wenn Default OFF ist, und die Fahrzeug Erkennung auf PV schaltet.

andig, 2023-07-23:
> Es wird dann disabled, und nicht wieder enabled, obwohl der mode geändert wurde.

Bevor ich da was ändere würde ich gerne verstehen, warum das der Fall ist.

GrimmiMeloni, 2023-07-23:
> > Es wird dann disabled, und nicht wieder enabled, obwohl der mode geändert wurde.
> 
> Bevor ich da was ändere würde ich gerne verstehen, warum das der Fall ist.

Ich habe das ganze durchgekaut - siehe unten.
Es kam zu einem Timeout beim disable() - der hat dann das ganze ins Ungleichgewicht gebracht.
Wenn der Timeout nicht zuschlägt, hätte es funktioniert. Das heißt aber das diese Variante ggf. nicht mehr ausreichend fehlertolerant ist. 

Den Loadpoint.mode erst nach der Vehicle detection zu erkennen hat in solchen Fällen nur den Effekt das der Mode nicht gewechselt wird, und es nicht zum charger disable kommen kann. Sollte also hier erstmal nicht betrachtet werden, da es kein Fix für das von mir beobachtete Problem (Laden startet nicht mehr) ist.

Meine (unbestätigte) Vermutung ist, daß die Implementierung von easee.(charger).Enabled() nicht sauber arbeitet und ein true zurück gibt, wo keins kommen sollte.

Ausgangssituation:
Wallbox: Easee
Default Mode Loadpoint: off
Default Mode Vehicle: PV
Kein Fahrzeug angesteckt, d.h. Loadpoint.enabled = false

Jetzt passiert folgendes.
1. Fahrzeug wird verbunden
2. Easee startet automatisch den Ladevorgang, ab jetzt ist bereits `lp.charger.Enabled() = true`
3. EVCC Interval startet, `Loadpoint.Update()` startet
5. `Loadpoint.mode = api.ModeOff` wird in lokaler Variable `mode` gespeichert
6. Loadpoint führt Vehicle detection aus
7. ab hier ist `Loadpoint.mode  = api.ModePV`, aber das wird nicht mehr berücksichtigt da die lokale Variable nicht aktualisiert wird
8. `syncCharger()` erkennt `lp.charger.Enabled() (true) != lp.enabled (false)`
9. `syncCharger()`setzt `lp.enabled` auf den Wert vom Charger, also `true`
10. Loadpoint erreicht switch block und entscheided auf Basis der lokalen Variable `mode` (off)
11. case `api.modeOff` ruft `setLimit(0, true)` auf
12. `LoadPoint.setLimit()` erkennt in L705 das `0A >= lp.MinCurrent` ist. Lokale Variable `enabled` ist `false`. Dies wird verglichen mit `lp.enabled` (immer noch `true`)
13. obiger Vergleich führt in L712 dazu, das `charger.Enable(false)` aufgerufen wird
14. Easee charger API Aufruf für `pause_charge` erfolgt
15. Timeout auf Easee API schlägt zu, dies führt u.a. dazu, daß `lp.enabled = true` bleibt

16. nächstes evcc Intervall startet
17. `lp.charger.Enabled()` liefert weiterhin `true` (DCC ist > 6A, und MODE 2) <--- 🐞 **Ich vermute hier den Bug**
18. `syncCharger()` erkennt `lp.chargerEnabled() == lp.enabled`
19. switch block erneut, diesmal mit `api.ModePV`
20. `setLimit()` L705 erkennt das `chargeCurrent > minCurrent` ist, und das `lp.enabled = true` ist
21. deshalb wird kein `lp.charger.enable(true)` aufgerufen
22. ab hier finden nur noch API calls auf dem Charger statt die den Current Wert verändern, aber es wird nicht mehr enabled.

GrimmiMeloni, 2023-07-24:
Habe heute nochmal verifiziert. Das Problem oben ist ein Folgefehler von https://github.com/evcc-io/evcc/issues/9100. Es wird durch den Change hier nicht verursacht - vielleicht aber erst sichtbar. In jedem Fall aber hier erstmal off topic.

@andig ich würde gerne #9085 unabhängig davon auch mit reinnehmen, da dies das ganze nochmal streamlined. Die lokale Mode Variable wird vorher sowieso nicht verwendet, daher dürfte es auch keinen Seiteneffekt haben wenn diese erst im letzten Moment (und nach der Vehicle ID) ermittelt wird.

Hat denn schon jemand die Änderung hier mit einer anderen Box außer der Easee getestet?

loeffelpan, 2023-07-24:
Ich hätte ein passendes Setup mit Elli-Wallbox. Bin aber erstens gerade an der Nordsee im Urlaub (bis Sonntag) und hab zweitens keine Ahnung, wie ich einen Build erstelle und laufen lassen kann. Gibts dafür ne Doku? Dann probiere ich das gern. Mit Kommandozeile, git usw. kann ich grundsätzlich ungehen.

andig, 2023-07-25:
@GrimmiMeloni das sind m.E. 2 bugs die sich überlagern:
- `Opmode`=02: PR liegt vor, Test bei Dir. Danach könntest Du v4 oben nochmal nachtesten
- zweiter Fehler: `Update()` arbeitet mit veralteter Variable: das macht es langsamer (+1 Intervall), sollte aber dann nicht zu Blockaden führen. Wäre Klasse wenn Du das im Test bestätigen könntest.

GrimmiMeloni, 2023-07-25:
@loeffelpan ist dein Setup dockerized? Dann könntest Du einen meiner builds nehmen. Die liegen im docker hub.

@andig Ja, genau. Danke für die Zusammenfassung, das entspricht auch meinem Verständnis.

loeffelpan, 2023-07-26:
> @loeffelpan ist dein Setup dockerized? Dann könntest Du einen meiner builds nehmen. Die liegen im docker hub.

Nein, ist es nicht. Aber wollte ich im Prinzip eh.
Könnte ich also über den weg machen. Welches Image muss ich probieren bzw. welche PR sind wo drin?

GrimmiMeloni, 2023-07-26:
> Welches Image muss ich probieren bzw. welche PR sind wo drin?

Das aktuellste Image ist `grimmimeloni/evcc:nightly-2` (build id `16e5385`). Dieses basiert auf master (von gestern Mittag) und enthält #9075 (inkl. #9085) und #9100

loeffelpan, 2023-07-26:
Dann probiere ich den gerne aus, wenn ich am Sonntag oder Montag aus dem Urlaub zurück bin.

Was ist denn mit #9069? Hab in der Diskussion ehrlich gesagt nicht verstanden, ob der Ansatz jetzt verworfen wurde.

GrimmiMeloni, 2023-07-26:
> Was ist denn mit #9069? Hab in der Diskussion ehrlich gesagt nicht verstanden, ob der Ansatz jetzt verworfen wurde.

Verworfen nicht. Aber die wahrscheinliche Lösung in #9075 ist deutlich robuster. Wie gut #9069 funktioniert hängt unter anderem vom eingestellten Intervall ab. Aus meiner Sicht ist #9075 die präferierte Lösung.

GrimmiMeloni, 2023-07-27:
Habe den oben erwähnten Build mit der Kombination https://github.com/evcc-io/evcc/pull/9075 (inkl. https://github.com/evcc-io/evcc/pull/9085) und https://github.com/evcc-io/evcc/issues/9100 getestet. 

Ich habe dabei noch einen Bug in der `Enable()` Implementierung der Easee gefunden. Wenn nach einem erfolgreichen `resume` der anschließende DCC reset auf 6A schief geht (was er gerne mal tut), kommt zwar der Error aus `Enable()` zurück, die Box lädt dann aber trotzdem mit 32A. Ich werde die Easee Implementierung dahin gehend nochmal etwas verbessern, das sehe ich aber erstmal hier als off-topic.

Leider fixte in der Situation das nächste Interval das ganze nicht, weil der Loadpoint in `setLimit()`

1. `chargeCurrent` 6A ausrechnet, was dem Wert des vorherigen Intervalls entspricht. Daher ruft der Loadpoint kein `charger.MaxCurrent()` auf.
2. der Charger korrekt als `Enabled()` erkannt wird, und daher auch kein erneuter Aufruf von `Enable()` erfolgt

Besonders der erste Punkt ist blöd, weil der Loadpoint genug Daten hat um zu erkennen, daß nicht mit 6A geladen wird.

```
[lp-1  ] DEBUG 2023/07/26 12:53:54 charge power: 7622W
[lp-1  ] DEBUG 2023/07/26 12:53:54 charge currents: [31.7 0.008 0.01]A
[lp-1  ] DEBUG 2023/07/26 12:53:54 detected active phases: 1p
[lp-1  ] DEBUG 2023/07/26 12:53:54 pv charge current: 0A = 6A + -20.7A (4762W @ 1p)
```

In diesem Fall ergibt Charge Power durch 6A ca. 1200V ⚡️⚡️⚡️ 😁 - bzw. ein Delta von knapp 6.2kW zwischen erwarteter und tatsächlicher Charge Power. 

@andig Könnten wir den Loadpoint in https://github.com/evcc-io/evcc/blob/7d6a3914c82d8d3fea8fa502923b1460571c97a4/core/loadpoint.go#L666 um einen sanity check erweitern der solche Situationen erkennt und der dann dazu führt das `charger.MaxCurrent()` aufgerufen wird, auch wenn es laut State vom Loadpoint nicht notwendig ist?

Oder eine andere Alternative: `setLimit` hat bereits ein `force` flag. Wie wäre es wenn wir dieses bei Erkennung von Charger out of sync dieses setzen, und https://github.com/evcc-io/evcc/blob/7d6a3914c82d8d3fea8fa502923b1460571c97a4/core/loadpoint.go#L666 um `required` als alternative Bedingung erweitern?

andig, 2023-07-28:
OT: @GrimmiMeloni lass uns den Easee-spezifischen Teil bitte mal in https://github.com/evcc-io/evcc/issues/9145 weiter diskutieren. Das ist hier eigentlich alles OT. Kernproblem dabei bleibt doch: wir versuchen aus einem asynchronen Easee API ein synchrones zu machen. Dabei gibts viele edge cases. Eine Empfehlung könnte sein, Easee z.B. nur mit `interval: 1m` zu betreiben. Es bleiben denn interne edge cases bei mehreren API-Calls in Sequenz.

Hier scheint erstmal Variante 4 aus https://github.com/evcc-io/evcc/issues/8298#issuecomment-1646604582 weiter keine bekannten Defects zu haben. Ich würde diese jetzt nochmal mit einer Stromkorrektur anreichern die der Easee das Leben einfacher machen sollte für weitere Tests.

loeffelpan, 2023-07-29:
> > Welches Image muss ich probieren bzw. welche PR sind wo drin?
> 
> Das aktuellste Image ist `grimmimeloni/evcc:nightly-2` (build id `16e5385`). Dieses basiert auf master (von gestern Mittag) und enthält #9075 (inkl. #9085) und #9100

Ich hab das jetzt mit dem Image und meiner Elli-Box mal ausprobiert. Default-Mode ist `off` (da öffentlich zugänglich), Vehicle-Mode ist `pv`. `inverval` war zunächst auf `10s` wie der Konfigurations-Assistent das vorgibt.

Hier mal mein Log mit vier Connects und meinen Beobachtungen dazu.
[_evcc_logs.txt](https://github.com/evcc-io/evcc/files/12207382/_evcc_logs.txt)

```
[lp-1  ] INFO 2023/07/29 12:07:30 car connected
[lp-1  ] INFO 2023/07/29 14:13:10 car connected
```
Beim ersten und zweiten Versuch hat die Erkennung ne knappe Minute gedauert. Hat natürlich erstmal gestoppt und dann wieder gestartet nachdem die Erkennung durch war.

Jetzt hab ich `interval` auf `30s` geändert.
```
[lp-1  ] INFO 2023/07/29 15:26:06 car connected
[lp-1  ] INFO 2023/07/29 15:38:06 car connected
```
Beim dritten und vierten Versuch war die Erkennung innerhalb einer Sekunde durch und evcc hat den Vorgang laufen lassen.

Wie ich vermutet hatte klappt das also gut, sofern die Fahrzeugerkennung bis zum Intervall fertig wird. Dann lädt er durch ohne erst zu stoppen. Falls nicht stoppt er zunächst (richtigerweise weil Gastfahrzeug) und startete dann wieder.
Die Implementierung für diese Diskussion wäre also aus meiner Sicht in Ordnung.

Ich möchte aber nochmal anregen mit jeglichem Handeln von evcc auf die Fahrzeugerkennung (oder deren Timeout) zu warten, da erst dann alle Informationen für eine korrekte Entscheidung vorliegen. Allerdings könnte ich auch verstehen, dass die Reaktionszeit von evcc dann etwas lang erscheint. Vielleicht sollte man das konfigurierbar machen, ob er auf die Erkennung wartet oder nicht?

andig, 2023-07-29:
Ab morgen im Nightly verfügbar.

> Ich möchte aber nochmal anregen mit jeglichem Handeln von evcc auf die Fahrzeugerkennung (oder deren Timeout) zu warten, da erst dann alle Informationen für eine korrekte Entscheidung vorliegen. 

Der Root cause ist ja, dass einige Charger ein Eigenleben haben. Das wird prinzipbedingt immer mit Nachteilen verbunden sein da es dann zwei "steuernde" Instanzen gibt.

closed 2023-07-29 by andig