evcc-io/evcc #8300: Support `minSoc` and `targetSoc` for vehicles using offline vehicle template

DerAndereAndi, 2023-06-04:
**Is your feature request related to a problem? Please describe.**
The vehicle template `offline` currently does not allow to set the `minSoc` and `targetSoc` values.

**Describe the solution you'd like**
If the wallbox to vehicle connection is handled via ISO15118-2 and the SoC is provided via that connection, setting an target SoC via the UI and also defaults for both via the template should be possible.

andig, 2023-06-04:
Macht Sinn, PR welcome!

DerAndereAndi, 2023-06-04:
An open question is: How to handle the case that the connection from the EVSE to EV falls back to PWM and SoC values are not available any longer.

andig, 2023-06-04:
Like any other api error- no more soc updates.

DerAndereAndi, 2023-06-04:
Sure. But doesn't this have consequences on the charging process in PV mode? And therefor for the UI also?

closed 2023-06-06 by andig