evcc-io/evcc #8311: Communication issues

Jo-sebove, 2023-06-04:
### Describe the bug

Hello everyone, 
I've been running evcc for more than a month now and I have recurrent failure with the communication during the charging process. Please note that I run evcc on macOS not on linux.

### Steps to reproduce

1. Plug the car in and let the software control the charging process.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: d0b6c81b427ed8cc0c2ce58c829fedf061849eb420f993a26512735986a59bed

interval: 10s # control cycle interval

sponsortoken: 
****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.1.85  
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.1.85  
  name: pv2
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.1.86  
  name: pv3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.1.11  
  name: wallbox5

vehicles:
- type: template
  template: bmw 
  title: i3  
  user: joseph.fochesato@hotmail.fr  
  password: wUfxig-zyvgam-qakci0  
  vin: Wby7z41040vd82368  
  capacity: 27  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv  
  minSoc: 100  
  targetSoc: 100  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev4

loadpoints:
- title: go-eCharger
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv3
```

### Log details

```text
[main  ] INFO 2023/06/04 23:46:32 evcc 0.117.4
[main  ] INFO 2023/06/04 23:46:32 using config file: /Users/josephfochesato/evcc.yaml
[main  ] INFO 2023/06/04 23:46:32 starting ui and api at :7070
[db    ] INFO 2023/06/04 23:46:32 using sqlite database: /Users/josephfochesato/.evcc/evcc.db
[main  ] ERROR 2023/06/04 23:46:33 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'bmw': unexpected status: 401 (Unauthorized)
[lp-1  ] WARN 2023/06/04 23:46:33 locking phase config to 3p for switchable charger
[site  ] INFO 2023/06/04 23:46:33 site config:
[site  ] INFO 2023/06/04 23:46:33   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/06/04 23:46:33     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/06/04 23:46:33     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/06/04 23:46:33     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/06/04 23:46:33   vehicles:
[site  ] INFO 2023/06/04 23:46:33     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/06/04 23:46:33 loadpoint 1:
[lp-1  ] INFO 2023/06/04 23:46:33   mode:        pv
[lp-1  ] INFO 2023/06/04 23:46:33   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/06/04 23:46:33   meters:      charge ✓
[lp-1  ] INFO 2023/06/04 23:46:33     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/06/04 23:46:33 phase timer inactive
[lp-1  ] DEBUG 2023/06/04 23:46:33 pv timer inactive
[lp-1  ] DEBUG 2023/06/04 23:46:33 guard timer inactive
[lp-1  ] INFO 2023/06/04 23:46:33 vehicle updated: unknown -> i3 (offline)
[lp-1  ] DEBUG 2023/06/04 23:46:34 max charge current: 6A
[site  ] DEBUG 2023/06/04 23:46:34 ----
[lp-1  ] DEBUG 2023/06/04 23:46:34 charge power: 0W
[site  ] DEBUG 2023/06/04 23:46:34 pv power: 0W
[site  ] DEBUG 2023/06/04 23:46:34 grid power: 3006W
[site  ] DEBUG 2023/06/04 23:46:34 site power: 3006W
[lp-1  ] DEBUG 2023/06/04 23:46:34 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/04 23:46:34 charge meter total import: 2091.162kWh
[lp-1  ] DEBUG 2023/06/04 23:46:34 charger status: B
[lp-1  ] INFO 2023/06/04 23:46:34 car connected
[lp-1  ] DEBUG 2023/06/04 23:46:34 pv timer elapse
[lp-1  ] DEBUG 2023/06/04 23:46:34 pv timer inactive
[lp-1  ] DEBUG 2023/06/04 23:46:34 charger: guard elapse
[lp-1  ] DEBUG 2023/06/04 23:46:34 guard timer inactive
[lp-1  ] DEBUG 2023/06/04 23:46:34 charge meter total import: 2091.162kWh
[lp-1  ] ERROR 2023/06/04 23:46:34 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': unexpected status: 401 (Unauthorized)
[lp-1  ] DEBUG 2023/06/04 23:46:34 charger disable
[lp-1  ] DEBUG 2023/06/04 23:46:34 wake-up timer: stop
[lp-1  ] DEBUG 2023/06/04 23:46:34 switched phases: 3p
[site  ] DEBUG 2023/06/04 23:46:44 ----
[lp-1  ] DEBUG 2023/06/04 23:46:44 charge power: 0W
[site  ] DEBUG 2023/06/04 23:46:44 pv power: 0W
[site  ] DEBUG 2023/06/04 23:46:44 grid power: 2960W
[site  ] DEBUG 2023/06/04 23:46:44 site power: 2960W
[lp-1  ] DEBUG 2023/06/04 23:46:44 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/04 23:46:44 charge meter total import: 2091.162kWh
[lp-1  ] DEBUG 2023/06/04 23:46:44 charger status: B
[lp-1  ] DEBUG 2023/06/04 23:46:44 pv charge current: 0A = 0A + -4.29A (2960W @ 3p)
[lp-1  ] DEBUG 2023/06/04 23:46:44 pv enable timer reset
[lp-1  ] DEBUG 2023/06/04 23:46:44 pv timer inactive
[site  ] DEBUG 2023/06/04 23:46:54 ----
[lp-1  ] DEBUG 2023/06/04 23:46:54 charge power: 0W
[site  ] DEBUG 2023/06/04 23:46:54 pv power: 0W
[site  ] DEBUG 2023/06/04 23:46:54 grid power: 3966W
[site  ] DEBUG 2023/06/04 23:46:54 site power: 3966W
[lp-1  ] DEBUG 2023/06/04 23:46:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/04 23:46:54 charge meter total import: 2091.162kWh
[lp-1  ] DEBUG 2023/06/04 23:46:54 charger status: B
[lp-1  ] DEBUG 2023/06/04 23:46:54 pv charge current: 0A = 0A + -5.75A (3966W @ 3p)
[site  ] DEBUG 2023/06/04 23:47:04 ----
[lp-1  ] DEBUG 2023/06/04 23:47:04 charge power: 0W
[site  ] DEBUG 2023/06/04 23:47:04 pv power: 0W
[site  ] DEBUG 2023/06/04 23:47:04 grid power: 3025W
[site  ] DEBUG 2023/06/04 23:47:04 site power: 3025W
[lp-1  ] DEBUG 2023/06/04 23:47:04 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/04 23:47:04 charge meter total import: 2091.162kWh
[lp-1  ] DEBUG 2023/06/04 23:47:04 charger status: B
[lp-1  ] DEBUG 2023/06/04 23:47:04 pv charge current: 0A = 0A + -4.38A (3025W @ 3p)
[site  ] DEBUG 2023/06/04 23:47:14 ----
[lp-1  ] DEBUG 2023/06/04 23:47:14 charge power: 0W
[site  ] DEBUG 2023/06/04 23:47:14 pv power: 0W
[site  ] DEBUG 2023/06/04 23:47:14 grid power: 3029W
[site  ] DEBUG 2023/06/04 23:47:14 site power: 3029W
[lp-1  ] DEBUG 2023/06/04 23:47:14 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/04 23:47:14 charge meter total import: 2091.162kWh
[lp-1  ] DEBUG 2023/06/04 23:47:14 charger status: B
[lp-1  ] DEBUG 2023/06/04 23:47:14 pv charge current: 0A = 0A + -4.39A (3029W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

[site  ] DEBUG 2023/06/04 23:48:04 ---- [lp-1  ] DEBUG 2023/06/04 23:48:04 charge power: 0W [site  ] DEBUG 2023/06/04 23:48:04 pv power: 0W [site  ] DEBUG 2023/06/04 23:48:04 grid power: 3009W [site  ] DEBUG 2023/06/04 23:48:04 site power: 3009W [lp-1  ] DEBUG 2023/06/04 23:48:04 charge currents: [0 0 0]A [lp-1  ] DEBUG 2023/06/04 23:48:04 charge meter total import: 2091.162kWh [lp-1  ] DEBUG 2023/06/04 23:48:04 charger status: B [lp-1  ] DEBUG 2023/06/04 23:48:04 pv charge current: 0A = 0A + -4.36A (3009W @ 3p)

Sillium007, 2023-06-05:
Please remove your password from your posted yaml file!

`[lp-1  ] ERROR 2023/06/04 23:46:34 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': unexpected status: 401 (Unauthorized)`

I don't know the process for BMW but it seems you need to check your credentials or re-authenticate.

andig, 2023-06-05:
Removed, but please set new password, it is burned. Looks like the credentials where wrong though, hence the 401.

Jo-sebove, 2023-06-05:
News password set, didn’t paid attention to the jaml file! Thank you for your awareness:-)