evcc-io/evcc #8337: MQTT Topics are not updated after reconnection to Wireless network

occtopus, 2023-06-06:
### Describe the bug

The mqtt topics for evcc are published completely. After some time it happens that not all topics are published anymore. Especially the evcc.status topic is shown as offline. Only after a service restart everything works as expected.

[image]

The screenshot demonstrates that some topics are update while others are not and at the same time the status is set to offline.

The data received via modbus tcp is received at processed correctly within evcc and displayed in the web gui. Only the mqtt topic publish seems to stop working.

### Steps to reproduce

Wait for a undefined time. It could be 1-2 weeks or some days. From the logs it seems that it happens when the raspberry pi rejoins the multicast group after a wireless disconnect.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval
site:
  - title: Zuhause # display name for UI
    meters:
      grid: my_grid # grid meter reference
      pv: # (pvs = veraltet)
        - my_pv # first pv meter reference
      battery: # (batteries = veraltet)
        - my_battery # battery meter reference
    residualPower: 100
    bufferSoc: 80
    prioritySoc: 66

mqtt:
  broker: 192.168.1.112:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: homeassistant
  password: ********

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    #Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.121.120 # Hostname
    port: 502 # Port    
  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv
    #Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.121.120 # Hostname
    port: 502 # Port
  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery
    #Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.121.120 # Hostname
    port: 502 # Port 
loadpoints:
    - title: Garage
      charger: wallbox1
chargers:
    - name: wallbox1
      type: custom
      status: # charger status A..F
        source: js
        script: |
          var res="A";
          res;
      enabled: 
        source: js
        script: |
          var res=1;
          res;
      enable: # set charger enabled state (true/false or 0/1)
        source: js
        script: |
          console.log(enable);
      maxcurrent:
        source: js
        script: |
          console.log(maxcurrent);
```

### Log details

```text
The log file is to huge for being provided completely. In the syslog I could observe the following:

evcc wpa_supplicant[566]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jun  5 04:52:15 evcc systemd-networkd[653]: wlan0: Gained carrier
Jun  5 04:52:20 evcc systemd-networkd[653]: wlan0: DHCPv4 address 192.168.1.119/24 via 192.168.1.1
Jun  5 04:52:20 evcc avahi-daemon[666]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.119.
Jun  5 04:52:20 evcc avahi-daemon[666]: New relevant interface wlan0.IPv4 for mDNS.
Jun  5 04:52:20 evcc avahi-daemon[666]: Registering new address record for 192.168.1.119 on wlan0.IPv4.

followed by the error 

Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/site/homePower: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/chargeCurrent: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/site/greenShare: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/sessionCo2PerKWh: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/sessionEnergy: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/chargeDuration: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/sessionPricePerKWh: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/chargedEnergy: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/charging: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/site/tariffEffectivePrice: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/connected: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/enabled: timeout
Jun 5 04:52:23 evcc evcc[737]: [mqtt ] ERROR 2023/06/05 04:52:23 send: evcc/site/warn: timeout

After this error it seems it ends up in the state where only some mqtt topics are published.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.117.4

andig, 2023-06-06:
Gibts vorher noch mqtt Einträge? Kann mosquitto von dem Rechner aus posten? Ist der mqtt Server überhaupt erreichbar? Systemlast auf dem mqtt server?

occtopus, 2023-06-07:
Ja es gibt vorher Einträge. Der mqtt server ist wohl temporär nicht erreichbar, da sich das WLAN neu verbindet. Allerdings geht es aus meiner Sicht darum, warum sich das System nicht vollständig von dem Network Disconnect erholt. Ist dieser Fall einmal aufgetreten, hilft nur ein Neustart von evcc. Ich habe auch schon mal 3 Tage gewartet um zu sehen ob der Zustand sich selbst löst.

andig, 2023-06-07:
> warum sich das System nicht vollständig von dem Network Disconnect erholt

Daher meine Frage:

> Gibts vorher noch mqtt Einträge

Gibt es einen Eintrag zum Disconnect? Falls nein sehen wir den gar nicht erst, dann gibts auch keinen Reconnect...

andig, 2023-06-07:
Wir drehen uns im Kreis. Es gibt einen Connection(Lost)Handler:

```go
// ConnectionLostHandler logs cause of connection loss as warning
func (m *Client) ConnectionLostHandler(client paho.Client, reason error) {
	m.log.ERROR.Printf("%s connection lost: %v", m.broker, reason.Error())
}

// ConnectionHandler restores listeners
func (m *Client) ConnectionHandler(client paho.Client) {
	m.log.DEBUG.Printf("%s connected", m.broker)

	m.mux.Lock()
	defer m.mux.Unlock()

	for topic := range m.listener {
		m.log.DEBUG.Printf("%s subscribe %s", m.broker, topic)
		go m.listen(topic)
	}
}
```

Wenn im Logfile kein disconnect zu sehen ist wird es auch keinen reconnect geben... Das mag ein Fehler in der Library sein aber erstmal müssten wir doch rausfinden, ob der disconnect überhaupt identifiziert wird.

eldrik, 2023-06-07:
also in meinem Log gibt es keine evcc disconnect Meldung noch wird eine stale Verbindung (ist sie ja scheinbar auch nicht) aufgrund einer Inaktivität vom Broker abgebaut.

Kann denn kein keepalive beim connect definiert werden und würde dies hier überhaupt helfen?

https://github.com/at-wat/mqtt-go/blob/master/keepalive.go

andig, 2023-06-07:
Das ist schon so:

```
//   Port: 1883
//   CleanSession: True
//   Order: True (note: it is recommended that this be set to FALSE unless order is important)
//   KeepAlive: 30 (seconds)
//   ConnectTimeout: 30 (seconds)
//   MaxReconnectInterval 10 (minutes)
//   AutoReconnect: True
```

occtopus, 2023-06-07:
Ich habe folgende Meldung im Log:
Jun  6 17:50:23 evcc evcc[737]: message repeated 9 times: [ [mqtt  ] ERROR 2023/06/06 17:50:23 send: evcc/site/error: timeout]
Jun  6 17:50:31 evcc evcc[737]: [mqtt  ] ERROR 2023/06/06 17:50:31 tcp://192.168.1.112:1883 connection lost: pingresp not received, disconnecting

andig, 2023-06-07:
Aber keine weitere „connected“ Meldung? Dann geht das entweder physisch nicht (=es gibt keine Verbindung) oder der reconnect funktioniert einfach nicht. Falls sich das reproduzieren ließe könntet ihr ein Ticket bei der verwendeten Library aufmachen. Evcc ist hier nur Consumer.

occtopus, 2023-06-08:
Leider keine weiteren  Connected Meldungen. Ist die Library dann mqtt-go?

P.S.: bzgl der Log Files sehr verständlich. Meins war nur 10 Mb groß ... Beim nächsten mal ist es aber dabei.

andig, 2023-06-08:
Wir verwenden paho- eigentlich der Gold-Standard siw. Ich mache mal zu bis es ein neues Logfile gibt. Ziel müsste sein, das Verhalten für Tests/Analyse zu reproduzieren.

closed 2023-06-08 by andig

occtopus, 2023-06-20:
@andig Ich konnte es endlich reproduzieren. Nach der Umstellung auf eine RJ45 Verbindung konnte ich es durch eine einfache Unterbrechung (13 min) der Netzwerkverbindung reproduzieren und ein deutlich umfangreicheres Log File erzeugen. Ich hoffe dies ist aussagekräftig genug.
Um Jun 20 19:55:16 habe ich das Kabel gezogen. Ab Jun 20 20:08:16 ist die Verbindung wieder hergestellt und es werden wieder korrekte Daten aus dem Wechselrichter via Modbus TCP ausgelesen.

[evcc-12min.log](https://github.com/evcc-io/evcc/files/11806238/evcc-12min.log)

andig, 2023-06-21:
Hast Du mal geschaut was im Log passiert wenn das Netzwerk wieder da ist?

andig, 2023-06-22:
Ich denke das Bild hier ist, dass wir bei Mqtt zwar einen `disconnect` sehen, aber keinen `reconnect`. Richtig?

Den Connect loggen wir nur auf debug:

```go
// ConnectionLostHandler logs cause of connection loss as warning
func (m *Client) ConnectionLostHandler(client paho.Client, reason error) {
	m.log.ERROR.Printf("%s connection lost: %v", m.broker, reason.Error())
}

// ConnectionHandler restores listeners
func (m *Client) ConnectionHandler(client paho.Client) {
	m.log.DEBUG.Printf("%s connected", m.broker)
	// ...
}
```

Kannst Du bitte bestätigen, dass das Mqtt Log auch Debug enthält? Falls nein bräuchten wir ein neues weil die These nicht überprüfbar ist. Wäre das Log vollständig könnte man es sehen ;)