evcc-io/evcc #8346: Myporsche not connecting

Jimnaldo, 2023-06-07:
### Describe the bug

Logfile:

vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'porsche': login failed: no auth code

### Steps to reproduce

connect porsche to charger
error: vehicles:
  -

### Configuration details

```yaml
vehicles:
  - name: Porsche Taycan
    type: template
    template: porsche
    user: info@addot.be # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: ******* # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    capacity: 83 # Akkukapazität in kWh # Optional
```

### Log details

```text
vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'porsche': login failed: no auth code
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.117.4

StefanSchoof, 2023-06-07:
Aus dem Bug Formular:

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.

andig, 2023-06-07:
In dem Fall war ich (via email) zufällig schon dran.

MarkusGH, 2023-06-14:
Kriegt man momentan leider auch mit der aktuellen Nightly:

```
Jun 14 15:30:16 homebridge evcc[559433]: [porsche] TRACE 2023/06/14 15:30:15 POST https://login.porsche.com/auth/api/v1/de/de_DE/public/login
Jun 14 15:30:16 homebridge evcc[559433]: [porsche] TRACE 2023/06/14 15:30:16 keeploggedin=false&password=***&resume=&sec=&state=eyJ6aXAiOiJERUYiLCJhbGciOiJkaXIiLCJlbmMiOiJBMTI4Q0JDLUhTMjU2Iiwia2lkIjoiOXMiLCJzdWZmaXgiOiJRd0hTelMuMTY4Njc1MDI1MyJ9..w7V8JVJkb6bs2-FwgVNMtA.4DI4cCDbvBKcjXXQ4-o8BQb2FZM32ksHS2WtbpVGqesnZ2qXNPYDsDnq3oplLHsN4HhNIuCwCgYFZK
83FTRjqm6zk2DomSOF-AJrywJAMAGaLcFkM5wcA3N8vdrHVegh.PA271Fwku_3fijF-Oqppyw&thirdPartyId=&username=***
Jun 14 15:30:16 homebridge evcc[559433]: [porsche] TRACE 2023/06/14 15:30:16 GET https://login.porsche.com/auth/de/de_DE/redirect
Jun 14 15:30:16 homebridge evcc[559433]: [porsche] TRACE 2023/06/14 15:30:16 GET https://login.porsche.com/as/authorization.oauth2?response_type=code&client_id=cloud_oidc&redirect_uri=https%3A%2F%2Flogin.porsche.com%2Fpa%2Foidc%2Fcb&state=eyJ6aXAiOiJERUYiLCJhbGciOiJkaXIiLCJlbmMiOiJBMTI4Q0JDLUhTMjU2Iiwia2lkIjoiOXMiLCJzdWZmaXgiOiJxTXZYNDYuMTY4Njc1
MDI1MyJ9..KDjETadVWsjOfdknJTSb2w.oUCQJ-6w8KHclT1GY_ySL2OEMFqJUkITAi9EVWybxUKRRIc0sjP3MHVZWdmbzhfT1kDv4SfO6tshNZUtj6GKG2ycpA7SqcURkSHaVXmWmvxv2kVUScJe-xr6Nz-HSRcF.29CZ75X9lsUHjEP5vZGVfQ&nonce=T6FvrMH2l9jntZrBQz8cd5rpTDEvaXBCS15QM5HekAA&acr_values=high&scope=openid%20profile%20address%20email%20phone&vnd_pi_requested_resource=https%3A%2F%2Flogin.p
orsche.com%2Fauth%2Fde%2Fde_DE%2Fredirect&vnd_pi_application_name=myaccess+-+auth+-+web
Jun 14 15:30:16 homebridge evcc[559433]: [porsche] TRACE 2023/06/14 15:30:16 <!DOCTYPE html>
Jun 14 15:30:16 homebridge evcc[559433]: <!-- template name: http.error.page.template.html -->
Jun 14 15:30:16 homebridge evcc[559433]: <html lang="en" dir="ltr">
Jun 14 15:30:16 homebridge evcc[559433]: <head>
Jun 14 15:30:16 homebridge evcc[559433]: #011<title>My Porsche</title>
Jun 14 15:30:16 homebridge evcc[559433]: #011<base href="https://login.porsche.com/"/>
Jun 14 15:30:16 homebridge evcc[559433]: #011<meta name="robots" content="noindex, nofollow"/>
Jun 14 15:30:16 homebridge evcc[559433]: #011<meta http-equiv="Content-Type" content="text/html; charset=UTF-8"/>
Jun 14 15:30:16 homebridge evcc[559433]: #011<meta name="viewport" content="initial-scale=1.0, minimum-scale=1.0, maximum-scale=1.0, user-scalable=no"/>
Jun 14 15:30:16 homebridge evcc[559433]: #011<!--<link href="../static/pa/assets/css/ping.css" rel="stylesheet" />-->
Jun 14 15:30:16 homebridge evcc[559433]: #011<link href="assets/css/ping.css" rel="stylesheet"/>
Jun 14 15:30:16 homebridge evcc[559433]: #011<link href="assets/css/porsche-connect-ui.css" rel="stylesheet"/>
Jun 14 15:30:16 homebridge evcc[559433]: </head>
Jun 14 15:30:16 homebridge evcc[559433]: <body>
Jun 14 15:30:16 homebridge evcc[559433]: <div class="pcc-site">
Jun 14 15:30:16 homebridge evcc[559433]: #011<div class="pcc-widget widget-navigation no-nav">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011<!-- w_header-porsche-crest.html -->
Jun 14 15:30:16 homebridge evcc[559433]: #011#011<div class="crest-container">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011<div class="row nav-row">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011<div class="col-xs-12">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011<div class="container-pcc-widget">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011<h3 class="crest-line">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011#011<span class="crest-wrapper">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011#011#011<a class="porsche-crest" href="https://www.porsche.com/countries/myporsche/"
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011#011#011   title="My Porsche">My Porsche</a>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011#011</span>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011</h3>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011<section class="pcc-content">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011<div class="row">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011<!-- title.template.html -->
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011<div class="pcc-widget pcc-widget-col widget-pagetitle">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011<div class="pcc-widget-container pagetitle-container">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011<div class="row txt-center">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011<div class="col-xs-12">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011#011<h1 class="title">My Porsche - An error has occurred</h1>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011<div class="pcc-widget pcc-widget-col widget-pageintro title-separator">
[... cut ...]
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011#011<img src="assets/images/banner-error.jpg" class="img-responsive">
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011#011#011<!-- nba.template.html (removed) -->
Jun 14 15:30:16 homebridge evcc[559433]: #011#011</div>
Jun 14 15:30:16 homebridge evcc[559433]: #011</section>
```
J

andig, 2023-06-14:
Da hat Porsche wohl fleissig weiter geschraubt :(

Jimnaldo, 2023-06-19:
is there a fix already?: main  ] ERROR 2023/06/19 20:46:29 creating vehicle Porsche Taycan failed: cannot create vehicle 'template': cannot create vehicle 'porsche': login failed: no auth code

psattl, 2023-06-19:
> is there a fix already?: main ] ERROR 2023/06/19 20:46:29 creating vehicle Porsche Taycan failed: cannot create vehicle 'template': cannot create vehicle 'porsche': login failed: no auth code

No, with 0.118.1 - I have still no connection with Porsche. Same problem here.

andig, 2023-06-19:
Which is exactly why this issue is still open!

Jimnaldo, 2023-06-19:
> Which is exactly why this issue is still open!

okay, it wasn't clear for me ;-) my german is not that good, thx

andig, 2023-06-19:
Basic support is already working again:

```
Soc:           54%
Capacity:      0.0kWh
Charge status: no emobility api
Range:         202km
Odometer:      12273km
Finish time:   no emobility api
Climater:      no emobility api
```

snapscanii, 2023-06-23:
Hallo zusammen,

gibt es mittlerweile eine Lösung für das Porsche auth Problem? Was muss man tun? Bin noch neu hier und hoffe auf Verständnis für diese Frage. 

Ansonsten ist evcc absolute Spitze :)

closed 2023-06-25 by andig

andig, 2023-06-27:
> Läuft hier das Token wegen Nichtbenutzung ab?

Das... müsste in der Porsche Doku stehen. Ich hab mal Cookie aktiviert, bitte nochmal probieren. Danach bliebe noch token heartbeat...

MarkusGH, 2023-06-29:
Mit der 0.118.1+1687919667 ist das Problem immer noch da

andig, 2023-06-29:
Sagt mir nichts. Ist der Commit da drin?

MarkusGH, 2023-06-29:
Du stellst Fragen :-) Nach meinem Verständnis ist das der letzte erfolgreiche Build (der von gestern, da der nightly Build heute nicht erfolgreich gebaut wurde. 
evcc -v meldet: evcc version 0.118.1 (4d516d58)

So ging es heute morgen los:

`
Jun 29 08:22:59 evcc[3804271]: [porsche] TRACE 2023/06/29 08:22:59 POST https://identity.porsche.com/oauth/token
Jun 29 08:22:59 evcc[3804271]: [porsche] TRACE 2023/06/29 08:22:59 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&grant_type=refresh_token&refresh_token=v1.MdShQoEKFgH9Ke4iBp8-VKtzdhqjb0zMlltsJDTawHmzeP_Z8puSL38RotY0rdtmCoAAAD6JcU_StUCDyVqX7NA
Jun 29 08:22:59 evcc[3804271]: --
Jun 29 08:22:59 evcc[3804271]: {"error":"invalid_grant","error_description":"Unknown or invalid refresh token."}
Jun 29 08:22:59 evcc[3804271]: [porsche] TRACE 2023/06/29 08:22:59 POST https://identity.porsche.com/oauth/token
Jun 29 08:22:59 evcc[3804271]: [porsche] TRACE 2023/06/29 08:22:59 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&grant_type=refresh_token&refresh_token=v1.MS3oC0qqmnOyAeqT0aLjoSfSi357VAi8R4aCwA3XwGghiW5ax31xsR3stD0q5hvgdvEAF4OgJ4JJ5TIpOr_3KBg
Jun 29 08:22:59 evcc[3804271]: --
Jun 29 08:22:59 evcc[3804271]: {"error":"invalid_grant","error_description":"Unknown or invalid refresh token."}
Jun 29 08:22:59 evcc[3804271]: [lp-1  ] WARN 2023/06/29 08:22:59 vehicle soc: Get "https://api.porsche.com/vehicle-data/de/de_DE/status/WP1ZZZ9YZNDA15232": oauth2: "invalid_grant" "Unknown or invalid refresh token." (ignored by estimator)
...
Jun 29 08:23:05 evcc[3804271]: [porsche] TRACE 2023/06/29 08:23:05 POST https://identity.porsche.com/oauth/token
Jun 29 08:23:05 evcc[3804271]: [porsche] TRACE 2023/06/29 08:23:05 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&grant_type=refresh_token&refresh_token=v1.MdShQoEKFgH9Ke4iBp8-VKtzdhqjb0zMlltsJDTawHmzeP_Z8puSL38RotY0rdtmCoAAAD6JcU_StUCDyVqX7NA
Jun 29 08:23:05 evcc[3804271]: --
Jun 29 08:23:05 evcc[3804271]: {"error":"invalid_grant","error_description":"Unknown or invalid refresh token."}
Jun 29 08:23:05 evcc[3804271]: [porsche] TRACE 2023/06/29 08:23:05 POST https://identity.porsche.com/oauth/token
Jun 29 08:23:05 evcc[3804271]: [porsche] TRACE 2023/06/29 08:23:05 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&grant_type=refresh_token&refresh_token=v1.MS3oC0qqmnOyAeqT0aLjoSfSi357VAi8R4aCwA3XwGghiW5ax31xsR3stD0q5hvgdvEAF4OgJ4JJ5TIpOr_3KBg
Jun 29 08:23:05 evcc[3804271]: --
Jun 29 08:23:05 evcc[3804271]: {"error":"invalid_grant","error_description":"Unknown or invalid refresh token."}
`

MarkusGH, 2023-06-29:
Will man da vielleicht einfach ein error handling einbauen, das in der Situation neu verbindet?
Denn wenn z. B. bei Porsche der Server neu gestartet wird ist das Token doch auch ungültig (falls es dort nicht persisitiert wird).

MarkusGH, 2023-06-29:
> Und gibts irgendeine Indikation, nach welcher Zeit der Refresh Token ausläuft? Hast Du mal geschaut, wie lange der initiale Access Token gültig ist?

Letzte erfolgreiche Abfrage war gestern 21:04.

andig, 2023-06-29:
Wenns irgend geht würde ich das gerne vermeiden und user/passwort nicht länger im Speicher halten als notwendig.

MarkusGH, 2023-06-29:
Schön und gut, aber die Erstanmeldung muss ja auch irgendwie gehen.
Und die Daten liegen ja eh unverschlüsselt in der Config.

andig, 2023-06-29:
Darüber zu diskutieren, was "jemand" coden könnte ist irgendwie unbefriedigend ;)

MarkusGH, 2023-06-29:
Alles gut - ich kann nur Vorschläge machen, da ich an dem Teil nicht die geringste Ahnung habe. Alles wie im richtigen Leben :-)

guidob2000, 2023-06-30:
Funktioniert bei irgendwem zur Zeit die Abfrage eines Porsche Taycan?
Ich bekomme nur noch folgende Meldung:
_vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'porsche': login failed: no auth code_
Die Login Daten sind korrekt und wurden auch nicht geändert.

MarkusGH, 2023-06-30:
Porsche PHEV funktioniert mit dem Nightly Build

guidob2000, 2023-06-30:
Bei mir nicht. Wie kann ich feststellen, ob es geladen wurde? Version 0.118.1 ist drauf

andig, 2023-06-30:
Ich mache hier mal zu. Wie oben geschrieben: *nightly* Version.