evcc-io/evcc #8354: UI-Menü "Hilfe benötigt?" erweitern

VolkerK62, 2023-06-08:
Idee ist, die Hilfe um den Punkt "Problem melden" zu erweitern.
Damit soll dann `evcc discuss` aufgerufen werden.

andig, 2023-07-03:
@naltatis wie findest Du die Idee? Dürfte ich das Dir assignen?

naltatis, 2023-07-03:
Mir ist noch nicht ganz klar wie ihr euch das vorstellt. Eigener API Endpunkt der dann die evcc.yaml und co ermittelt und bereitstellt? Ich find das prinzipiell gut, sehe da aber das Potential, dass wir dann über eine API geschützte Config-Daten offen rausgeben. Klar es wird redactet, aber das ist ja nicht immer perfekt (plugins und so).

Ist aber genau die gleiche Frage, die ich mir bei den Config-APIs stelle. Die einfach offen im eigenen Heimnetz zu haben oder sich drauf zu verlassen, dass der User sich da selbst ne Authentifizierung vorbaut finde ich nicht ausreichend. Würde bei der Funktion auf jeden Fall eine Art Admin-Passwortabfrage o.ä. erwarten um es Secure by Default zu machen.

VolkerK62, 2023-07-03:
Manches stelle ich mir dann doch wohl zu einfach vor ;-) (wenn man halt keine Ahnung hat)
Ich dachte, man könnte einfach, statt auf dem Kommandozeile `evcc discuss` aufzurufen, im UI einen Menüpunkt anklicken.

andig, 2023-07-09:
Die Security ist tatsächlich schlechter als bei `evcc discuss`. Da braucht man zusätzlich noch Shellzugang- hier kann das jeder Anwender über das Netz. Vor diesem Hintergrund wäre ich auch lieber defensiv- wontfix :(

closed 2023-07-09 by andig