evcc-io/evcc #8368: Wrong outgoing power value for solaredge inverter

botux76, 2023-06-09:
### Describe the bug

Hi,
my evcc is well configured and nearly working perfectly except from this issue:

The power value for outgoing current of my solaredge SE-5000 inverter is not correct.
I have added 2 screenshots taken at the same time to proof it.
And from my point of view it does not make any sense if evcc is showing 3.5kW going out while only 2.5kW are produced by PV.
Most probably the wrong value of the modbus record is taken, so I would like to discuss with the maintainer of the solaredge inverter module what to change here.

Kind regards,
Tobias

[image]
[image]

### Steps to reproduce

1. Run evcc with the configuration from below using a solaredge SE-5000 inverter
2. Wait for sun :-)
...

### Configuration details

```yaml
meters:
- type: template
  template: solaredge-hybrid 
  id: 2  
  host: 192.168.8.40  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 2  
  host: 192.168.8.40  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solaredge-hybrid 
  id: 2  
  host: 192.168.8.40  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
```

### Log details

```text
Jun 09 07:57:51 homepi evcc[5684]: [eebus ] DEBUG 2023/06/09 07:57:51 avahi - error resolving service: Timeout reached
Jun 09 07:57:55 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:57:55 ----
Jun 09 07:57:55 homepi evcc[5684]: [lp-1  ] DEBUG 2023/06/09 07:57:55 charge power: 0W
Jun 09 07:57:55 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:57:55 pv power: 2796W
Jun 09 07:57:55 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:57:55 battery soc: 100%
Jun 09 07:57:55 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:57:55 battery power: 0W
Jun 09 07:57:55 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:57:55 grid power: -3507W
Jun 09 07:57:55 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:57:55 site power: -3507W
Jun 09 07:57:55 homepi evcc[5684]: [lp-1  ] DEBUG 2023/06/09 07:57:55 charger status: A
Jun 09 07:57:56 homepi evcc[5684]: [eebus ] DEBUG 2023/06/09 07:57:56 avahi - error resolving service: Timeout reached
Jun 09 07:58:01 homepi evcc[5684]: [eebus ] DEBUG 2023/06/09 07:58:01 avahi - error resolving service: Timeout reached
Jun 09 07:58:05 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:05 ----
Jun 09 07:58:05 homepi evcc[5684]: [lp-1  ] DEBUG 2023/06/09 07:58:05 charge power: 0W
Jun 09 07:58:05 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:05 pv power: 2745W
Jun 09 07:58:05 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:05 battery soc: 100%
Jun 09 07:58:05 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:05 battery power: 0W
Jun 09 07:58:05 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:05 grid power: -3528W
Jun 09 07:58:05 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:05 site power: -3528W
Jun 09 07:58:05 homepi evcc[5684]: [lp-1  ] DEBUG 2023/06/09 07:58:05 charger status: A
Jun 09 07:58:15 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:15 ----
Jun 09 07:58:15 homepi evcc[5684]: [lp-1  ] DEBUG 2023/06/09 07:58:15 charge power: 0W
Jun 09 07:58:15 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:15 pv power: 2770W
Jun 09 07:58:15 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:15 battery soc: 100%
Jun 09 07:58:15 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:15 battery power: 0W
Jun 09 07:58:15 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:15 grid power: -3498W
Jun 09 07:58:15 homepi evcc[5684]: [site  ] DEBUG 2023/06/09 07:58:15 site power: -3498W
Jun 09 07:58:15 homepi evcc[5684]: [lp-1  ] DEBUG 2023/06/09 07:58:15 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.117.4

andig, 2023-06-09:
Zeig bitte mal

    evcc meter

passend zum Screenshot

botux76, 2023-06-09:
Hi Andi!
Kann ich gern gleich posten. Aber ich habe schon eine Teilantwort.
Mein komplettes Setup sieht so aus
- 8kW PV
- Solaredge SE-5000
- Solaredge SE-2000
- Telsa Powerwall über Solaredge Storedge angebunden

Die Steuerung erfolgt ausschließlich über den SE-5000. Ich bekomme alle Werte über Modbus/TCP - auch die vom SE-2000.
Allerdings werden im EVCC die Werte des SE-2000 nicht berücksichtigt, nur die Leistung des SE-5000 fließt ein.
Kann EVCC die Werte automatisch aufaddieren? Oder ist die Funktion nicht vorhanden?

$ evcc meter
[eebus ] DEBUG 2023/06/09 08:26:34 mdns: announce
[eebus ] DEBUG 2023/06/09 08:26:34 mdns: using avahi
[eebus ] DEBUG 2023/06/09 08:26:34 starting websocket server on :4712
grid1
-----
Power: -498W

pv2
---
Power: 3122W

battery3
--------
Power: -0W
Soc:   100%

[eebus ] DEBUG 2023/06/09 08:26:35 mdns: stop announcement