evcc-io/evcc #8372: Hausverbrauch wird falsch berechnet

mseele, 2023-06-09:
### Describe the bug

Seit einer Weile stelle ich das Problem fest das der Hausverbrauch manchmal als 0 angezeigt wird und dadurch die ganze Überschussladung durcheinander kommt und immer auch mit einigem an Netzstrom geladen wird.
Noch im Frühjahr hat es - meine ich - problemlos funktioniert. Die Technik noch die evcc Konfiguration hat sich bei mir verändert.

Hier ein Video und zwei Bilder von Tibber bei dem man den Effekt schön sieht:

https://github.com/evcc-io/evcc/assets/15141672/95710e4d-6d33-45e6-9a13-170d3488353a

[image]

[image]

Für weiteres Debugging etc. stehe ich sehr gerne zur Verfügung.

### Steps to reproduce

1. Auto mit PV Überschussladen
...

### Configuration details

```yaml
[main  ] INFO 2023/06/09 12:06:49 evcc 0.117.4
[main  ] INFO 2023/06/09 12:06:49 using config file: /etc/evcc.yaml
[db    ] INFO 2023/06/09 12:06:50 using sqlite database: /home/mseele/.evcc/evcc.db

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:9091
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 9091

log: info
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
  - name: grid
    type: template
    template: solarlog
    usage: grid
    host: 192.168.178.5
    port: 502

  - name: pv
    type: template
    template: solarlog
    usage: pv
    host: 192.168.178.5
    port: 502

chargers:
  - name: wallbox
    type: template
    template: go-e
    host: 192.168.178.7
[... 646 lines cut ...]
[lp-1  ] DEBUG 2023/06/09 11:49:12 charge power: 4270W
[site  ] DEBUG 2023/06/09 11:49:12 pv power: 7016W
[site  ] DEBUG 2023/06/09 11:49:12 grid power: -559W
[site  ] DEBUG 2023/06/09 11:49:12 site power: -559W
[lp-1  ] DEBUG 2023/06/09 11:49:12 charge voltages: [233 236 236]V
[lp-1  ] DEBUG 2023/06/09 11:49:12 detected connected phases: 3p
[lp-1  ] DEBUG 2023/06/09 11:49:12 charge currents: [6 5.9 6]A
[lp-1  ] DEBUG 2023/06/09 11:49:12 detected active phases: 3p
[lp-1  ] DEBUG 2023/06/09 11:49:12 charge meter total import: 3142.000kWh
[lp-1  ] DEBUG 2023/06/09 11:49:12 charger status: C
[lp-1  ] DEBUG 2023/06/09 11:49:12 soc estimated: 70.44% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/06/09 11:49:12 vehicle soc: 70%
[lp-1  ] DEBUG 2023/06/09 11:49:12 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/06/09 11:49:12 vehicle range: 350km
[lp-1  ] DEBUG 2023/06/09 11:49:12 pv charge current: 6.81A = 6A + 0.81A (-559W @ 3p)
[lp-1  ] DEBUG 2023/06/09 11:49:13 charge meter total import: 3142.000kWh
```

### What type of operating system are you running?

Linux

### Version

0.117.4

VolkerK62, 2023-06-09:
Das Problem ist `solarlog` als Grid-Meter. https://docs.evcc.io/docs/devices/meters#solarlog

```
[lp-1  ] DEBUG 2023/06/09 11:45:12 charge power: 11010W
[site  ] DEBUG 2023/06/09 11:45:12 pv power: 6955W
[site  ] DEBUG 2023/06/09 11:45:12 grid power: -4822W

[lp-1  ] DEBUG 2023/06/09 11:45:22 charge power: 11020W
[site  ] DEBUG 2023/06/09 11:45:22 pv power: 6961W
[site  ] DEBUG 2023/06/09 11:45:22 grid power: 6996W
```
Innerhalb von 10 Sek von 4822W Einspeisung auf 6996W Netzbezug. Solche Sprünge sieht man im gesamten Verlauf des Log.

Das Problem hatte ich am Anfang auch. 
Jetzt nutze ich SolarLog nur noch für PV und als Grid habe ich einen Shelly3EM.

premultiply, 2023-06-09:
Solarlog ist in jeder Hinsicht nur ein absolute Notlösung. Wenn die SD-Karte gerade nicht kaputt ist oder das Teil mal wieder mit sich selbst beschäftigt ist, ist die gesamte Messung über S0-Zähler sowas von ungenau und selten aktualisiert dass es keinen Nutzen macht.

Quasi jede andere Lösung ist besser.

andig, 2023-06-09:
@VolkerK62 sollten wir das einfach ausbauen oder im Solarlog Template wenigstens abraten?