evcc-io/evcc #8379: Cannot Read/write coils with modbus plugin

kscholty, 2023-06-09:
**Is your feature request related to a problem? Please describe.**
I want to implement a custom charger based on a phoenix-em-eth controller. 
This controller uses coils for the "Enabled" flag. Therefore, I need to be able to read and write coils.

**Describe the solution you'd like**

- Add `"coil"` for type when reading.
- Add` "writesinglecoil"` to type when writing

Ideally the `"decode"` would automatically be `"bool"`, otherwise allow setting` "decode"` to `"bool"`.

**Describe alternatives you've considered**
Alternatively the `"decode"` type `"bool"` could automatically switch the `"type"` to` "coil"`. But that would not work when writing, since this has no type. Therefore, we would need a `"decode"` for writing as well.

**Additional context**
Looking into the code "RegisterOperation" in util/modbus/modbus.go looks like a good starting point...

StefanSchoof, 2023-06-09:
But why do you want a custom charger for a charger that is Supported? https://docs.evcc.io/docs/devices/chargers/#em-cp-pp-eth-modbus-tcp

kscholty, 2023-06-09:
> But why do you want a custom charger for a charger that is Supported? https://docs.evcc.io/docs/devices/chargers/#em-cp-pp-eth-modbus-tcp

Because I enhanced my wallbox with a phase switcher. It uses 2 of the outputs of the controller to switch between 1 and 3 phases.
See also my question #8373
Since this is not something a lot of people will use I doubt that I can bring this into evcc (@andig ?) I wanted to try to use a custom charger module. Unfortunatley, it seems that's not so easy.

premultiply, 2023-06-09:
Apart from this issue here:
Can you please add a discussion about the switching circuit you have designed and added?

Maybe it can be added officially to the charger implementation if it fits some basic safety rules and other users can rebuild it.

kscholty, 2023-06-09:
Just opened PullRequest  #8385
This should fix this one here.

kscholty, 2023-06-10:
> Apart from this issue here: Can you please add a discussion about the switching circuit you have designed and added?
> 
> Maybe it can be added officially to the charger implementation if it fits some basic safety rules and other users can rebuild it.

See #8386

StefanSchoof, 2023-06-10:
Ich habe im Kopf, dass der custom charger keine 1p/3p Wechsel kann. Wenn das stimmt, löst das dein Problem nicht.

andig, 2023-06-10:
Kann er mittlerweile 🙄

closed 2023-06-11 by andig

Steffen294, 2023-06-24:
Hallo kscholty,

ich benötige auch die Funktion um ein Modbus Coil schalten zu können.
Wie sieht die yaml config dazu aus? Die Modbus Doku bringt mich nicht weiter.

Grüße Steffen

kscholty, 2023-06-24:
Hi,
ich habe die Doku angepasst, aber der Pull Request ist noch nicht approved.
Schau mal hier
https://github.com/evcc-io/docs/pull/399
Ich hoffe, das hilft dir.

PR #8385 (2023-06-11): Reading and writing coils in modbus plugin 
This tries to fix #8379 
It implements the register type coil. This will automatically set decode to bool (1 byte boolean).
Also write singecoil can be used as type now. This will also automatically use bool as datatype.
I hope I didn't miss anything