evcc-io/evcc #8380: Start with FULLLOAD "Clean energy available" without set target time and only 7min remaining

PanNwt, 2023-06-09:
### Describe the bug

Start with FULLLOAD "Clean energy available" without set target time and only 7min remaining.

After setting to 1 Phase only PV charging started even with max Current without any need.

### Steps to reproduce

1. Start charging with PV Mode 
2. The cleanest Energy hours must be shortly before charging end
3. Wait until some minutes before end of the cleanest hour

- [image]
- [image]
- [Log with both LPs.txt](https://github.com/evcc-io/evcc/files/11709252/Log.with.both.LPs.txt)

### Configuration details

```yaml
network:
  host: 192.168.<x.yyy>
  port: 80 # uri for ui
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: <available>

telemetry: true
plant: <set>

# log settings
log: info
levels:
  core: trace
  lp-1: trace
  lp-2: trace

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: grid
  type: discovergy
  user: <myEmailAddress>
  password: <discovergyPassword> # password
  meter: <grid>
  scale: 1
- name: pv
  type: discovergy
  user: <myEmailAddress>
  password: <discovergyPassword> # password
  meter: <production>
  scale: -1

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: go-eCharger
  type: template
  template: go-e
  host: <192.168.x.yyy>
- name: go-eHomeFix
  type: template
  template: go-e-v3
  host: <192.168.x.yyy> 
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
[... 354 lines cut ...]
[lp-1  ] DEBUG 2023/06/09 20:05:41 vehicle range: 315km
[lp-1  ] DEBUG 2023/06/09 20:05:42 charger: guard elapse
[lp-1  ] DEBUG 2023/06/09 20:05:42 guard timer inactive
[lp-1  ] DEBUG 2023/06/09 20:05:42 charger disable
[lp-1  ] DEBUG 2023/06/09 20:05:42 wake-up timer: stop
[lp-1  ] DEBUG 2023/06/09 20:05:47 charge power: 0W
[lp-1  ] DEBUG 2023/06/09 20:05:57 charge power: 0W
[lp-1  ] DEBUG 2023/06/09 20:05:57 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/09 20:05:57 charge meter total import: 2819.691kWh
[lp-1  ] DEBUG 2023/06/09 20:05:57 charger status: B
[lp-1  ] INFO 2023/06/09 20:05:57 stop charging <-
[lp-1  ] DEBUG 2023/06/09 20:05:57 charge meter total import: 2819.691kWh
[lp-1  ] DEBUG 2023/06/09 20:05:58 soc estimated: 79.52% (vehicle: 79.00%)
[lp-1  ] DEBUG 2023/06/09 20:05:58 vehicle soc: 80%
[lp-1  ] DEBUG 2023/06/09 20:05:58 vehicle range: 316km
[lp-1  ] DEBUG 2023/06/09 20:06:07 charge power: 0W
```

### What type of operating system are you running?

Docker container

### Version

v0.117.4

PanNwt, 2023-06-09:
Hatte jetzt noch den Fehler gesehen, dass evcc diesen Parameter an dieser Stelle nicht möchte. Wo gehört er hin?

andig, 2023-06-09:
> the default threshold 135 g/kWh is used and all cars are started again.

was meinst du damit? Wo soll dieses Default her kommen? Zeig mal bitte deine Settings.

andig, 2023-06-09:
Ich mache mal zu. Es ist unklar was hier der Fehler sein soll oder wo im Logfile der zu sehen wäre. Es ist unklar, was ein „fullload“ ist. Es gibt auch kein spezielles Handling für „x Minuten vor Ladeende“.

closed 2023-06-09 by andig

PanNwt, 2023-06-10:
Das war ja mein Frage! Ich hatte keine Ahnung, dass in der Detailansicht der Energieflüsse sich ein Settingsdialog fürs Laden versteckt. Sorry überraschend.

PanNwt, 2023-06-10:
Und wenn evcc bei 2,x PV auf 3phasig 16A für fehlende 0,5 kWh schaltet ist dies ein Fullload(charge).

PanNwt, 2023-06-11:
Sorry nochmal für den falsche Ton am Anfang.

Nichtsdestotrotz würde ich mir für die Zukunft zwei Dinge wünschen:
1. etwas mehr Details, welche Settings gemeint sind.
2. etwas mehr Zeit um auf Fragen reagieren zu dürfen. Ein closed gibt einem das Gefühl, des zu langsam zu sein.

andig, 2023-06-11:
@naltatis wollen wir das "saubere Energie verfügbar" und die entsprechende Preisnachricht vllt. noch um die gültige Schwelle und den aktuellen Wert ergänzen? Falls "autocharge=true" Beides auch noch mit ins Logfile für site aufnehmen?

naltatis, 2023-06-12:
Das Anzeigen der eingestellten Schwelle in der Status-Nachricht finde ich sinnvoll. Von da aus sollte sich dann auch die Schwelle verändern lassen (Netzladen Dialog). Und ja, im Log sollte das natürlich auch stehen.

Den Punkt, dass die Grid und Batterie Einstellung aktuell etwas versteckt sind nehme ich auch mit, bzw. wurde auch schon an anderer Stelle angemerkt. Aber das würde ich als ein separates Thema für ggf. später sehen.