evcc-io/evcc #8381: BMW API not working in v0.117.4

fxfxfx666, 2023-06-09:
### Describe the bug

After the last update to v0.117.4 my BMW is no longer recognized:
[image]
[image]

### Steps to reproduce

1. connect the car to the charger
2. car is no longer recognized, see pictures, and the battery status is not visible

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.178.123
  name: grid1
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.178.125
  password: xxx
  name: pv2

chargers:
- type: template
  template: keba
  host: 192.168.178.184
  name: wallbox4

vehicles:
- type: template
  template: bmw
  title: iX3
  user: xxx
  password: xxx
  vin: WBYxxx
  capacity: 80
  name: ev3

loadpoints:
- title: Einfahrt
  charger: wallbox4
  vehicle: ev3
  mode: pv # charge mode (off, now, minpv, pv)
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
```

### Log details

```text
lp-1  ] WARN 2023/06/09 20:53:22 phases not configured, assuming 3p
[semp  ] WARN 2023/06/09 20:53:22 SEMP_BASE_URL unspecified, using http
://192.168.178.154:7070 instead
[site  ] INFO 2023/06/09 20:53:22 site config:
[site  ] INFO 2023/06/09 20:53:22   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/06/09 20:53:22     grid:      power ✓ energy ✓ curre
nts ✓
[site  ] INFO 2023/06/09 20:53:22     pv 1:      power ✓ energy ✓ curre
nts ✓
[site  ] INFO 2023/06/09 20:53:22   vehicles:
[site  ] INFO 2023/06/09 20:53:22     vehicle 1: range ✗ finish ✗ statu
s ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/06/09 20:53:22 loadpoint 1:
[lp-1  ] INFO 2023/06/09 20:53:22   mode:        pv
[lp-1  ] INFO 2023/06/09 20:53:22   charger:     power ✓ energy ✓ curre
nts ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/06/09 20:53:22   meters:      charge ✓
[lp-1  ] INFO 2023/06/09 20:53:22     charge:    power ✓ energy ✓ curre
nts ✓
[lp-1  ] DEBUG 2023/06/09 20:53:22 phase timer inactive
[lp-1  ] DEBUG 2023/06/09 20:53:22 pv timer inactive
[lp-1  ] DEBUG 2023/06/09 20:53:22 guard timer inactive
[lp-1  ] INFO 2023/06/09 20:53:22 vehicle updated: unknown -> iX3 (offl
ine)
[lp-1  ] DEBUG 2023/06/09 20:53:22 max charge current: 6A
[site  ] DEBUG 2023/06/09 20:53:22 ----
[lp-1  ] DEBUG 2023/06/09 20:53:22 charge power: 0W
[site  ] DEBUG 2023/06/09 20:53:22 pv power: 222W
[site  ] DEBUG 2023/06/09 20:53:22 grid power: 423W
[site  ] DEBUG 2023/06/09 20:53:22 grid powers: [76 344 2]W
[site  ] DEBUG 2023/06/09 20:53:22 grid currents: [0.81 1.72 0.614]A
[site  ] DEBUG 2023/06/09 20:53:22 site power: -577W
[lp-1  ] DEBUG 2023/06/09 20:53:22 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/09 20:53:22 charge meter total import: 2863.442k
Wh
[lp-1  ] DEBUG 2023/06/09 20:53:22 charger status: B
[lp-1  ] DEBUG 2023/06/09 20:53:22 charge start energy: 2.86e+03kWh
[lp-1  ] INFO 2023/06/09 20:53:22 car connected
[lp-1  ] DEBUG 2023/06/09 20:53:22 pv timer elapse
[lp-1  ] DEBUG 2023/06/09 20:53:22 pv timer inactive
[lp-1  ] DEBUG 2023/06/09 20:53:22 charger: guard elapse
[lp-1  ] DEBUG 2023/06/09 20:53:22 guard timer inactive
[lp-1  ] DEBUG 2023/06/09 20:53:22 charge meter total import: 2863.442k
Wh
[lp-1  ] DEBUG 2023/06/09 20:53:22 charger vehicle id: 93a7a4de00000000
[lp-1  ] ERROR 2023/06/09 20:53:22 vehicle soc: vehicle not available: 
cannot create vehicle 'template': cannot create vehicle 'bmw': unexpect
ed status: 401 (Unauthorized)
[lp-1  ] DEBUG 2023/06/09 20:53:22 pv charge current: 0.836A = 0A + 0.8
36A (-577W @ 3p)
[lp-1  ] DEBUG 2023/06/09 20:53:22 pv disable timer reset
[lp-1  ] DEBUG 2023/06/09 20:53:22 pv timer inactive
[site  ] DEBUG 2023/06/09 20:53:32 ----
[lp-1  ] DEBUG 2023/06/09 20:53:32 charge power: 0W
[site  ] DEBUG 2023/06/09 20:53:32 pv power: 222W
[site  ] DEBUG 2023/06/09 20:53:32 grid power: 424W
```

### What type of operating system are you running?

Linux

### Version

0.117.4

andig, 2023-06-09:
Working fine here. Please send credentials to info@evcc.io.

closed 2023-06-09 by andig

Hofyyy, 2023-06-10:
Had the same problem. 
Go to the bmw homepage.
They will force a passwort change. then it will work again.

fxfxfx666, 2023-06-10:
> Had the same problem.
> Go to the bmw homepage.
> They will force a passwort change. then it will work again.

Thanks, that was it. Sorry for the confusion!