evcc-io/evcc #8396: charger: invalid status: E

hardoverflow, 2023-06-10:
### Describe the bug

After updated from 0.117.4 to 0.118.0 i got the the following error message:

```
Jun 10 13:26:33 raspberrypi evcc[1426]: [lp-1  ] ERROR 2023/06/10 13:26:33 charger: invalid status: E
Jun 10 13:26:43 raspberrypi evcc[1426]: [lp-1  ] ERROR 2023/06/10 13:26:43 charger: invalid status: E
Jun 10 13:26:53 raspberrypi evcc[1426]: [lp-1  ] ERROR 2023/06/10 13:26:53 charger: invalid status: E
```

### Steps to reproduce

Update from 0.117.4 to 0.118.0.

### Configuration details

```yaml
---
network:
  schema: http
  host: ***redacted***
  port: 7070

log: info
levels:
  cache: error

plant: ***redacted***

interval: 10s

meters:
  - type: template
    template: solaredge-inverter
    id: 1
    host: ***redacted***
    port: 1502
    usage: grid
    modbus: tcpip
    name: grid1
  - type: template
    template: solaredge-inverter
    id: 1
    host: ***redacted***
    port: 1502
    usage: pv
    modbus: tcpip
    name: pv1

chargers:
  - name: alfen
    type: template
    template: alfen
    modbus: tcpip
    id: 1
    host: ***redacted***
    port: 502
loadpoints:
  - title: Carport
    charger: alfen
    mode: pv
    phases: 0
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.4588
  feedin:
    type: fixed
    price: 0.0673
  co2:
    type: grünstromindex
    zip: ***redacted***

site:
  title: My home
  meters:
    grid: grid1
    pv: pv1

modbusproxy:
  - port: 1502
    uri: ***redacted***:1502

mqtt:
  broker: localhost:1883
  topic: evcc
  user: evcc
  password: ***redacted***

influx:
  url: ***redacted***
  database: evcc
  token: ***redacted***
  org: ***redacted***

telemetry: true

sponsortoken: ***redacted***
```

### Log details

```text
root@raspberrypi:~# evcc --log debug
[main  ] INFO 2023/06/10 13:30:40 evcc 0.118.0
[main  ] INFO 2023/06/10 13:30:40 using config file: /etc/evcc.yaml
[main  ] INFO 2023/06/10 13:30:40 starting ui and api at :7070
[db    ] INFO 2023/06/10 13:30:50 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/06/10 13:30:50 connecting evcc-1382451112 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/06/10 13:30:50 tcp://localhost:1883 connected
[proxy-1502] DEBUG 2023/06/10 13:30:50 modbus proxy for 10.189.21.220:1502 listening at :1502
[alfen ] DEBUG 2023/06/10 13:30:52 detected 3p alfen
[site  ] DEBUG 2023/06/10 13:30:52 co2 tariff
[site  ] INFO 2023/06/10 13:30:52 site config:
[site  ] INFO 2023/06/10 13:30:52   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/06/10 13:30:52     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/06/10 13:30:52     pv 1:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2023/06/10 13:30:52 loadpoint 1:
[lp-1  ] INFO 2023/06/10 13:30:52   mode:        pv
[lp-1  ] INFO 2023/06/10 13:30:52   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/06/10 13:30:52   meters:      charge ✓
[lp-1  ] INFO 2023/06/10 13:30:52     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2023/06/10 13:30:52 co2 tariff
[lp-1  ] DEBUG 2023/06/10 13:30:52 phase timer inactive
[lp-1  ] DEBUG 2023/06/10 13:30:52 pv timer inactive
[lp-1  ] DEBUG 2023/06/10 13:30:52 guard timer inactive
[site  ] DEBUG 2023/06/10 13:30:53 ----
[lp-1  ] DEBUG 2023/06/10 13:30:53 charge power: 0W
[site  ] DEBUG 2023/06/10 13:30:53 co2 tariff
[site  ] DEBUG 2023/06/10 13:30:53 pv power: 6829W
[site  ] DEBUG 2023/06/10 13:30:53 grid power: -6448W
[site  ] DEBUG 2023/06/10 13:30:53 site power: -6448W
[lp-1  ] DEBUG 2023/06/10 13:30:53 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/10 13:30:53 charge meter total import: 10.322kWh
[lp-1  ] ERROR 2023/06/10 13:30:53 charger: invalid status: E
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.0

200Puls, 2023-06-10:
Im encoutering the same issue after Update.

Using ABB Terra AC with OCPP.

hardoverflow, 2023-06-10:
May be related to https://github.com/evcc-io/evcc/pull/8073

andig, 2023-06-11:
That is intentional since evcc doesn't handle status E at all. According to https://www.goingelectric.de/forum/viewtopic.php?t=36700&start=10:

> In der Anleitung von Mennekes liest man für Zustand E "Kurzschluß in der Leitung".

/cc @premultiply

hardoverflow, 2023-06-12:
I have checked the wallbox. I can confirm that it works perfectly. Screenshot attached.

[image]

hardoverflow, 2023-06-12:
A quick view in the [Implementation of Modbus Slave TCPIP for Alfen NG9xx platform.pdf](https://alfen.com/file-download/download/public/1610) section 3.4.1.

IMHO this is not an "Error". Only state "F" is an error state. Does anyone see this differently?

[image]

andig, 2023-06-12:
Aus https://evsim.gonium.net:

> Die Zustände E und F sind Fehlerzustände, in denen die Stromversorgung zum Elektroauto unterbrochen wird. E zeigt einen Kurzschluss zwischen CP und PE an. Der Zustand F kennzeichnet einen Ausfall der Wallbox: Zwischen CP und PE besteht keine Verbindung.

Selbst wenn E kein Fehlerzustand wäre: wir wüssten nichts damit anzufangen. Das ist jetzt ein echtes Dilemma :(

andig, 2023-06-12:
@hardoverflow Vorschlag: Frage an den Support: WB sagt E, laut Standard ist E Fehler, was ist los...

hardoverflow, 2023-06-14:
I have contacted the Alfen Support with the following content:

> Greetings,
> 
> i use a smart charging management software ([evcc.io](http://evcc.io/)) together with Alfen Eve Single-line Pro and LBA (Load Balancing Active) and i have a general question about the standard.
> 
> The thing is, that the wallbox default mode 3 listening state (without a car connection) is “State E”. I found different meanings or statements about the states. First i have found, that State E and F are error states.
> 
> “The states E and F are fault states in which the power supply to the electric car is interrupted. E indicates a short circuit between CP and PE. State F indicates a failure of the wallbox: There is no connection between CP and PE.”
> 
> https://evsim.gonium.net/#der-control-pilot-cp
>
> In the Alfen documentation about Modbus TCP/IP section 3.4.1. (table) i have found that “State F” is only a error state.
> 
> [Implementation of Modbus Slave TCPIP for Alfen NG9xx platform.pdf](https://alfen.com/file-download/download/public/1610)
>
> Could you please bring a little light into the dark here? Which standard do you adhere to? Or is there a deviation from the standard here and why? Where can I read about this? (Reference of ..)
> 
> Many thanks to you!

engelchrisi, 2023-06-14:
Same issue happy with Innogy Ebox since 1.18. A regression.
Because of the state it seems that the box never realizes that a cable was unlocked/removed and therefore always "thinks" that it is loading.

hardoverflow, 2023-06-14:
Just ask ChatGPT. (Without rating!)

[image]

premultiply, 2023-06-14:
Klingt solide ist aber völliger Blödsinn der sich in keinem Standard abbildet.

andig, 2023-06-14:
It seems different wallboxes or charge controllers have different interpretations of standards. We'll add model-specific mapping of status. Is it correct to say that E ist generally perceived as A, i.e. "no car attached"?

engelchrisi, 2023-06-14:
Auch wenn dies ein Alfen Thread ist, ich hänge mal meine gesammelten Infos zur Innogy Ebox an. es gibt noch keinen separaten Innogy Thread, macht wohl auch keinen Sinn.
Das sind die Stati Änderungen von meiner Box bei einem Ladevorgang mit ID Karte.
E: Keine Kabel angeschlossen
B: Kabel und Auto angeschlossen
C; Ladevorgang
B: Ladevorgang abgeschlossen (Kabel und Auto angeschlossen)
[InnogyEboxLadestatus.pdf](https://github.com/evcc-io/evcc/files/11749505/InnogyEboxLadestatus.pdf)

thizmo, 2023-06-15:
Kann ich bestätigen. Gleicher Fehler auch bei mir, evcc v0.118.0 und Alfen S-Line, 3Ph.
Mit dem Update auf v0.118.0

closed 2023-06-15 by andig

hardoverflow, 2023-06-16:
This is the answer from Alfen Support:

> Beste / Dear Daniel Schade,
> 
> thank you for your message and your advice. You are right, the document is incomplete at this point; state E is also an error state.
> 
> So if the EVSE is in state E or F, it means that there is an error; Communication via the EVSE is not possible. It is also not possible for an EV to charge in these states.
> 
>  
> 
> I hope I have answered your question well.
> 
>  
> 
> Met vriendelijke groet / Kind regards / Mit freundlichen Grüßen,
> 
> Mohammed
> 
> Alfen Charging Equipment Service

engelchrisi, 2023-06-20:
I can confirm it is fixed with 118.1 for Innogy Ebox - thanks for the fast reaction

brandac, 2023-12-09:
Hi everyone, with my Alfen Wallbox I also now get the error "charger: invalid status: F" using evcc version 0.122.1 and latest WB firmware.

StorageFan, 2023-12-19:
I had the same issue today with my Alfen Eve Single Pro-Line. I always got the "charge status: invalid status: F" error back from evcc charger. I tried to reboot the Wallbox using ACE Installer. And for a minute or so the error was gone after rebooting. But then it came back. So I went to the Wallbox to see if the display would show anything. And lo and behold the problem was that I had left the type 2 charging cable still in the wallbox (I had only plugged out the other end from the car the day before). 

Once I had unplugged the charging cable from the wallbox, the error message went a away and "charge status: A" was reported back from the wallbox. I don't know if that will solve the issues reported above as well, but I was baffled how easy a solution sometimes can be. And also a bit disappointed that leaving the cable in the Wallbox to just use it like a Wallbox with a fixed cable doesn't seem to be an option. At least not one without producing a constant stream of error messages in evcc.

premultiply, 2023-12-19:
Please ask Alfen for a better configuration or maybe just to fix the firmware.

andig, 2023-12-19:
F is an error state according to standard. You can either patch evcc source to ignore F or you'll have to live with the error messages. Alfen shouldn't  use F(ault) for this kind of issue.

StorageFan, 2023-12-19:
Good luck with Alfen getting them to do anything if you're not an electrician but just the owner and user of a wallbox.
With one of the last Firmwares they destroyed the first and last line in the logfiles that are shown in the ACE Installer. Usually these lines contained a summary of the time, date, kWh charged and a few other pieces of data. Since that Firmware update it only shows "[Unknown]" in the logfile if you view it from the ACE Installer. I opened a ticket in their system, sent them emails (including to a leading technician) and despite several Firmware updates and updates to the ACE Installer, which have been released in the past months (I think at least 3 or 4 of them) this minor issue hasn't been fixed. So I won't waste any of my time with another issue they will probably be even more happy to ignore. 

And just to avoid any misunderstanding, I wasn't asking for help with my error message here. I was simply trying to help others with the same error message. It cost me roughly two hours trying to debug a perfectly correct evcc.yaml file only to find out that the friggin cable was the culprit for the error message popping up in evcc every 30 seconds or so. So my hope is that others might find this thread when searching for the problem and get it solved faster than I did.

StorageFan, 2023-12-20:
Does anyone here know if there's an option in the settings of the Alfen Wallbox to make it accept a Typ 2 charging cable being plugged in permanently? So that one can use it as if it were a charger with an integrated charging cable?

Right now the F fault state occurs in less than 3 minutes after plugging in a cable. So if you're not fast enough connecting your car with the other end of the cable, the box goes into the error mode F and you can't charge your car anymore. 
T
o resolve that error you **must** unplug the cable in the box. I wasn't able to get the Alfen charger to start charging my car at all until I took out the cable from the box and plugged it back in.