evcc-io/evcc #8397: Variable Einstellungen für Hausspeicher

stritti, 2023-06-10:
**Is your feature request related to a problem? Please describe.**
Je nach Jahreszeit oder Wetterlage würde ich gerne den Akku für das Haus unterschiedlich stark priorisieren wollen. Dazu muss ich aktuell immer die config-Datei anpassen

**Describe the solution you'd like**
Eine Anpassung des Mindestladestatus für den Hausspeicher in Prozent, den man über das UI anpassen kann.

HooLeeChit, 2023-06-10:
Du kannst doch auf der Oberfläche wenn du auf den Ladestand der Batterie klickst, diesen variabel einstellen! 

[image]

stritti, 2023-06-10:
Oh stimmt. Das hatte ich noch gar nicht entdeckt gehabt. Danke!

stritti, 2023-06-10:
Eventuell könnte man das auch noch bei den Einstellungen verlinkten?

closed 2023-06-10 by stritti

andig, 2023-06-11:
/cc @naltatis