evcc-io/evcc #8401: OpenEVSE issue

RikVerstijnen, 2023-06-10:
### Describe the bug

OpenEVSE is connected via API connection (not OCPP).

Error "charger: json: cannot unmarshal number 34.067667 into Go struct field .watthour of type int" is shown.

Car is connected to charger, but EVCC shows it is not.

OpenEVSE firmware: 8.2.3.EU
OpenEVSE Wifi firmware: v4.2.2

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: my_charger
    type: template
    template: openevse
    host: 192.168.2.205
    user: ****
    password: ****
```

### Log details

```text
[open***] TRACE 2023/06/10 21:58:16 GET http://192.168.2.205/status
[open***] TRACE 2023/06/10 21:58:16 {"mode":"STA","wifi_client_connected":1,"eth_connected":0,"net_connected":1,"ipaddress":"192.168.2.205","emoncms_connected":0,"packets_sent":0,"packets_success":0,"mqtt_connected":1,"ocpp_connected":0,"rfid_failure":0,"ohm_hour":"NotConnected","free_heap":159492,"comm_sent":1009249,"comm_success":1008836,"rapi_connected":1,"***_connected":1,"amp":0,"voltage":230,"power":0,"pilot":16,"temp":527.5,"temp_max":562.5,"temp1":false,"temp2":false,"temp3":false,"temp4":527.5,"state":254,"status":"disabled","flags":1280,"vehicle":1,"colour":6,"manual_override":1,"freeram":159492,"divertmode":1,"srssi":-87,"time":"2023-06-10T19:58:17Z","local_time":"2023-06-10T21:58:17+0200","offset":"+0200","session_elapsed":68,"session_energy":0,"total_energy":0.034067667,"total_day":0,"total_week":0,"total_month":0,"total_year":0,"total_switches":228,"elapsed":68,"wattsec":0,"watthour":34.067667,"gfcicount":0,"nogndcount":1,"stuckcount":0,"solar":0,"grid_ie":0,"charge_rate":0,"divert_update":518724,"divert_active":false,"shaper":0,"shaper_live_pwr":0,"shaper_cur":0,"shaper_updated":false,"service_level":2,"limit":false,"ota_update":0,"config_version":7,"claims_version":225,"override_version":1,"schedule_version":1,"schedule_plan_version":1,"limit_version":1,"vehicle_state_update":518724,"tesla_vehicle_count":false,"tesla_vehicle_id":false,"tesla_vehicle_name":false}
[lp-1  ] ERROR 2023/06/10 21:58:16 charger: json: cannot unmarshal number 34.067667 into Go struct field .watthour of type int
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.0

andig, 2023-06-11:
Looks like we need a new api definition.

/cc @sparkyson not sure where you got the api definition from so I couldn't upgrade it?

closed 2023-06-11 by andig

sparkyson, 2023-06-12:
Thanks for the fix! 
For future reference, this is the OpenAPI spec: 
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/api.yml
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/tree/master/models

andig, 2023-06-12:
Unfortunately creating code from spec doesn‘t work without manual steps.