evcc-io/evcc #8405: [main  ] FATAL 2023/06/11 11:45:23 cannot create charger 'wallbox3': cannot create charger 'template': invalid key: fw2

Dosepower, 2023-06-11:
### Describe the bug

Running evcc at homeassistant.

was all time running changed nothing now i getting these errors.

cannot create charger 'wallbox3'
cannot create charger 'template'
invalid key: fw2

### Steps to reproduce

just started evcc
restart
same issue

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
  
network:  
  
  schema: http  
  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  
  port: 7070  
  
  
  
log: debug  
  
levels:  
  
  cache: error  
  
  
  
# unique installation id  
  
plant: *****  
  
  
interval: 10s # control cycle interval  
  
  
  
meters:
  
- type: template  
  
  template: varta   
  
  usage: grid    
  
  host: 192.168.178.10    
  
  port: 502    
  
  name: grid1  
  
- type: template  
  
  template: kostal-piko-legacy   
  
  usage: pv    
  
  host: 192.168.178.32    
  
  user: *****  
  password: *****  
  name: pv2
  
- type: template
  
  template: shelly-1pm 
  
  usage: pv  
  
  host: 192.168.178.105  
  
  name: pv3
  
- type: template  
  
  template: varta   
  
  usage: battery    
  
  host: 192.168.178.10    
  
  port: 502    
  
  name: battery4  
  
    
  
    
  
chargers:  
  
- type: template  
  
  template: tinkerforge-warp   
  
  fw2: true    
  
  host: 192.168.178.68    
  
  port: 1883    
  
  topic: Dosepower    
  
  user: *****  
  password: *****  
  name: wallbox3  
  
  
  
vehicles:  
  
- type: template
  
  template: offline 
  
  title: Kona  
  
  capacity: 39  
  
  name: ev3
  
  phases: 1
  
  mode: pv
  
  mincurrent: 6
  
  maxcurrent: 32
  
- type: template  
  
  template: id   
  
  title: ID 4    
  
  user: *****  
  password: *****  
  vin: *****  
  capacity: 77    
  
  name: ev2
  
  phases: 3
  
  mode: pv
  
  mincurrent: 10
  
  maxcurrent: 16
  
  
  
  
  
loadpoints:  
  
- title: Carport  
  
  charger: wallbox3  
  
  vehicle: ev3  
  
  mode: pv  
  
  phases: 3  
  
  mincurrent: 6  
  
  maxcurrent: 32  
  
  resetOnDisconnect: true  
  
  
  
site:  
  
  title: Zuhause  
  
  meters:  
  
    grid: grid1  
  
    pv:  
  
    - pv2
  
    - pv3
  
    battery:  
  
    - battery4 
  
  residualPower: 100
  
  prioritySoc: 50
  
  bufferSoc: 80
  
  bufferStartSoc: 90
  
  
  
tariffs:
  
  currency: EUR # (default EUR)
  
  grid:
  
    # static grid price
  
    type: fixed
  
    price: 0.39 # [currency]/kWh
  

  
  feedin:
  
    # rate for feeding excess (pv) energy to the grid
  
    type: fixed
  
    price: 0.24 # [currency]/kWh
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/06/11 11:45:23 evcc 0.118.0
[main  ] INFO 2023/06/11 11:45:23 using config file: /config/evcc.yaml
[main  ] INFO 2023/06/11 11:45:23 starting ui and api at :7070
[db    ] INFO 2023/06/11 11:45:23 using sqlite database: /data/evcc.db
[main  ] FATAL 2023/06/11 11:45:23 cannot create charger 'wallbox3': cannot create charger 'template': invalid key: fw2
[main  ] FATAL 2023/06/11 11:45:23 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

andig, 2023-06-11:
> invalid key: fw2

Remove fw2.

closed 2023-06-11 by andig

Dosepower, 2023-06-11:
thy is working