evcc-io/evcc #8406: BMW offline for four days

schr-ma, 2023-06-11:
### Describe the bug

My BMW i3 has been displayed as offline for four days, see screenshot. The following message appears in EVCC: 

> vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': unexpected status: 401 (Unauthorized)

The My BMW app works fine, I can also log in to BMW with my credentials. The configuration has been working for many weeks without any problems. The problem occurs with v0.117.4 latest as well as v0.117.4 nightly.
[image]

### Steps to reproduce

Simply open EVCC, the vehicle is immediately displayed as offline.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: ****** # .local suffix announces the hostname on MDNS
  port: ******

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: ******

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: ****** 
  password: ******  
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: ****** 
  password: ******  
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: ******  
  password: ******  
  name: battery3

chargers:
- type: template
  template: go-e-v3 
  host: ******  
  name: wallbox4
- type: template
  template: go-e-v3 
  host: ******  
  name: wallbox5

vehicles:
- type: template
  template: bmw
  title: BMWi3s
  user: ******
  password: ******.
  vin: ******
  capacity: 38
  phases: 3
  mode: pv
[... 148 lines cut ...]
2023-06-11T09:46:48.363959548Z	stdout	[site  ] INFO 2023/06/11 11:46:47     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
2023-06-11T09:46:48.363916670Z	stdout	[site  ] INFO 2023/06/11 11:46:47     pv 1:      power ✓ energy ✓ currents ✗
2023-06-11T09:46:48.363872453Z	stdout	[site  ] INFO 2023/06/11 11:46:47     grid:      power ✓ energy ✗ currents ✗
2023-06-11T09:46:48.363828729Z	stdout	[site  ] INFO 2023/06/11 11:46:47   meters:      grid ✓ pv ✓ battery ✓
2023-06-11T09:46:48.363781760Z	stdout	[site  ] INFO 2023/06/11 11:46:47 site config:
2023-06-11T09:46:48.363737185Z	stdout	[site  ] WARN 2023/06/11 11:46:47 vehicle 'MercedesEQV300' does not support automatic detection
2023-06-11T09:46:48.363666925Z	stdout	[site  ] WARN 2023/06/11 11:46:47 vehicle 'BMWi3s (offline)' does not support automatic detection
2023-06-11T09:46:48.363615526Z	stdout	[site  ] WARN 2023/06/11 11:46:47 battery configured but residualPower is missing (add residualPower: 100 to site)
2023-06-11T09:46:48.363569067Z	stdout	[site  ] WARN 2023/06/11 11:46:47 deprecated: use 'battery' instead of 'batteries'
2023-06-11T09:46:48.363472436Z	stdout	[site  ] WARN 2023/06/11 11:46:47 deprecated: use 'pv' instead of 'pvs'
2023-06-11T09:46:47.445858570Z	stdout	[site  ] DEBUG 2023/06/11 11:46:47 static grid tariff
2023-06-11T09:46:47.067176599Z	stdout	[main  ] ERROR 2023/06/11 11:46:47 creating vehicle bmwi3s failed: cannot create vehicle 'template': cannot create vehicle 'bmw': unexpected status: 401 (Unauthorized)
2023-06-11T09:46:46.040147166Z	stdout	[db    ] INFO 2023/06/11 11:46:46 using sqlite database: /root/.evcc/evcc.db
2023-06-11T09:46:45.535357474Z	stdout	[main  ] INFO 2023/06/11 11:46:45 starting ui and api at :7070
2023-06-11T09:46:45.532806954Z	stdout	[main  ] INFO 2023/06/11 11:46:45 using config file: /etc/evcc.yaml
2023-06-11T09:46:45.493657582Z	stdout	[main  ] INFO 2023/06/11 11:46:45 evcc 0.117.4 (05e22817)
```

### What type of operating system are you running?

Docker container

### Version

v0.117.4

andig, 2023-06-11:
My i3 works fine. Please provide credentials to info@evcc.io.

closed 2023-06-11 by andig

schr-ma, 2023-06-11:
Thank you for this quick reply - and sorry. In fact, I forgot to create a new password that complies with BMW's new password policy. I thought as long as the My BMW app works, it's not because of that.

andig, 2023-06-11:
> BMW's new password policy

Thank you, good to know!