evcc-io/evcc #8428: Schaltschutz und Warp2 Wallbox: evcc sendet zeitgleich "0" und >"6000" mA

binderth, 2023-06-12:
### Describe the bug

Ich betreibe eine WARP2 Wallbox (Smart) mit evcc.

Heute hat das Fahrzeug bis 80% geladen und vor 5min waren die 80% erreicht. Ich habe dann in der GUI auf 85% erhöht und erhalte:

1. in der GUI den Fehler "charger out of sync: expected disabled, got enabled"
2. selbe Warnung im logfile alle 10sekunden
3. per MQTT zuerst den Befehl `evse\external_current` auf "0", direkt danach auf "6663"
4. ich höre in der Wallbox den Schütz ziehen

5. sobald der Schaltschutz vorbei ist, wird wieder "normal" nurmehr die >6A payloads gesendet.

### Steps to reproduce

1. Auto laden
2. warten auf Ladeziel-Erreichung
3. Ladeziel erhöhen
4. während des Schaltschutzes wird "0" und die Ampere-Zahl an die Wallbox gesendet

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http

  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. Host must be set to the docker host's name
  #host: evcc.local
  host: 0.0.0.0

  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

  # per default, evcc should be available at http://evcc.local:7070

interval: 10s # control cycle interval
log: trace

meters:
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
- type: custom
  name: PVErzeugung
  power:
    source: mqtt
    topic: devices/Kostal/totalPVPower
    timeout: 2m
- type: custom
  name: Netzbezug
  power:
    source: mqtt
    topic: devices/Kostal/totalActivePower
    timeout: 2m
- type: custom
  name: BYD10
  power:
    source: mqtt
    topic: devices/Kostal/actualBatteryChargeDischargePower
    timeout: 2m
  soc:
    source: mqtt
    topic: devices/Kostal/batteryActualSoc
    timeout: 2m

chargers:
# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
- type: template
  template: tinkerforge-warp
  #fw2: true
  host: 192.168.78.18
  port: 1883
  topic: devices/Warp2
  timeout: 30s
  name: WarpWallbox

vehicles:
# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/evcc-io/config#vehicles
- name: 500e
  type: template
  template: offline
  title: Fiat 500e
  capacity: 22
- name: EV6
  type: custom
  title: Kia EV6 # display name for UI
  capacity: 77 # byttery capacity (kWh)
  soc: # battery soc (%)
    source: http
    uri: http://192.168.78.10:8080/rest/items/EV6_batteryStatus
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
  range: # optional electric range (km)
    source: http
    uri: http://192.168.78.10:8080/rest/items/EV6_availableRange 
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
  status: # FahrzeugStatus (%)
    source: http
    uri: http://192.168.78.10:8080/rest/items/EV6_vehicleStatus
    method: GET
    headers:
    - content-type: application/json
    jq: .state|split(" ")[0]|split(".")[0]
  onIdentify: # set defaults when vehicle is identified
    mode: pv
    minSoC: 25
    targetSoC: 80

tariffs:
# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
  currency: EUR # (default EUR)
  grid:
    #type: fixed
    #price: 0.319 # [currency]/kWh
    type: tibber
    token: "xxx"
    homeid: "yyy"
  feedin:
    type: fixed
    price: 0.065 # [currency]/kWh

loadpoints:
# loadpoint describes the charger, charge meter and connected vehicle
- title: Warp
  charger: WarpWallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  vehicle: 500e

site:
# site describes the EVU connection, PV and home battery
  title: Zuhause
  meters:
    grid: Netzbezug
    pvs:
    - PVErzeugung
    batteries:
    - BYD10
  prioritySoC: 66
  bufferSoC: 75
  residualPower: 600

mqtt:
  broker: 192.168.78.18:1883
  clientid: evccClient
  user: m2m
  password: openHAB2m
  topic: servers/evcc
```

### Log details

```text
leider hatte ich aufgrund vergangener Fehleranalysen evcc noch immer auf "info" laufen, von daher leider nicht genügend Info ausser alle 10sec nach erreichten 80%:

[lp-1  ] WARN 2023/06/12 16:40:04 charger out of sync: expected disabled, got enabled
```

Ein Nachstellen nach Neustart von evcc war leider? bzw. zum Glück?! nicht mehr erfolgreich. d.h. nachdem ich in der evcc.yaml auf TRACE geschaltet hatte und evcc neu gestartet habe, lief der Schaltschutz wie erwartet, ohne widersprüchliche Befehle.
```

### What type of operating system are you running?

Docker container

### Version

0.118.0

andig, 2023-06-12:
Ohne Logfile nicht zu analysieren, bitte einschl. mqtt für die warp.

closed 2023-06-12 by andig

binderth, 2023-06-13:
habe versucht, es eben nochmal nachzustellen, leider nicht erfolgreich.
Versuche es später nochmal mit mehr Sonne aka >4,1kWh Überschuss, hat evtl. damit zu tun...