evcc-io/evcc #8432: Easee: wrong API used for enabling smart charging LED

GrimmiMeloni, 2023-06-12:
### Describe the bug

evcc currently uses the charger settings API to enable the smart charging LEDs on Easee wallboxes.
According to the API docs, this call shall only be used to **disable** the smart charging LEDs.
Enabling shall happen via a dedicated command instead, see [here](https://developer.easee.cloud/reference/post_api-chargers-id-commands-smart-charging).

The impact of this bug seems low to zero, as the LEDs still change color. However the following Product Update messages state that smart charging is NOT active. This makes reading the log messages somewhat confusing as reality and API responses are out of sync. Cleaning this up feels like the right thing to do.

### Steps to reproduce

1. Just attach a car to the callbox.
2. Observe 
3.
...

### Configuration details

```yaml
Not required in this case, as this is not controlled by any config.
```

### Log details

```text
[easee ] TRACE 2023/06/12 16:58:14 POST https://api.easee.com/api/chargers/EHCUMYZ7/settings
[easee ] TRACE 2023/06/12 16:58:14 {"smartCharging":true}
--
[{"device":"EHCUMYZ7","commandId":11,"ticks":638221786948600230}]
[easee ] TRACE 2023/06/12 16:58:15 ProductUpdate EHCUMYZ7: LEDMODE 21
[easee ] TRACE 2023/06/12 16:58:15 ProductUpdate EHCUMYZ7: SMART_CHARGING false
[easee ] TRACE 2023/06/12 16:58:15 ProductUpdate EHCUMYZ7: SMART_CHARGING false
[easee ] TRACE 2023/06/12 16:58:16 CommandResponse EHCUMYZ7: {SerialNumber:EHCUMYZ7 ID:11 Timestamp:2023-06-12 14:58:14.860023 +0000 UTC DeliveredAt:2023-06-12 14:58:14.106 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638221786948600230}
```

### What type of operating system are you running?

Linux

### Version

0.118.0

GrimmiMeloni, 2023-06-12:
@andig please assign, happy to have a look

cc @naltatis

andig, 2023-06-12:
Don't spend too much effort there: a) it works and b) it is not substantial for functionality. Can leave as-is.

GrimmiMeloni, 2023-06-19:
Just gave this a quick spin today. The result is actually the same as enabling via settings.

```
[easee ] TRACE 2023/06/19 19:55:38 POST https://api.easee.com/api/chargers/EHXXXXXX/commands/smart_charging
[easee ] TRACE 2023/06/19 19:55:38 
POST /api/chargers/EHXXXXXX/commands/smart_charging HTTP/1.1
Host: api.easee.com
User-Agent: Go-http-client/1.1
Content-Length: 2
Authorization: Bearer ey...2c
Content-Type: application/json
Accept-Encoding: gzip

{}

HTTP/2.0 202 Accepted
Content-Length: 63
Access-Control-Allow-Origin: *
Content-Type: application/json; charset=utf-8
Date: Mon, 19 Jun 2023 17:55:38 GMT
X-Amz-Apigw-Id: Gxx4IHKJDoEFXUA=
X-Amzn-Remapped-Connection: keep-alive
X-Amzn-Remapped-Content-Length: 63
X-Amzn-Remapped-Date: Mon, 19 Jun 2023 17:55:38 GMT
X-Amzn-Remapped-Server: Kestrel
X-Amzn-Requestid: 0cab5b79-1b20-4aff-b496-eaf0c182b6ef

{"device":"EHXXXXXX","commandId":11,"ticks":638227941381616400}
[easee ] TRACE 2023/06/19 19:55:40 ProductUpdate EHXXXXXX: LEDMODE 22
[easee ] TRACE 2023/06/19 19:55:40 ProductUpdate EHXXXXXX: SMART_CHARGING false
[easee ] TRACE 2023/06/19 19:55:40 ProductUpdate EHXXXXXX: SMART_CHARGING false
[easee ] TRACE 2023/06/19 19:55:40 CommandResponse EHXXXXXX: {SerialNumber: EHXXXXXX ID:11 Timestamp:2023-06-19 17:55:38.16164 +0000 UTC DeliveredAt:2023-06-19 17:55:36.748 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638227941381616400}
```

Closing as won't do.

closed 2023-06-19 by GrimmiMeloni

andig, 2023-06-19:
Perfect 👍🏻