evcc-io/evcc #8436: UI: Show cost limit when green/cheap-charging is active

naltatis, 2023-06-12:
Show active smartCostLimit in loadpoint status text and provide an option to change it from there.

_Originally posted by @naltatis in https://github.com/evcc-io/evcc/issues/8380#issuecomment-1588120973_

mhibu, 2023-06-25:
Ich kann das SmartCostLimit nirgendswo einstellen, weder in der Config (unter site) noch im Interface (nicht im Planer und auch nicht bei den Einstellungen). Es wird jedenfalls ständig unter 17cent geladen obwohl ich das eigentlich nicht möchte.

naltatis, 2023-06-26:
Ist aktuell noch etwas versteckt. Klick mal auf den aktuellen Netzpreis.

[image]

andig, 2023-09-09:
@naltatis findest Du hier noch einen guten Platz für einen Indikator oder Text?

PR #10297 (2023-10-13): Publish smartCostActive active
@naltatis Anfrage aus https://github.com/evcc-io/evcc/discussions/10293#discussioncomment-7273711. Das haben wir noch nicht, oder?

also fixes #8436
closed 2023-10-13 by andig