evcc-io/evcc #8441: Smart air conditioner

froemer4773, 2023-06-13:
Since you already have all the surplus calculations at hand, couldn't you introduce another type of consumer, the smart air conditioning system? If the car is full, then switch on the air conditioner...

StefanSchoof, 2023-06-14:
You can switch other devices like switches https://docs.evcc.io/docs/devices/chargers#schaltbare-steckdosen or even add your custom charger https://docs.evcc.io/docs/devices/chargers#manuell 

There is also a PR for support of the Stiebel ISG #7207

andig, 2023-06-14:
You can already define custom chargers and assign priorities to chargers. Looks like the feature is already available :) Configuration will get easier when we implement https://github.com/evcc-io/evcc/pull/7338.

closed 2023-06-14 by andig