evcc-io/evcc #8447: Ich habe heute mit dem Nightly Build evcc_0.118.0+1686450921 erneut getestet, ob die Funktion der Ladestromänderung funktioniert. Das Verhalten ist weiterhin so wie bei 0.117.4+xxxx beobachtet:

ilion1, 2023-06-13:
Ich habe heute mit dem Nightly Build evcc_0.118.0+1686450921 erneut getestet, ob die Funktion der Ladestromänderung funktioniert. Das Verhalten ist weiterhin so wie bei 0.117.4+xxxx beobachtet:
1. Wenn kein Auto angeschlossen ist, kann man den maximalen Ladestrom verringern und der Wert bleibt stehen.
2. Sobald ein Fahrzeug lädt (getestet mit KIA e-Niro 3-phasig und Mercedes Hybrid 2-phasig), wird nach wenigen Sekunden auf den Standardwert (16 A) zurück gesetzt. Man kann während des Ladens den Strom ändern aber auch hier erfolgt ein Rücksetzen auf 16A.
3. Der minimale Ladestrom lässt sich auch während des Ladens ändern und bleibt auf dem eingestellten Wert, wie es sein sollte.
Hier der zeitliche Ablauf sowie Konfiguration und Log-File im Trace-Modus als Anhang.

> 12:09:51: Fahrzeug auf Vehicle 1 gestellt
> 12:09:55: Ziel SOC auf 100% gestellt
> 12:10:59: Max Ladestrom manuell auf 14 A
> 12.11:45: Auto angesteckt lädt mit 16A
> 12:12:40: Max Ladestrom manuell auf 12A
> 12:12:46: Reset auf 16A
> 12:13:15: Min 10A (Wert bleibt dauerhaft bestehen)
> 12:14:00: Max Ladestrom manuell auf 12A
> 12:14:06: Reset auf 16A
>

Hier die zugehörigen Einträge in der Log-Datei:
```
[lp-1  ] DEBUG 2023/06/11 12:09:48 set charge mode: now
[lp-1  ] DEBUG 2023/06/11 12:09:51 set min current: 6
[lp-1  ] DEBUG 2023/06/11 12:09:51 set target soc: 70
[lp-1  ] DEBUG 2023/06/11 12:09:55 set target soc: 100
[lp-1  ] DEBUG 2023/06/11 12:10:59 set max current: 14
[lp-1  ] DEBUG 2023/06/11 12:11:47 set max current: 16
[lp-1  ] DEBUG 2023/06/11 12:12:40 set max current: 12
[lp-1  ] DEBUG 2023/06/11 12:12:46 set max current: 16
[lp-1  ] DEBUG 2023/06/11 12:13:15 set min current: 10
[lp-1  ] DEBUG 2023/06/11 12:14:00 set max current: 12
[lp-1  ] DEBUG 2023/06/11 12:14:06 set max current: 16
```
[EVCC-log.zip](https://github.com/evcc-io/evcc/files/11714423/EVCC-log.zip)

_Originally posted by @ilion1 in https://github.com/evcc-io/evcc/discussions/8412_

andig, 2023-06-13:
/cc @DerAndereAndi überschreibt eebus das?

DerAndereAndi, 2023-06-14:
- Um `12:10:59` wird das max gesetzt, bevor ein Fahrzeug angeschlossen wird. Das läuft aber ins leere, da kein Fahrzeug angeschlossen ist und damit die Wallbox auch kein max setzen kann
- Sobald das Fahrzeug angeschlossen ist, wird es sofort laden und zwar mit maximaler in der Wallbox über DIP Schalter maximaler Stromstärke. Das lässt sich nicht verhindern oder ändern
- Wenn dann das Anstecken an evcc übermittelt wird, übermittelt die Wallbox min und max Stromstärken. Diese werden einmalig gesetzt, siehe `12:11:47`
- Um `12:12:40` wird das Limit wieder extern auf `12` gesetzt, aber ich sehe nicht dass das Limit auch zur Wallbox gesendet wird. Ich sehe im Code momentan nicht, wie der Aufruf überhaupt zum Wallbox Code geführt wird und einen Aufruf von `setLimit` im loadpoint führt. Dazu kenne ich den Code nicht.
- Im EEBUS Code werden NUR Limits die am Fahrzeug gesetzt sind berücksichtigt und damit die Werte der Wallbox überschrieben. Ansonsten funktioniert das ganze System nicht mit ISO15118 und IEC61851.

Lösung: Limits nicht am Loadpoint sondern beim identifizierten Fahrzeug setzen.

DerAndereAndi, 2023-06-14:
Alternativer Gedanke: man übernimmt die Werte der Wallbox nur, wenn die Wallbox auch eine `capability: iso15118` hat, und evtl. das Auto auch. Die entsprechenden vehicle templates müssten das dann aber auch bekommen.

Optional sollte man das in den Einstellungen auch ausschalten können bei der Wallbox. In der Regel kann man da die Kommunikation auf IEC61851 einschränken.

DerAndereAndi, 2023-06-15:
Ja, hatte da was durcheinander gebracht. Natürlich wird da nichts geschickt.

Das Grundproblem ist, dass es eben Szenarien gibt bei denen sich die Ladegrenzen ändern. Und zwar wenn die Verbindung von der Wallbox zum Fahrzeug zwischen IEC61851 und ISO15118 wechselt. Die Wallbox weiss wann dies geschieht und auch welche Werte dann verfügbar sind. Die Ladelimits kommen dann bei ISO15118 auch immer vom Auto selbst! Aber diese sind leider auch nicht immer korrekt, der ID.3 liefert z.b. momentan 1,5A, aber erst bei 3A lädt er "richtig". Von daher würde ich gerne generell diese Werte als Anschlusswerte lassen und von der Wallbox auch mitteilen/setzen lassen.

Jetzt könnte man das so umbauen, dass die Änderung immer nur dann gemacht wird, wenn der Modus sich geändert hat und neue Limits zu evcc übermittelt wurden. Momentan wird das bei jedem Aufruf von `Status()` gemacht.

Sauber könnte vielleicht so etwas sein:
- Ladelimits am Loadpoint ist immer das was der Anschluss hergibt, entweder manuell oder von der Wallbox.
- Das gibt die "äußere" Grenze an.
- Am Fahrzeug kann man dann eine "innere" Grenze angeben, die wird an den äußeren Grenzen "abgeschnitten". Und diesen Wert sollte man auch von der API nutzen, wenn das (warum auch immer) zur Laufzeit geändert werden soll.
- Evtl. macht es Sinn hier optional auch eine IEC61851 und eine ISO15118 Grenze angeben zu können, wenn ein Fahrzeug technisch beides kann, aber in beiden Fällen sich unterschiedlich verhällt

Beispiel 1:
- Loadpoint: 6A - 16A
- Fahrzeug: 8A - 16A
- Resultat: 8A - 16A

Beispiel 2:
- Loadpoint 2A  - 16A
- Fahrzeug: 3A - 16A
- Resultat: 3A - 16A

Beispiel 3:
- Loadpoint 6A - 16A
- Fahrzeug: 8A - 20A
- Resultat: 8A - 16A

Beispiel 4:
- IEC61851:
  - Loadpoint: 6A - 16A
  - Fahrzeug: 8A - 16A
  - Resultat: 8A - 16A
- ISO15118:
  - Loadpoint: 1,5A - 16A
  - Fahrzeug: 3A - 16A
  - Resultat: 3A - 16A

Es wäre wichtig, ISO15118 mal zu berücksichtigen ansonsten bleibt das ein Hack der immer wieder zu Problemen führt. Wie hier. Aber das sag ich ja nicht erst seit jetzt ;)

premultiply, 2023-06-15:
Am Ende ist das ähnlich komplex und mehrschichtig wie bei den Phasen. 😜

Viele IEC-Wallboxen liefern zumindest die Werte für die eingestellte oder feste obere eigene Stromgrenze. Manche auch (noch) die des angesteckten Kabels. Die 6A unten sind hier ja bei IEC durch den Standard vorgegeben.

Hier wäre der Mechanismus daher zumindest in der Grundform ähnlich zu ISO möglich.

Richtig, nichts von alledem wird derzeit wirklich berücksichtigt ausser der Nutzerangaben.

andig, 2023-06-15:
> Das Grundproblem ist, dass es eben Szenarien gibt bei denen sich die Ladegrenzen ändern. 

> Am Ende ist das ähnlich komplex und mehrschichtig wie bei den Phasen. 

Yep. Ich hab mich gefragt, ob es nicht wenigstens ein kurzfristiges kleines Plaster geben sollte. Heute schauen wir in `SyncCharger` ob die Statuus noch passen, falls nein steuern wir nach. Für die Ströme, zumindest für den Maximalstrom tun wir das nicht. Dabei gibt es zwei Szenarien: 

1. Iststrom > (neuer) Maximalstrom -> wir müssten reduzieren, und zwar immer
2. Iststrom < (neuer) Maximalstrom -> wir müssten erhöhen, zumindest im Sofortladen und Cheap Modus

Beides sind sicher eher seltene Fälle die nur bei API-Nutzung oder eben bei ISO zuschlagen, wären aber relativ einfach zu machen.

Der Dritte Fall- Minimalstrom und MinPV würde sich über die aktuelle Regelung selbst fixen.

closed 2023-07-12 by github-actions[bot]