evcc-io/evcc #8451: evse-din Template / configure / device name

premultiply, 2023-06-14:
And I just tried `evcc configure` , it takes `"device name"` and places it in `device:` (where `device: /dev/ttyUSB0` should go), and `name:` is always `wallbox1` as a result.

So template is broken it seems. EDIT - not broken but rather confusing. 
It would be better if instead of "`device name`" it would ask for "`device`" AND would suggest-show `/dev/ttyUSB0` by default.

_Originally posted by @RTTTC in https://github.com/evcc-io/evcc/discussions/8437#discussioncomment-6165755_

andig, 2023-06-14:
See defaults.yaml:

```
  - name: modbusdevice
    description:
      de: Gerätename
      en: Device name
    help:
      de: USB-RS485 Gerätename
      en: USB-RS485 device name
```

RTTTC, 2023-06-14:
Funny, for me it seemed that it is asking for a device name , such as "Charger number 2" or "Green charger"
And naturally test failed due to this :)

RTTTC, 2023-06-14:
Device name would be fine if it would show default `/dev/ttyUSB0` in grey, (similar to how it shows `port` `502`)

Goal here would be to reduce/eliminate confusion and support calls from users, as much as possible, especially people that are new to evcc, people with less IT knowledge, amateurs :)
It would be especially beneficial if it would also help with all device templates, where `/dev/ttyUSB0`is used, not just a few people that use din controllers :D

RTTTC, 2023-06-14:
Or maybe just remove `name`?

RTTTC, 2023-06-16:
I think it's good practice to check for errors.

andig, 2023-06-23:
Example has been added.

closed 2023-06-23 by andig