evcc-io/evcc #8454: Tibber Strompreis geht seit Update auf 118 nicht mehr

abeamstart, 2023-06-14:
### Describe the bug

[main  ] ERROR 2023/06/14 12:23:04 failed configuring grid tariff: cannot create tariff 'tibber': 1 error(s) decoding:

* '' has invalid keys: cheap

### Steps to reproduce

Siehe Fehlermeldung.

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
     type: tibber
     cheap: 0.28 # [currency]/kWh
     token: "richtigertoken"
     homeid: "richtigehomeid" # optional
```

### Log details

```text
[main  ] ERROR 2023/06/14 12:23:04 failed configuring grid tariff: cannot create tariff 'tibber': 1 error(s) decoding:

* '' has invalid keys: cheap
```

### What type of operating system are you running?

Linux

### Version

118

VolkerK62, 2023-06-14:
`cheap` gibt es schon länger nicht mehr
https://docs.evcc.io/docs/reference/configuration/tariffs#type-tibber

wurde ersetzt durch `smartCostLimit`
https://docs.evcc.io/docs/reference/configuration/site#smartcostlimit-ehemals-cheap

abeamstart, 2023-06-14:
Super! Vielen Dank für die schnelle Hilfe.
Config hatte noch bis zum Update so funktioniert.
Habe nun unter "site" "smartCostLimit:" ergänzt und bei tibber "cheap" entfernt.

closed 2023-06-14 by abeamstart