evcc-io/evcc #8468: Als Tool möchte ich die Ladung des höher Priorisierten Fahrzeugs Selbständig starten

xerion3800, 2023-06-15:
**Aktuelles Verhalten:**
Ich habe zwei Ladestationen. Bei den Fahrzeugen ( nicht bei loadpoints) habe ich `priority `vergeben.
Morgens fährt Auto mit `priority: 2` weg und Auto `priority: 1` verbleibt an der Ladestation. Irgendwann am Morgen fängt dann das Überschussladen an und regelt entsprechend hoch. Jetzt kommt gegen Mittag das Auto mit `priority: 2` wieder nach Hause und wird angesteckt. Der volle Überschuss (z.B. 10 kW) geht dann weiter in Auto mit `priority: 1` und Auto mit `priority: 2` fängt nicht an zu laden. Ich muss dann erst  die Ladung von dem Auto mit `priority: 1` kurz unterbrechen damit das Fahrzeug 2 das Laden beginnt, danach schalte ich das erste Fahrzeug wieder zu damit der Überschuss anhand der Priorität verteilt wird.

**Erwartetes Verhalten:**
Wenn ein Fahrzeug schon am Laden ist und genügend Überschuss für zwei Fahrzeuge vorhanden ist, soll das nachträglich angesteckte Fahrzeug mit `minCurrent` anfangen zu laden. Nach dem der Lade Vorgang gestartet ist, soll dann die Regelung entsprechend der vergebene Priorität wieder übernehmen.

xerion3800, 2023-06-15:
Lieder ist das gestrige Logfile schon überschriebe worden. Ich werde es wieder nachstellen sobald die Autos wieder gefahren wurden.

xerion3800, 2023-06-15:
Obwohl in Slack es schon diskutiert wurde und dort hieß, dass das Verhalten aktuell normal ist.
Aus der Doku würde ich das auch so interpretieren:
_priority
Dient während der Ladung der Priorisierung von Loadpoints untereinander. Hat keinen Einfluss auf den Start der Ladung._

andig, 2023-06-15:
Davon sehe ich im PR nichts: https://github.com/evcc-io/evcc/pull/6107

> Hat keinen Einfluss auf den Start der Ladung.

Ich denke was hier gemeint war ist, dass jeder LP individuell über den Start entscheidet ohne dabei auf andere zu schauen. Unabhängig davon muss eine LP mit höherer Prio das Leistungsbudget des LPs mit niedrigerer Prio konsumieren.

/cc @VolkerK62

VolkerK62, 2023-06-15:
>Unabhängig davon muss eine LP mit höherer Prio das Leistungsbudget des LPs mit niedrigerer Prio konsumieren und starten

Das Thema kam kurz nach der Implementierung schon mal auf. Da hieß es: Priority greift nur, wenn geladen wird und führt nicht dazu, dass der LP mit der höheren Prio eingeschaltet wird. Deshalb ist dieser Hinweis in der Doku ergänzt worden https://github.com/evcc-io/docs/pull/341
Wäre prima, wenn es anders wäre ;-)

xerion3800, 2023-06-22:
Heute konnte ich mal wieder testen. Es entsprach zwar nicht ganz dem dem Aufbau wie letztes Mal, da wetterbedingt nicht zu viel Überschuss zur Verfügung steht. Aber das Starten des zweiten (später angeschlossenen ) EV hat diesmal auf jeden Fall funktioniert. 
Unterschied zum letzten Mal war nur dass das bereits ladende EV schon auf 3-Phasen umgeschaltet hatte, heute war nur 1-Phasig möglich, sofern es überhaupt mit der Logik zu tun haben kann. Ich beobachte mal weiter und sofern sich was ergibt stelle ich dann ein Log zur Verfügung.

xerion3800, 2023-07-05:
Heute habe ich das Problem wieder. LP2 lädt mit den verfügbaren Überschuss und LP1 wird angesteckt. Die Ladung von LP1 startet aber leider nicht.
Im Anhang das Log dazu was hoffentlich etwas Licht ins Dunkle bringt.
Im Log sieht man das Verbinden von LP1 um 2023-07-05T10:51

[_evcc-io_logs.txt](https://github.com/evcc-io/evcc/files/11957265/_evcc-io_logs.txt)

VolkerK62, 2023-07-05:
```
2023-07-05T05:15:27.105620490Z [site  ] WARN 2023/07/05 07:15:27 vehicle 'Hyundai IONIQ 5' does not support automatic detection
2023-07-05T05:15:27.105674187Z [site  ] WARN 2023/07/05 07:15:27 vehicle 'Hyundai IONIQ 5 (Schnell-Laden)' does not support automatic detection
2023-07-05T05:15:27.106767497Z [site  ] WARN 2023/07/05 07:15:27 vehicle 'Sungrow SBR 128 (offline)' does not support automatic detection
2023-07-05T05:15:27.107514414Z [site  ] WARN 2023/07/05 07:15:27 vehicle 'Cupra Formentor' does not support automatic detection
2023-07-05T05:15:27.108299323Z [site  ] WARN 2023/07/05 07:15:27 vehicle 'Cupra Formentor (Schnell-Laden)' does not support automatic detection
2023-07-05T05:15:27.108341319Z [site  ] WARN 2023/07/05 07:15:27 vehicle 'VELO DE VILLE' does not support automatic detection
```

Du hast die `priority` bei den Fahrzeugen hinterlegt. Wenn diese nicht erkannt werden, kann `priority` nicht wirken.

Im Log ist zwar zu erkennen, dass ein Auto verbunden wird, aber nicht welches. Obwohl Soc und Odo übermittelt werden.
```
2023-07-05T10:51:17.818249305Z [lp-1  ] INFO 2023/07/05 12:51:17 car connected
2023-07-05T10:51:17.830309740Z [lp-1  ] DEBUG 2023/07/05 12:51:17 vehicle soc: 72%
2023-07-05T10:51:17.832478253Z [lp-1  ] DEBUG 2023/07/05 12:51:17 vehicle range: 32km
```

xerion3800, 2023-07-05:
> ```
> 2023-07-05T05:15:27.105620490Z [site  ] WARN 2023/07/05 07:15:27 vehicle 'Hyundai IONIQ 5' does not support automatic detection
> 2023-07-05T05:15:27.105674187Z [site  ] WARN 2023/07/05 07:15:27 [...]

Hallo Volker, danke für deine Auswertung. Das klingt auf jeden Fall plausibel, war mir nicht bewusst das priority an die Detection geknüpft ist.
Da es ja immer wieder Probleme mit den Herstellern "APIs" gibt habe ich mich vor langer Zeit davon verabschiedet. Ist ja noch gar nicht so lange her das Hyundai wieder was geändert hatte. ;-) Das hat mich immer genervt, darum habe ich mir selber was mit OBD, WIFI und Python für Zuhause gebaut. Das klappt alles super und die Daten kommen per MQTT. Was kann ich denn jetzt noch tun, damit das Fahrzeug automatisch "erkannt" wird?

VolkerK62, 2023-07-05:
`priority` ist Bestandteil von `onIdentify` 
Zeig doch mal bitte den `vehicles` Teil deiner Konfig.

VolkerK62, 2023-07-05:
editiere das mal bitte, damit es richtig formatiert ist
code komplett markieren und dann im Menüband `<>` klicken

andig, 2023-07-05:
@xerion3800 priority liesse sich natürlich auch statisch an die Ladepunkte hängen- ich vermute, das hattest du im Sinne?

VolkerK62, 2023-07-05:
unabhängig von andig´s Hinweis ...
zur Erkennung fehlt der Status. Ich vermute schwer, den kriegst du per OBD nicht, oder?

und, losgelöst von dem Thema: beim Ioniq muss `phases: 3` rein.
`phases: 0` funktioniert nur am Loadpoint, wenn die Wallbox das unterstützt.

xerion3800, 2023-07-05:
> @xerion3800 priority liesse sich natürlich auch statisch an die Ladepunkte hängen- ich vermute, das hattest du im Sinne?

An die Ladepunkte eher nicht. Ich erkläre mal kurz den Aufbau meiner Umgebung, damit ihr das evtl. besser nachvollziehen könnt.

Ich habe zwei Ladepunkte die technisch identisch sind. Also  beide können 11 kW und Phasenumschaltung. Dann habe ich zwei Fahrzeuge. Ein PHEV und BEV.  Der PHEV kann nur einphasig laden der BEV dreiphasig. Als ich anfangs `phases:` bei den Ladepunkten definiert hatte, bekam ich bei dem PHEV immer das Problem, das EVCC den LP auf drei Phasen schalten wollte, das bedeutete dann immer wieder Ladeunterbrechungen. Damit das nicht mehr passiert hatte ich dann `phases` dann nach `vehicles ` verschoben. Seitdem ist das Problem behoben. Ich hätte natürlich auch beim LP einfach auf `phases: 1` stellen können, aber genau das möchte ich nicht, da es mal vorkommt das die Autos untereinander getauscht werden und dann soll natürlich das BEV auch an dem LP dreiphasig Überschuss laden können.

Was ich aber noch nicht ganz verstanden habe, warum `priority` mit `onIdentify `gekoppelt ist. Wenn ich die Fahrzeuge innerhalb der LP tausche dann kann ich das schnell in der UI unter Fahrzeuge anpassen. Warum muss sich das Fahrzeug von alleine identifizieren damit `priority` greift? Meine Erwartung wäre, wenn ich schon `priority` je Fahrzeug definieren kann und das Fahrzeug z.B. durch `onDisconnect` wieder ausgewählt wurde, das es dann doch egal sein sollte ob es manuell durch UI oder `onDisconnect` erfolgt ist. Wichtig wäre doch nur das ein Fahrzeug richtig zugweisen ist und egal durch welche Aktion. Oder übersehe ich da einen UseCase das es automatisch erfolgen muss?

andig, 2023-07-05:
> Damit das nicht mehr passiert hatte ich dann phases dann nach vehicles  verschoben. Seitdem ist das Problem behoben. Ich hätte natürlich auch beim LP einfach auf phases: 1 stellen können

Der LP ist 0p oder 3p, das Auto 1p.

> Was ich aber noch nicht ganz verstanden habe, warum priority mit onIdentify gekoppelt ist.

Weil die Prio am Fahrzeug hängt.

> und das Fahrzeug z.B. durch onDisconnect wieder ausgewählt wurde

Disconnect wählt kein Fahrzeug- es sei denn zurück auf das Defaultfahrzeug des Ladepunktes falls du das hast.

xerion3800, 2023-07-05:
> Der LP ist 0p oder 3p, das Auto 1p.

Die beiden LP sind 0p da beide Phasenumschaltung unterstützen.

> Disconnect wählt kein Fahrzeug- es sei denn zurück auf das Defaultfahrzeug des Ladepunktes falls du das hast.

Ja ich habe bei beiden LP Default Fahrzeuge definiert, darum hätte ich erwarte das diese Zuweisung für priority ausreicht?

xerion3800, 2023-07-05:
Hier noch mal die LP Config wo ich die Default Fahrzeug definiert habe, das müsste doch passen, oder?
```
loadpoints:
  - title: Parkplatz
    charger: goe_charger1
    vehicle: cupra
    mode: pv # Standardmodus zum Laden
    phases: 0 # Automatische Umschaltung zwischen 1 und 3 Phasen
    enable:
      threshold: 0 # 0 = Export die minimale Ladeleistung erreicht haben
      delay: 1m
    disable:
      threshold: 200 # Maximaler Netzbezug von 200W ist erlaubt
      delay: 6m
    minCurrent: 6 # Mindestens 6A
    maxCurrent: 16 # Maximal 16A
    resetOnDisconnect: true

  - title: Carport
    charger: goe_charger2
    vehicle: IONIQ5
    mode: pv # Standardmodus zum Laden
    phases: 0
    enable:
      threshold: 0 # 0 = Export die minimale Ladeleistung erreicht haben
      delay: 1m
    disable:
      threshold: 200 # Maximaler Netzbezug von 200W ist erlaubt
      delay: 6m
    minCurrent: 6 # Mindestens 6A
    maxCurrent: 16 # Maximal 16A
    resetOnDisconnect: true

  - title: Tasmota Steckdose
    charger: tasmota_plug
    vehicle: e_bike
    mode: pv # Standardmodus zum Laden
    phases: 1
    enable:
      threshold: 0 # 0 = Export die minimale Ladeleistung erreicht haben
      delay: 1m
    disable:
      threshold: 200 # Maximaler Netzbezug von 200W ist erlaubt
      delay: 1m
    minCurrent: 4 # Mindestens 4A
    maxCurrent: 4 # Maximal 4A

  - title: Sungrow SH10RT
    charger: SH10RT
    meter: SBR128_meter
    vehicle: SBR128
    mode: pv # Standardmodus zum Laden
    phases: 3
    enable:
      threshold: 0 # 0 = Export die minimale Ladeleistung erreicht haben
      delay: 1m
    disable:
      threshold: 200 # Maximaler Netzbezug von 200W ist erlaubt
      delay: 3m
    minCurrent: 6 # Mindestens 6A
    maxCurrent: 6 # Maximal 6A
    # resetOnDisconnect: true
    soc:
      poll:
        mode: charging
        interval: 5m
      estimate: true

```

xerion3800, 2023-07-05:
> aber ja, das sollte so sein. Siehst Du im Log ob das Defaultfahrzeug aktiviert wird?

Das müsste ich morgen mal prüfen. Aber in der UI sehe ich ja das die Zuweisung der Fahrzeuge passt, das sollte dann doch auch aussagekräftig sein, oder nicht?

xerion3800, 2023-07-06:
Wenn das Fahrzeug angesteckt wird und schon in der UI ausgewählt ist, kommt "nur" ein `car connected`:
```
[lp-2  ] INFO 2023/07/06 07:59:27 car connected
[lp-2  ] DEBUG 2023/07/06 07:59:27 pv timer elapse
[lp-2  ] DEBUG 2023/07/06 07:59:27 pv timer inactive
[lp-2  ] DEBUG 2023/07/06 07:59:27 charger: guard elapse
[lp-2  ] DEBUG 2023/07/06 07:59:27 guard timer inactive
[lp-2  ] DEBUG 2023/07/06 07:59:27 charge total import: 389.869kWh
[lp-2  ] DEBUG 2023/07/06 07:59:27 vehicle soc: 96%
[telegram] DEBUG 2023/07/06 07:59:27 sending to ***
[lp-2  ] DEBUG 2023/07/06 07:59:27 vehicle range: 389km
[lp-2  ] DEBUG 2023/07/06 07:59:27 targetSoc reached: 96.0% > 80%
```

wenn ich aber vor dem Abstecken das Fahrzeug z.B, auf Gastfahrzeug stelle dann wird das Fahrzeug durch `resetOnDisconnect `auch wieder "verbunden".

```
[lp-2  ] DEBUG 2023/07/06 08:04:07 charger status: A
[lp-2  ] INFO 2023/07/06 08:04:07 car disconnected
[lp-2  ] INFO 2023/07/06 08:04:07 vehicle updated: unknown -> Hyundai IONIQ 5
[lp-2  ] DEBUG 2023/07/06 08:04:07 set charge mode: pv
[lp-2  ] DEBUG 2023/07/06 08:04:07 set target soc: 80
[lp-2  ] DEBUG 2023/07/06 08:04:07 set charge mode: pv
[lp-2  ] DEBUG 2023/07/06 08:04:07 set min current: 6
```

VolkerK62, 2023-07-06:
... und ich würde mal vermuten, dass dann auch `priority` richtig funktioniert.
Denn jetzt kommen auch die anderen Parameter `mode` und  targetsoc`.

xerion3800, 2023-07-06:
> ... und ich würde mal vermuten, dass dann auch `priority` richtig funktioniert.
> Denn jetzt kommen auch die anderen Parameter `mode` und  `targetsoc`.

Okay aber das wäre dann ja ein Problem das ich vor dem abstecken erst ein anderes Fahrzeug auswählen muss. Also wenn das Fahrzeug schon zugewiesen ist und nur angesteckt und wieder angesteckt wird sollte sich an der `priority` Logik dich nichts ändern, oder?

VolkerK62, 2023-07-06:
Keine Ahnung, wie das im code hinterlegt ist.
Alternative bis zur Klärung: Standardfahrzeuge am Loadpoint entfernen und nach dem anstecken im UI per Hand auswählen.

VolkerK62, 2023-07-06:
Noch ne Idee. Hinterleg priority auch beim loadpint, analog zu den Fahrzeugen. Wenn die dann Mal anders angesteckt werden, musst du das Auto per Hand im UI zuweisen.

xerion3800, 2023-07-06:
> Noch ne Idee. Hinterleg priority auch beim loadpint, analog zu den Fahrzeugen. Wenn die dann Mal anders angesteckt werden, musst du das Auto per Hand im UI zuweisen.

Danke für deine Unterstützung kann ich bei Gelegenheit auch genre teste, aber für mich sieht es nach einer Definitionslücke aus und da würde ich gerne @andig seine Meinung zu wissen bevor wir weitere Workarounds suchen.

VolkerK62, 2023-07-06:
ist evtl ein Selbstgemachter.
Setz mal an den Loadpoints `resetOnDisconnect: false`.

andig, 2023-07-06:
> für mich sieht es nach einer Definitionslücke aus

Was habt ihr da jetzt als Lücke konkret identifiziert? Dann kann ich mal versuchen das nachzustellen.

> Setz mal an den Loadpoints resetOnDisconnect: false.

Das sollte nix ändern. Erst kommen die Defaults, dann die vom Fahrzeug:

	// set defaults
	if lp.ResetOnDisconnect {
		lp.applyAction(lp.onDisconnect)
	}

	// override global defaults with default vehicle
	if lp.defaultVehicle != nil {
		lp.applyAction(lp.defaultVehicle.OnIdentified())
	}

xerion3800, 2023-07-07:
> Was habt ihr da jetzt als Lücke konkret identifiziert? Dann kann ich mal versuchen das nachzustellen.

Aktuell sieht es so aus als wenn ein Fahrzeug nur erkannt wenn es den Status **selber** übermittelt oder wenn das Fahrzeug vor dem abstecken in der UI verändert wurde.  Wenn also ein Fahrzeug schon "identifiziert" wurde und das sollte m.E. egal sein ob ich das als User in der UI, es durch `ResetOnDisconnect ` oder automatisch erfolgt ist.

andig, 2023-07-07:
Steps to reproduce?

xerion3800, 2023-07-07:
- Die Autos werden per Default durch `ResetOnDisconnect` "automatisch" zugewiesen.
- Fahrzeug wird abgesteckt und wird anschließend wieder angesteckt.
-  Im Log taucht dann nur das auf: `car connected` und das soll laut @VolkerK62 die Ursache sein. Hier hätte dann der Fahrzeugname stehen müssen, wenn ich es richtig verstanden habe.
Auszüge der config bzgl. LP und Vehicles sind weiter oben im Issue zu finden. Vehicle werden per MQTT versorgt.

andig, 2023-07-09:
Ich denke ich hab den Fehler gefunden. Sieht aus als wäre https://github.com/evcc-io/evcc/pull/6417 nur eine halbe Implementierung gewesen.

andig, 2023-07-09:
@xerion3800 wärst Du in der Lage, den PR selbst zu compilieren und zu testen? 
@VolkerK62 wäre Klasse wenn Du nochmal querchecken könntest ob das Offlinetemplate noch wie erwartet funktioniert und konfigurierbar ist. Das hab ich drastisch vereinfacht.

VolkerK62, 2023-07-09:
testen kann ich leider nix. 
Das ist schon schwer kastriert. Für meine Zwecke ok.
Aber da wird es bestimmt Widerspruch geben ;)  Sollte (wenn möglich) so konfigurierbar sein, wie ein normales Fahrzeug, nur das der Soc fehlt.

Was steckt hinter   `{{ include "vehicle-identify" . }}` ?

andig, 2023-07-09:
Das ist auch so. Steckt hinter dem `preset` (=eine Sammlung von Werten, siehe `defaults.yaml`) das dann mit der `include` Zeile geredendert wird.

xerion3800, 2023-07-09:
Ja kompilieren und testen könnte ich. Kann aber nicht versprechen ob ich es kurzfristig schaffe, da wir kurz vor dem Urlaub sind.

xerion3800, 2023-07-09:
@andig kann ich das "nur" testen wenn ich das Fahrzeug mit Prio1 dran habe und am Laden ist und dann Prio 2 später hinzukommt ? Oder kann ich das auch anders testen durch an- und abstecken und man sieht was im Log?  Wenn es "einfacher" geht dann könnte ich das ja schon mal versuchen?

andig, 2023-07-10:
Bin nicht sicher ob ich dich verstehe. Du musst halt dein Szenario testen das bisher nicht funktioniert hat. Sollte jetzt gehen. Falls nein gibts zusätzliches Logging zur Fehleranalyse.

xerion3800, 2023-07-10:
Okay dann muss ich das Verhalten mal wieder nachstellen. Das wird aber wohl Dauern... wenig Sonne, Autos so gut wie voll und ab Ende der Woche im Urlaub :-)
Ich dachte das es daran lag, das die Autos nicht erkannt wurden. So hatte ich @VolkerK62 zumindest verstanden . Im Log stand dann nur `car connected`. Wenn ich dich aber richtig verstanden habe und den PR richtige deute, wurde "nur" was an der Priority geändert und somit wurde ich im Log immer noch `car connected` bekommen und nicht daran festmachen, dass es jetzt funktioniert?

andig, 2023-07-10:
Die Prio des erkannten Fahrzeuges wurde einfach nicht übernommen. Lass es uns vereinfachen- ich mache den PR fertig und wenn das Problem in Zukunft wieder auftaucht meldest Du Dich einfach nochmal :)

closed 2023-07-10 by andig