evcc-io/evcc #8478: sponsorship unavailable! / telemetry: upload failed: unexpected status: 403 (Forbidden)

TobiasHuber1980, 2023-06-15:
### Describe the bug

`sponsorship unavailable! `
`telemetry: upload failed: unexpected status: 403 (Forbidden)`
`Network Error: API request failed https://api.evcc.io/v1/total`

### Steps to reproduce

Fehler tritt in der aktuellen nightly auf  0.118.0 (1ec690c6)

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.118.0 (1ec690c6)

StefanSchoof, 2023-06-15:
Klingt nach dem Problem aus #8117

Es fehlt meines Wissens hier ein Log mit entsprechend Log Level damit hier was von dem Team gemacht werden kann.

TobiasHuber1980, 2023-06-15:
Ja, das passt zu #8117 
Ich mache dann hier dicht.

closed 2023-06-15 by TobiasHuber1980