evcc-io/evcc #8482: Ansicht Ladevorgänge kleiner Anzeigefehler beim Download Button

StevieC121176, 2023-06-15:
### Describe the bug

In der mobile Version (Portrait Modus) der neuen Ansicht der Ladevorgänge wird der Download Button für die Gesamtübersicht der Ladevorgänge teilweise abgeschnitten (IOS) sh. Screenshot
Die Werte für CO2, Preis, Sonnenanteil usw. können seitlich geschellt werden, die Download Buttons bleiben an der Stelle.
Evtl. Wäre eine Anordnung der beiden Buttons untereinander in dieser mobilen Variante besser?

<details>
 <summary>Screenshots</summary>

[image]

</details>

### Steps to reproduce

.

### Configuration details

```yaml
nicht relevant
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.118.0 (1ec690c6)

naltatis, 2023-06-16:
Danke fürs melden. Die Ansicht ist mir scheinbar durchgerutscht.

StevieC121176, 2023-06-17:
Gefällt mir übrigens sehr gut mit der monatlichen Ansicht und den Filtern. Da bin ich echt schon auf die graphische Auswertung gespannt.

naltatis, 2023-06-17:
Danke! Um die Erwartungen etwas zu bremsen. Auf unserer Prioliste kommt erst das Thema Config-UI. Danach dann die grafische Darstellung hier 😀

PR #8504 (2023-06-17): Improve CSV Download Buttons Layout for Small Screens
fixes #8482

large screen
[image]

small screen
[image]
closed 2023-06-17 by naltatis