evcc-io/evcc #8488: Tibber Preise fehlen seit Update 0.118.0

MikeRow1972, 2023-06-16:
### Describe the bug

Seit dem Update auf 0.118.0 werden keine Tibber Preise mehr angezeigt. Dadurch auch keine Berechnungen der Ladevorgänge. Fehlermeldungen werden keine ausgegeben

### Steps to reproduce

1. update zu  0.118.0
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  fiat: trace

# unique installation id
plant: XXXXX

interval: 15s # control cycle interval

sponsortoken: XXXXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true
# mqtt message broker
mqtt:
  broker: 192.168.200.135:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  # user:
  # password:

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    cheap: 0.2 # [currency]/kWh
    token: "XXXXX"
    homeid: "XXXXX" # optional
#   type: fix    
#   price: 0.238 # [currency]/kWh
#	type: tibber
#	cheap: 0.2 # [currency]/kWh
#	token: "XXXXX"
#	homeid: "XXXXX" # optional

  feedin:
    type: fixed
    price: 0.0673 # [currency]/kWh

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.200.113  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.200.11  
  password: XXXXX
  name: pv2
- type: template
  template: sma-inverter 
  usage: battery  
  host: 192.168.200.122  
  password: XXXXX
  name: battery3

chargers:
- type: template
  template: abb 
  id: 1  
  host: 192.168.200.89
  port: 502  
  modbus: rs485tcpip  
  name: wallbox5

vehicles:
- type: template
  template: fiat 
  title: Fiat 500e  
  user: XXXXX@outlook.de  
  password: XXXXX 
  vin: XXXXX  
  capacity: 44  
  pin: XXXXX  
  name: fiat
  targetSoc: 80 # Bis zu welchem Ladestand (Soc) soll das Fahrzeug geladen werden (Optional)
- #type: template
  #name: my_car
  #title: Grandland
  #template: opel
  #user: XXXXX
  #password: XXXXX
  #vin: XXXXX
  #capacity: 12
  type: template
  template: flobz
  name: grandland
  title: Grandland
  url: http://192.168.200.164:5000
  vin: XXXXX # Erforderlich
  capacity: 12 # Akkukapazität in kWh (Optional) 
loadpoints:
- title: ABB
  charger: wallbox5
  mode: pv # charge mode (off, now, minpv, pv)
  phases: 1
  mincurrent: 6
  maxcurrent: 15.659
  resetOnDisconnect: true
  enable:
    delay: 1m
    threshold: 100 #Alter Wert -300
  disable:
    delay: 3m
    threshold: 200 # Ein maximaler Netzbezug von 200W ist erlaubt Alter Wert 0
  guardDuration: 5m

site:
  title: Home
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  residualPower: 200 # Haus Grundlast Alter Wert 100
  bufferSoc: 88 # Hausbatterie wird oberhalb Soc 90% als Puffer genutzt
  prioritySoc: 80 # Hausbatterie bekommt bis zum Soc 65% Prio beim laden
hems:
  type: sma
  AllowControl: false
```

### Log details

```text
Jun 16 06:48:28 EVCC2 evcc[663]: [main  ] ERROR 2023/06/16 06:48:28 failed configuring grid tariff: cannot create tariff 'tibber': 1 error(s) decoding:
Jun 16 06:48:28 EVCC2 evcc[663]: * '' has invalid keys: cheap
```

### What type of operating system are you running?

Linux

### Version

0.118.0

Sillium007, 2023-06-16:
Soweit ich es spontan sehe gibt es 'cheap' nicht mehr:
https://github.com/evcc-io/evcc/commit/7c3ca53f0274df479f5253e4f77eb8d40516f768#diff-63a44d6b909e276ff54ac2a62845fdb2505509d26072160b16cfb37f3cb6de4eL38

VolkerK62, 2023-06-16:
Die Fehlermeldung sagt es:` * '' has invalid keys: cheap`
https://docs.evcc.io/docs/reference/configuration/tariffs#type-tibber
https://docs.evcc.io/docs/reference/configuration/site#smartcostlimit-ehemals-cheap

MikeRow1972, 2023-06-16:
Ok, danke passt wieder. Cheap ist nicht nur zu smartCostLimit geworden sondern auch umgezogen in site. In den aktuellen Release Notes finde ich zu der Änderung nix, vielleicht bin ich auch Blind. Danke für die Tipps

closed 2023-06-16 by MikeRow1972

VolkerK62, 2023-06-16:
ist schon etwas länger her (Ende März), kam mit 0.115.0

MikeRow1972, 2023-06-16:
> ist schon etwas länger her (Ende März), kam mit 0.115.0

Ich bin am 12.06. von 0.117.0 auf 0.118.0 gegangen. Bei bei 0.117.0 ging es noch. Seltsam