evcc-io/evcc #8503: Fehlerhafte Berechnung der Ladeenergie

xrthommy, 2023-06-16:
### Describe the bug

Die Angezeigte Wert der Energie bei den Ladevorgängen weicht manchmal vom tatsächlichen Wert ab. Beispiel:
[image]
Die Zählerstände sind korrekt aber der "Geladen" Wert stimmt nicht. Habe den Verdacht, dass es nur auftritt wenn während des Ladevorganges eine Phasenumschaltung stattfindet.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true
plant: xxx

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  my_car_bmw_i3: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
#  - port: 5200
#    uri: homeassistant:502
#    rtu: true
#    readonly: true
[... 431 lines cut ...]
[site  ] DEBUG 2023/06/16 22:13:48 ----
[lp-1  ] DEBUG 2023/06/16 22:13:48 charge power: 0W
[lp-2  ] DEBUG 2023/06/16 22:13:49 charge power: 0W
[site  ] DEBUG 2023/06/16 22:13:49 dynamic grid tariff
[site  ] DEBUG 2023/06/16 22:13:49 pv power: 0W
[site  ] DEBUG 2023/06/16 22:13:49 battery soc: 70%
[site  ] DEBUG 2023/06/16 22:13:49 battery power: 756W
[site  ] DEBUG 2023/06/16 22:13:49 grid power: 11W
[site  ] DEBUG 2023/06/16 22:13:49 site power: 967W
[lp-1  ] DEBUG 2023/06/16 22:13:49 vehicle odometer: 77941km
[lp-1  ] DEBUG 2023/06/16 22:13:49 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/16 22:13:49 charge meter total import: 861.681kWh
[lp-1  ] DEBUG 2023/06/16 22:13:49 charger status: B
[lp-1  ] DEBUG 2023/06/16 22:13:49 available power -967W < 4140W min 3p threshold
[lp-1  ] DEBUG 2023/06/16 22:13:49 phase scale1p in 2m59s
[lp-1  ] DEBUG 2023/06/16 22:13:49 pv charge current: 0A = 0A + -1.4A (967W @ 3p)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.0

premultiply, 2023-06-17:
@andig Theoretisch wäre das denkbar falls die Box nach einer Umschaltung eine neue Session beginnt und wir dafür auch ChargedEnergy auswerten.

Habe den Eindruck, dass wir besser grundsätzlich ausschließlich TotalEnergy implementieren bzw. verarbeiten wenn das gibt und dann ChargedEnergy weglassen/ignorieren. ChargedEnergy nur wenn die Box sonst nix bietet.
Dürfte zuverlässiger sein.

andig, 2023-06-17:
Können wir gerne machen.

andig, 2023-06-17:
Von welchem Charger reden wir überhaupt? Ist nur die *Anzeige* in evcc falsch oder kommt der falsche Wert vom Charger? Ich mache mal zu bis es mehr Klarheit des Problems gibt.

closed 2023-06-17 by andig

xrthommy, 2023-06-17:
Der Charger ist ein go-e Charger Gemini flex (11kW). Hier der Eintrag für diese Ladung in der Log-Datei des Chargers. Da stimmt es, nur eben nicht in der Ladehistorie von evcc.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/xrtho/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/xrtho/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Session Number | Start | Ende | Dauer gesamt | Dauer aktiver Stromfluss | max. Leistung [kW] | max. Strom [A] | Energie [kWh] | Zaehlerstand Anfang [kWh] | Zaehlerstand Ende [kWh]
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
129 | 16.06.2023 04:21 | 16.06.2023 12:35 | 08:14:00 | 03:17:38 | 7,613 | 16 | 14,904 | 845,557 | 860,461

</body>

</html>

andig, 2023-06-18:
Ok, es geht also um den Go-E. Wir bräuchten bitte ein Log (`go-e: trace`) um zu verstehen, ob wir falsche Werte bekommen, vielleicht zum falschen Zeitpunkt fragen oder wo sonst die Ursache liegt. Wir holen die Daten ja schon vom Go-E API, es ist also unklar warum sie nicht stimmen :(

phrozen77, 2023-06-21:
Siehe auch #8017 - irgendwas ist da faul, auch heute wieder so einen Fall gehabt, Trace-Logs leider nicht vorhanden.

xrthommy, 2023-06-30:
Hallo andig,

 

hier ein Trace:

 

[go-e  ] TRACE 2023/06/30 20:00:52 GET http://xxx.xxx.xxx.xxx/api/status?filter=alw,car,eto,nrg,wh,trx,cards

[go-e  ] TRACE 2023/06/30 20:00:52 {"alw":false,"eto":952005,"car":4,"wh":7885.926,"trx":null,"nrg":[232.5,232.81,232.81,2.79,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}

 

Freundlichen Gruß

 

Thomas

andig, 2023-06-30:
Damit ist ohne Kontext des Ladevorgangs und konkrete Fehlerbeschreibung leider nix anzufangen.