evcc-io/evcc #8508: Index out of range

puzich, 2023-06-17:
### Describe the bug

EVCC hatte keine aktuellen Werte des Grids mehr. Fehler im Log:

Jun 17 11:32:33 evcc evcc[15726]: [site  ] ERROR 2023/06/17 11:32:33 grid energy: panic: runtime error: index out of range [3] with length 2

### Steps to reproduce

Keine Möglichkeit der Reproduzierung

### Configuration details

```yaml
<details><summary>Konfiguration (/root/evcc.yaml)</summary>

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****
sponsortoken: *****
interval: 10s # control cycle interval

chargers:
- name: wallbox1
  type: template
  template: easee
  user: *****
  password: *****
  charger: EHPW8R46
- name: wallbox2
  type: template
  template: easee
  user: *****
  password: *****
  charger: EHLC2AVR

vehicles:
- type: template
  template: bmw
  title: BMW i3
  user: *****
  password: *****
  vin: *****
 # phases: 3
 # icon: car
 # cache: 15m
  mode: pv
  minSoc: 80
  targetSoc: 100
  priority: 1
#  minCurrent: 6
#  maxCurrent: 16
  name: bmw_i3
- type: template
  template: tesla
  title: Tesla ECTO-1
  accessToken: *****
  refreshToken: *****
 # phases: 3
 # icon: car
  mode: pv
  minSoc: 50
  targetSoc: 85
 # minCurrent: 6
 # maxCurrent: 16
  priority: 2
  name: tesla_ecto1

meters:
  - name: my_grid
    type: template
    template: sofarsolar-g3
    usage: grid
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter fuer den Adapter
    delay: 1s # Optional
  - name: my_pv
    type: template
    template: sofarsolar-g3
    usage: pv
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter fuer den Adapter
    delay: 1s # Optional
  - name: my_battery
    type: template
    template: sofarsolar-g3
    usage: battery
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter fuer den Adapter
    delay: 1s # Optional

site:
  title: Home
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery
  prioritySoc: 50
  residualPower: 10
  bufferSoc: 80

loadpoints:
  - title: Carport links # display name for UI
    charger: wallbox1 # Wallbox Referenz
    vehicle: bmw_i3 # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)
  - title: Carport rechts # display name for UI
    charger: wallbox2 # Wallbox Referenz
    vehicle: tesla_ecto1 # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.28 # [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

</details>
Version: `0.118.0 (57232cbb)`
```

### Log details

```text
Jun 17 11:32:33 evcc evcc[15726]: [site  ] ERROR 2023/06/17 11:32:33 grid energy: panic: runtime error: index out of range [3] with length 2
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.0 (57232cbb)

andig, 2023-06-17:
Falsche Modbus Antwort vom Zähler. Das dürfte eigentlich nicht passieren, wird von der verwendeten Bibliothek aber nicht abgefangen.

puzich, 2023-06-17:
Ja. Aber der Fehler kam ganz häufig im Log über einen längeren Zeitraum. Es scheint für mich so, dass wenn einmal die Modbus Abfrage falsch lief, dies sich auch bei den anderen Abfragen nicht mehr fängt. Ich musste den EVCC Daemon einmal neu starten.

andig, 2023-06-17:
Mag sein. Ohne trace log nicht zu bewerten.

puzich, 2023-06-17:
Shit. Habe ich natürlich nicht mitlaufen lassen. Ahnte ich ja nicht dass ich den mal für etwas nicht reproduzierbares brauche. Einen Tipp für mich für die Zukunft?

andig, 2023-06-18:
Das sollte `modbus: trace` in den Loglevel sein.

andig, 2023-07-05:
Fixed im nightly ab morgen danke https://github.com/evcc-io/modbus :)

closed 2023-07-05 by andig