evcc-io/evcc #8510: Zahl und Prozentsymbol nach DIN 5008 formatieren

naltatis, 2023-06-17:
In der Vergangenheit ist mehrfach diskutiert worden [1](https://github.com/evcc-io/evcc/pull/6564), [2](https://github.com/evcc-io/evcc/pull/8480#discussion_r1233007182) wie die Kombination aus Zahl und Prozentzeichen dargestellt werden soll (88% vs 88 %). Aktuell verwenden wir konsistent die Schreibweise ohne Leerzeichen.

Der [Duden empfiehlt](https://de.wikipedia.org/wiki/Prozentzeichen#Schreibweisen) hier eine Schreibweise mit [schmalem geschützten Leerzeichen](https://de.wikipedia.org/wiki/Geschütztes_Leerzeichen#Schmales_geschütztes_Leerzeichen). Wir halten es für sinnvoll auf diese Schreibweise umzustellen. Allerdings sind dafür Anpassungen an einigen Stellen in der Software notwendig um konsistent zu bleiben.

- Anpassungen in allen Vue Templates über eine zentrale formatter Funktion (Loadpoint, Vehicle, Energyflow, Sessions, Battery Settings, ...)
- Anpassung aller Übersetzungstexte. Aktuell wird oft so etwas verwendet: `Unter {percent}% wird ...`. Hier muss zukünftig der gesamte Ausdruck inkl. Prozentzeichen über den Platzhalter kommen damit wir das schmale geschützte Leerzeichen nicht der Sprachpflege überlassen. Ziel: `Unter {percent} wird ...`

StefanSchoof, 2023-06-17:
Meines Wissens nach ist es Sprachabhängig, ob ein Leerzeichen gesetzt wird oder nicht. Siehe zum Beispiel https://style.ons.gov.uk/house-style/percentages/#:~:text=Use%20the%20percentage%20symbol%20with%20no%20space%20between%20it%20and%20the%20number.&text=Always%20write%20%E2%80%9Cpercentage%E2%80%9D%20and%20never,a%20sequence%20of%20decimal%20numbers.

Um es also auch in anderen Sprachen richtig zu machen, müsste man es also Sprachabhängig machen.

naltatis, 2023-06-18:
@StefanSchoof ok, danke für den Hinweis.

Diese Unterscheidung bilden die Browser glücklicherweise bereits über `new Intl.NumberFormat(..., {style: "percent"})` ab. Das müssten wir dann hier nutzen. Leider gibt diese Funktion bei Ländern mit Abstand zwischen Zahl und Prozent immer einen "non-breaking space" (`\u00A0`) und kein "small non-breaking space" (`\u202F`) aus. Das müssten wir dann noch ersetzten.

Heißt die Formatierungsfunktion sähe vermutlich so ähnlich aus:

```
fmtPercent(value, digits) {
  return new Intl.NumberFormat(this.$i18n.locale, {
        style: "symbol",
        minimumFractionDigits: digits,
        maximumFractionDigits: digits,
      })
      .format(value/100)
      .replace(/\u00A0/g, "\u202F")
}
```

Soweit ich das überprüft habe ist der Unterschied zwischen den Locales "nur" ob es einen Abstand gibt oder nicht. `en-US`, `en-UK` haben keinen Abstand, `de-DE`, `no-NO`, `fr-CA` haben einen.

descodess, 2023-06-18:
Neben den **Prozentangaben** sollten auch alle **physikalischen Größen** konsistent – und wo nötig auch sprachabhängig – formatiert werden. Siehe beispielsweise:

- https://de.wikipedia.org/wiki/Physikalische_Gr%C3%B6%C3%9Fe#Formatierung
- https://de.wikipedia.org/wiki/Hilfe:TeX#Physikalische_Gr%C3%B6%C3%9Fen
- https://de.wikipedia.org/wiki/Prozentzeichen#Schreibweisen
- https://en.wikipedia.org/wiki/Percent_sign#Correct_style … Sprachabhängigkeiten für %

Vielleicht sollte der Formatierungsfunktion (z. B. allgemein `fmtPhysicalQuantity` statt `fmtPercent`) also auch die jeweilige Maßeinheit als weiterer Parameter übergeben werden, damit auch Maßeinheit-abhängige Formatierungen möglich werden. Beispiel: Deutsch: `12\u202F%` und `37\u202Fkm`, Englisch: `12%` und `37\u202Fkm`

p7lejo, 2023-07-01:
Ich finde es gut, dass ihr euch mit dem Thema beschäftigt. Tatsächlich würde ich es wohl bei einem (ganzen) Leerzeichen (auf deutsch) belassen, schon alleine, damit ältere Browser oder mobile Geräte auch damit (wahrscheinlicher) klarkommen.

goebelmeier, 2023-07-01:
> [...] alleine, damit ältere Browser oder mobile Geräte auch damit (wahrscheinlicher) klarkommen.

Gibt's denn Beispiele oder Erfahrungen wo das (noch) nicht geht?

PR #13880 (2024-05-17): UI i18n: percentage formatting
fixes #8510, fixes #13836

- format percentage values according to language-specific conventions
- uses `Intl.NumberFormat` for formatting
- uses DIN 5008 for German (see #8510)

overview for all existing languages:

[image]

**English**

[image]

**German**

[image]

**Arabic**

[image]
[image]
[image]
[image]
<img width="740" alt="Bildschirmfoto 2024-05-12 um 16 01 39" src="https://github.com/evcc-io/evcc/assets/152287/17a279f3-9c74-49
closed 2024-05-17 by naltatis