evcc-io/evcc #8511: Easee: Refreshtoken expired

ulrichSchreiner, 2023-06-17:
### Describe the bug

hi,

ich habe mein Auto vor einer Woche (am 11.06.) zum letzten mal geladen. Seitdem war es nciht mehr angesteckt. Heute (17.06.) nachmittags wurde es wieder Zeit und leider wurde mit 11kW geladen ohne umzuschalten. In der Oberfläche von EVCC sieht man oben rechts Meldungen. Diese sind auch im Log zu sehen Der Aufruf von `/api/accounts/refresh_token` schlägt fehl mit dem Fehler

~~~
{"errorCode":105,"errorCodeName":"RefreshTokenExpired","type":null,"title":"Refresh token expired","status":401,"detail":"[Empty in production]","instance":null,"extensions":{}}
~~~

Jeder Aufruf zu easee führt zu diesem Fehler; der `accessToken` ist abgelaufen und der `refreshToken` wohl mittlerweile auch, aber evcc versucht halt immer nen Refresh, der nicht geht. 

Im Log von evcc seh ich halt, dass seit einer Woche keine Calls mehr zu easee erfolgten. Wäre das passiert, wäre wohl zwischendurch der `accessToken` refreshed worden und es wäre dann auch ein neuer Refresh-Token gekommen. Da aber keine Ladevorgang für fast eine Woche war, gabs keinen Call und somit auch keinen Refresh und somit ist auch der Refresh-token abgelaufen.

Die Lösung ist ganz einfach: EVCC durchstarten, dann loggt der sich neu ein, holt nen neuen `accessToken` und gut ist. Nur: mein EVCC läuft in einer Cloud-Instanz als Dienst.  Dieses "Durchstarten" kann in meiner Familie halt nur ich, weil ich dort den systemd-Dienst restarten kann.

Es wäre gut, wenn evcc diesen Fehler selbst erkennen könnte. Denn es kann ja wirklich sein, dass auch der Service-Anbieter alle ausgestellten tokens invalidiert (musste ich bei einem eigenen Dienst mal machen). Die Clients sollten sich dann einfach neu einloggen.

Ggf. könnte evcc einfach 1x/stunde die aktuellen settings abrufen, die dann alle über signalr kommen. so eine Art rekonzilierung. Dann würde der Token zeitnah refreshd werden. 

Alternativ über die Oberfläche einen Restart-Button einführen; da mein systemd den Dienst neu startet, würde mir das helfen; aber andere haben ggf. ein Problem, wenn der Dienst nicht mehr hochkommt.

Am schönsten wäre es natürlich, der Rückgabewert würde erkannt und es erfolgt automatisch ein komplett neuer Login.

### Steps to reproduce

1. Auto mit EVCC/Easee laden
2. mind. 6 Tage lang **keinen** Ladevorgang initiieren, aber evcc muss durchlaufen
3. Auto anstecken
...

### Configuration details

```yaml
chargers:
- name: myeasee
  type: template
  template: easee
  user: "...myuser..."
  password: "...mypassword..."
  charger: mycharter
```
```

### Log details

```text
Dieser Log-Eintrag kommt nun jedesmal, wenn ein Call zu easee notwendig wäre (also recht häufig):

Jun 17 14:09:05 [easee ] TRACE 2023/06/17 14:09:05 POST https://api.easee.com/api/accounts/refresh_token
Jun 17 14:09:05 [easee ] TRACE 2023/06/17 14:09:05
Jun 17 14:09:05 POST /api/accounts/refresh_token HTTP/1.1
Jun 17 14:09:05 Host: api.easee.com
Jun 17 14:09:05 User-Agent: Go-http-client/1.1
Jun 17 14:09:05 Content-Length: 356
Jun 17 14:09:05 Accept: application/json
Jun 17 14:09:05 Content-Type: application/json
Jun 17 14:09:05 Accept-Encoding: gzip
Jun 17 14:09:05 
Jun 17 14:09:05 {"accessToken":"......}
Jun 17 14:09:05 
Jun 17 14:09:05 HTTP/2.0 401 Unauthorized
Jun 17 14:09:05 Content-Length: 177
Jun 17 14:09:05 Access-Control-Allow-Origin: *
Jun 17 14:09:05 Cache-Control: no-cache,no-store
Jun 17 14:09:05 Content-Type: application/problem+json; charset=utf-8
Jun 17 14:09:05 Date: Sat, 17 Jun 2023 12:09:05 GMT
Jun 17 14:09:05 Expires: -1
Jun 17 14:09:05 Pragma: no-cache
Jun 17 14:09:05 X-Amz-Apigw-Id: GqZPUFhUDoEFxfQ=
Jun 17 14:09:05 X-Amzn-Remapped-Content-Length: 177
Jun 17 14:09:05 X-Amzn-Remapped-Date: Sat, 17 Jun 2023 12:09:05 GMT
Jun 17 14:09:05 X-Amzn-Remapped-Server: Kestrel
Jun 17 14:09:05 X-Amzn-Requestid: 10a8d7c7-a0ab-49dc-8abb-ff7dfa6b63f1
Jun 17 14:09:05 
Jun 17 14:09:05 {"errorCode":105,"errorCodeName":"RefreshTokenExpired","type":null,"title":"Refresh token expired","status":401,"detail":"[Empty in production]","instance":null,"extensions":{}}
Jun 17 14:09:05 [lp-1  ] ERROR 2023/06/17 14:09:05 charger: Post "https://api.easee.com/api/chargers/XXXX/commands/pause_charging": unexpected status: 401 (Unauthorized)
Jun 17 14:09:18 [site  ] DEBUG 2023/06/17 14:09:18 ----
```
```

### What type of operating system are you running?

Linux

### Version

0.118.0

andig, 2023-06-17:
Wie sieht denn der Refreshtoken aus? Ist der wirklich expired?

ulrichSchreiner, 2023-06-17:
also der accesstoken ist wirklich expired

da steht das hier drin (mit jwt.io geparsed)
~~~
{
  "AccountId": XXXX,
  "UserId": XXXX,
  "unique_name": "Ulrich Schreiner",
  "role": "User",
  "nbf": 1686479307,
  "exp": 1686565707,
  "iat": 1686479307
}
~~~
und der expired am 12.juni.

der refreshtoken sieht so aus:

~~~
"refreshToken":"9FBQ7s62VSU3C+H5BS/KYQXP5RwbXG96FOsV09QKN8U="
~~~

wie man da jetzt erkennt ob der abgelaufen ist, weis ich nicht, ich wüsste nicht, dass es da einen standard gibt. imho sind refresh-tokens abhängig vom provider

und ja: das mit dem cronjob hab ich eingebaut. ist halt nicht so die feine art ....

andig, 2023-06-17:
Klar, der AT muss ja expiren. Wir könnten ja einfach 1x am Tag einen Request machen?

ulrichSchreiner, 2023-06-17:
wär für mich vollkommen ok. gerne auch 1/stunde :-). 

ich hab mit er 0.118 eh das problem, dass ich schon zum zweiten mal den fehler `max charge current 6A: must retry` hab und sich evcc nicht fängt. was hilft: bei easee in der app einfach "pause" anklicken, dann schlagen bei evcc über signalR eine menge updates auf und dann fängt es sich wieder.

das könnte man auch mit nem regelmässigen abruf der settings-daten, so dass easee einem via signal-R quasi den ganzen zustand "rausbläst". das würde solche kaputten zustände reparieren.

aber das ist ein anderes problem. in dem ticket hier gehts um token-refresh und ja, da würde 1rq/tag vollkommen ausreichend sein.

andig, 2023-06-17:
Probier mal das Nightly, sollte jetzt besser laufen.

closed 2023-06-18 by andig

tobiasbayer, 2023-06-18:
> ich hab mit er 0.118 eh das problem, dass ich schon zum zweiten mal den fehler `max charge current 6A: must retry` hab und sich evcc nicht fängt. was hilft: bei easee in der app einfach "pause" anklicken, dann schlagen bei evcc über signalR eine menge updates auf und dann fängt es sich wieder.

Gibt es dafür schon ein Ticket? Ich habe das Problem auch.

andig, 2023-06-18:
Bitte nightly verwenden und OT-Postings vermeiden. Das hat NICHTS mit den Refreshtokens zu tun.

tobiasbayer, 2023-06-18:
Ja eben. Daher ja auch die Frage nach einem anderen Ticket.