evcc-io/evcc #8542: Invalid EVSE state

RikVerstijnen, 2023-06-19:
### Describe the bug

Error "charger: invalid EVSE state" is shown.

### Steps to reproduce

Use configuration below.
Chose a charging mode other than "Off".
Error is shown.

OpenEVSE version: 8.2.3.EU
OpenEVSE Wifi version: v4.2.2

### Configuration details

```yaml
chargers:
  - name: my_charger
    type: template
    template: openevse
    host: ****
    user: ****
    password: ****
```

### Log details

```text
[lp-1  ] INFO 2023/06/19 20:31:54 stop charging <-
[open***] TRACE 2023/06/19 20:31:54 GET http://****/override
[open***] TRACE 2023/06/19 20:31:54 {"state":"disabled","charge_current":6,"auto_release":true}
[open***] TRACE 2023/06/19 20:32:04 GET http://****/status
[open***] TRACE 2023/06/19 20:32:04 {"mode":"STA","wifi_client_connected":1,"eth_connected":0,"net_connected":1,"ipaddress":"****","emoncms_connected":0,"packets_sent":0,"packets_success":0,"mqtt_connected":1,"ocpp_connected":0,"rfid_failure":0,"ohm_hour":"NotConnected","free_heap":165964,"comm_sent":801489,"comm_success":801467,"rapi_connected":1,"***_connected":1,"amp":0,"voltage":230,"power":0,"pilot":6,"temp":520,"temp_max":520,"temp1":false,"temp2":false,"temp3":false,"temp4":520,"state":254,"status":"disabled","flags":1280,"vehicle":1,"colour":6,"manual_override":1,"freeram":165964,"divertmode":1,"srssi":-84,"time":"2023-06-19T18:32:05Z","local_time":"2023-06-19T20:32:05+0200","offset":"+0200","session_elapsed":636,"session_energy":0,"total_energy":0.034067667,"total_day":0,"total_week":0,"total_month":0,"total_year":0,"total_switches":239,"elapsed":636,"wattsec":0,"watthour":34.067667,"gfcicount":0,"nogndcount":1,"stuckcount":0,"solar":0,"grid_ie":0,"charge_rate":0,"divert_update":510111,"divert_active":false,"shaper":0,"shaper_live_pwr":0,"shaper_cur":0,"shaper_updated":false,"service_level":2,"limit":false,"ota_update":0,"config_version":2,"claims_version":5,"override_version":5,"schedule_version":1,"schedule_plan_version":1,"limit_version":1,"vehicle_state_update":510111,"tesla_vehicle_count":1,"tesla_vehicle_id":"****","tesla_vehicle_name":"****"}
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.1

andig, 2023-06-19:
Seems `state` has moved to `status`. Apparently there are different openEVSE versions around. Without ability to tell apart which is which there is not fix possible.

andig, 2023-06-20:
/cc @sparkyson

RikVerstijnen, 2023-06-21:
This could be a hint: I am using the [v4 wifi board firmware](https://github.com/OpenEVSE/ESP32_WiFi_V4.x). Possibly the implementation is currently based on [v2](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x).

sparkyson, 2023-06-21:
I’m away from the home so I can’t check this week. 
This is the API reference: https://openevse.stoplight.io/docs/openevse-wifi-v4/e0ab0a4ad5e1e-set-the-manual-override

andig, 2023-06-21:
I was not able to regenerate the api from sources. As far as I'm concerned this would be wontfix. Would be happy to take a PR for a potenially different openevse v4.

andig, 2023-07-03:
@sparkyson würdest Du Dir das nochmal anschauen? Würde sonst als wontfix schließen. Vielen Dank!

andig, 2023-07-09:
friendly ping @sparkyson

premultiply, 2023-07-13:
Better to drop current support completly?

sparkyson, 2023-07-13:
Sorry, I missed the message. I will take a look, if we decide it's fine to continue support.
The API cannot be automatically generated from spec, but I can surely fix it manually, if this is acceptable.

Laser-freak, 2023-07-14:
@andig 
selbes verhalten bei mir ...
hab dann den manual override manuell per cli auf dem host der evcc ausgelöst, so wie [hier](https://openevse.stoplight.io/docs/openevse-wifi-v4/e0ab0a4ad5e1e-set-the-manual-override) gesetzt.

seit dem keine Fehlermeldung bei / nach umschalten des Modus im evcc.

andig, 2023-07-16:
@sparkyson wäre Klasse falls Du das machen willst. Habt leider nicht hin bekommen den Client zu generieren und mir ist auch das API Versioning nicht klar. Ich hatte zuletzt ja schon händisch einen Change gemacht (vmtl. am v3 Api). War das überhaupt korrekt? Brauchen wir jetzt einen Client je API Version? Alternative wäre noch das ohne den Client einfach "von Hand" zu machen, vmtl. wäre das in Summe auch weniger fehleranfällig und aufwändig. Auch das müsste aber jemand machen der es auch testen kann...

Falls Du PR machen möchtest sehr gerne- danke!

Laser-freak, 2023-07-16:
@andig 
Welche Informationen werden dazu benötigt ?
Betreibe eine OpenEvse und hab selbiges Problem der Meldung "invalid State" jedesmal wenn Befehle zur EVSE gesendet werden.
Dieses mit EVCC Version 0.117.4 bei OpenEvse Firmware 7.1.2 und Wifi 4.1.5 (ESP32)
Hab in den letzten Tagen mehrere Kombinationen ausprobiert und beim Testen der Versionen diese als soweit lauffähig ausgemacht.

Automatischer Ladestart, Stop und Leistungsanpassung an PV Überschuss funktionieren solang ich die EVSE per CLI in den manuellen Modus versetzt hab ( senden der Anfrage aus dem CLI des Host vom EVCC).

Ab EVCC 0.118.0 kamen bei gleicher OpenEvse WiFi Version 4.1.5 dann weitere Fehlermeldungen hinzu, 
was aber auch an nicht getätigtem Reset der EVSE oder WiFi Modul liegen kann ... es war irgendwann einfach zu spät ;-)
Teste die Tage weiter welche Version zusammen spielt, muss dazu aber erst den Fahrakku wieder etwas leeren um das die EVSE überhaupt plausibel spielen kann ...

andig, 2023-07-16:
Das müsste sich einfach jemand anschauen der Gerät und Knowhow hat...

sparkyson, 2023-07-16:
I'll look into it. I have the hardware and I'll update it to the latest version.

andig, 2023-07-18:
@sparkyson it would be nice to get rid of oapi-codegen in total. I've noticed it recently started bringing in a bunch of big new dependencies like:

```
go mod why -m github.com/gin-gonic/gin
# github.com/gin-gonic/gin
github.com/evcc-io/evcc/charger/openevse
github.com/deepmap/oapi-codegen/pkg/runtime
github.com/gin-gonic/gin
```

Given clients can't be generated anyway I'd suggest to switch to manual interface.

sparkyson, 2023-07-18:
InfluxDB is also using "github.com/deepmap/oapi-codegen/pkg/runtime" so I can't get rig of oapi-codegen unless you have some sage advise.

closed 2023-07-20 by andig

thekev, 2023-07-31:
I ran into this today. I failed to properly pull a fresh master branch, and ended up hacking on the same problem before I gave up.

Confirmed just now a fresh build does enable basic control and charge power measurement. Thank you.

I want to implement more interfaces, but I can’t deal with that generated api code. If I go for that, I’m going to rewrite it and simplify into something more manageable.

thekev, 2023-07-31:
Update: power measurement is a lie. I don't know how it's being derived, but it's not coming from the EVSE. Motivation to refactor increasing. ;)

sparkyson, 2023-07-31:
We removed the support for Currents at @andig recommendation because there's only one CT measuring current even for 3-phase installation.
There's no voltage sensor.
This also means there's no reliable way to measure power accurately.

thekev, 2023-07-31:
Ah, yes. Voltage from `/status` is a lie. I restarted the ESP and controller, and I'm getting less nonsense now. EVCC logs were reading 9200W when it was actually ~1.4kW - it was not delivering at the rate commanded by the `/override` call. Looks like the `power` property in `/status` is new in 4.2.2, along with additional energy meters. 

Btw, voltage can be informed via MQTT, Tesla API, or the Wifi API. EVCC would have to assume voltage is the same in 3-phase mode. That's the only capability of the EVSE, @andig do you require it to be exact to include it at all? From perusing the EVSE firmware source, I conclude toggling 3-phase only alters the power calculations, and not the phases used for charging. Is this your experience @sparkyson?

andig, 2023-08-02:
Voltage is only used to detect 1p vs 3p mains connection. If that's not possible since the values are fake we can just remove the api.

thekev, 2023-08-03:
@sparkyson I'm confused here. I cannot find `G7` or `S7` in the [controller firmware](https://github.com/OpenEVSE/open_evse/blob/master/firmware/open_evse/rapi_proc.cpp) for three phase control. What wifi/controller firmware are you using?

sparkyson, 2023-08-04:
Hi! Indeed, it is an experimental support for phase switching for OpenEVSE which doesn't support that out of the box.
This is the branch in case you're interested https://github.com/sparkyson/open_evse/tree/stable.
I never actually sent the pull request to the maintainer since I stopped using the hardware due to frequent hardware failure due to overheating. 
At this point, if someone is interested in continuing the support, they have my full support and appreciation, but, if not, we can (should?) probably remove the support from EVCC as well.

thekev, 2023-08-05:
It looks safe to leave there, at least until a feature with ID 7 shows up in mainline. :wink: It works fine with my single phase unit, which doesn't understand `G7`. That said, if it's not used by anyone, it might be wise to remove it.

In my local branch I've refactored and simplified the openevse api, and I'm happy with how it's working. I had a segfault today in an uncaught error condition, so I want to let it bake for a while longer before proposing a PR. I also haven't tested the http basic auth yet. I'll open another issue for this change when it's ready. I left out the voltage and current readings, but included `CurrentPower`. That will require 4.2.2, so I still need to work out a decorator for omitting it when not supported. These are the [changes made so far](https://github.com/evcc-io/evcc/compare/9a283814...thekev:evcc:openevse).

andig, 2023-08-05:
Feel free to open PR if you need help with any coding topics.

andig, 2023-08-19:
@thekev can we do anything to move this forward? I'd really like to get rid of the oapi step.

thekev, 2023-08-20:
> @thekev can we do anything to move this forward? I'd really like to get rid of the oapi step.

My branch has been running fine for a few weeks now. I’ll get a PR in as soon as I have time.

andig, 2023-09-29:
@thekev anything I can help with?

andig, 2023-09-29:
Note: I've grabbed https://github.com/evcc-io/evcc/pull/10110 from your repo- please object if you don't want that or it needs fixing. I can't test it.

thekev, 2023-10-17:
> Note: I've grabbed #10110 from your repo- please object if you don't want that or it needs fixing. I can't test it.

Sorry I’ve been unresponsive. New job, all my free time working on the house etc. That’s fine. It is working well for my single phase unit. There’s one patch that I haven’t pushed which fixes a minor issue where it’s expected that a charge current or disable override hasn’t yet been set. 

```
diff --git a/charger/openevse.go b/charger/openevse.go
index 1fd07f70..ed1222fe 100644
--- a/charger/openevse.go
+++ b/charger/openevse.go
@@ -65,7 +65,9 @@ func (c *OpenEVSE) setOverride() error {
 
        err := c.GetJSON(uri, &data)
        if err != nil {
-               return err
+               if err.Error() != "unexpected status: 404 (Not Found)" {
+                       return err
+               }
        }
 
        state := wbDisabled
```

andig, 2023-10-17:
I've put this into https://github.com/evcc-io/evcc/pull/10362

PR #9015 (2023-07-20): OpenEVSE: fix status handling
Fixes #8542