evcc-io/evcc #8548: 'Connection Refused' when using DSMR template to integrate slimmelezer+

bolisaf, 2023-06-19:
Hi,

I've used the [template ](https://docs.evcc.io/docs/devices/meters/#dsmr)described here to integrate a [Slimmelezer+](https://www.zuidwijk.com/product/slimmelezer-plus/) (wireless variant of the serial P1 reader)

i keep on getting a 'connection refused' error. in HA it's running fine, so my issue must be in evcc.yaml i think.

i'm using the latest version of EVCC. i've also read [this ](https://github.com/evcc-io/evcc/issues/2880)thread which discusses the same issue but couldn't find an answer
[image]

andig, 2023-06-19:
> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

closed 2023-06-19 by andig

premultiply, 2023-06-20:
Other system blocking the single possible TCP connection to the device?
Wrong port?
Who knows...

bolisaf, 2023-06-20:
> Other system blocking the single possible TCP connection to the device? Wrong port? Who knows...

hey, thanks for replying. I did turn off the Slimmelezer+ integration to HA to check if there was a single connection issue, but that did not solve the problem.
I then check on which port HA is talking to the P1 reader which is 6053. i changed evcc.yaml to match that port number, and now i'm getting an IO timeout error instead of a connection refused.

it just feels like i'm doing something which can't work. Can anyone confirm they have successfully integrated a Slimmelezer+ to EVCC?