evcc-io/evcc #8554: 1-3 Phasen Umschaltung wenn kein Fahrzeug angeschlossen ist.

buliwyf42, 2023-06-20:
Aktuell erfolgt die Umschaltung nicht, wenn kein Fahrzeug angeschlossen ist.

Folgendes Szenario:

Morgens wird das Auto abgesteckt, während es nur mit einer Phase geladen wird. 
Wenn ich es am Nachmittag wieder einstecke, fängt es erst mit einer Phase an und schaltet dann gleich wieder um auf 3. Phasen. 

Gleiches passiert auch andersherum.

Vorschlag:
Evcc könnte die Phasenumschaltung je nach Überschuss auch an die Wallbox schicken, wenn das Gerät nicht angesteckt ist.
Falls man das Fahrzeug dann ansteckt, würde es direkt mit der aktuell richtigen Anzahl an Phasen geladen.

andig, 2023-06-20:
Dann würden den ganzen Tag die Relais klackern... schöner wäre bei Ladebeginn erst die Phasen zu prüfen und dann umzuschalten. Tatsächlich sollte das eigentlich auch heute so sein, nur dass unnötigerweise noch eine Pause vor Umschaltung erfolgt die eigentlich nicht notwendig ist. Ich würde das lieber auf dem Weg lösen, dass wir die Pause vor der initialen Umschaltung eliminieren. Wenn ihr es anders seht können wir das gerne in eine Diskussion dazu verwandeln.

/cc @premultiply

closed 2023-06-20 by andig

buliwyf42, 2023-06-20:
Ich bin da eventuell etwas verwöhnt mit 19 kwp. Da bin ich halt recht fix über den 3,6 KW. Da würde die Wallbox vermutlich nicht allzu häufig umschalten.  
Das mit der Pause vor dem Ladebeginn finde ich gut. Prüfen was aktuell geht, umschalten und dann den Ladevorgang starten.