evcc-io/evcc #8556: 'Connection Refused' when using DSMR template to integrate slimmelezer+

bolisaf, 2023-06-20:
### Describe the bug

I've used the [template ](https://docs.evcc.io/docs/devices/meters/#dsmr)described here to integrate a [Slimmelezer+](https://www.zuidwijk.com/product/slimmelezer-plus/) (wireless variant of the serial P1 reader)

i keep on getting a 'connection refused' error. in HA it's running fine, so my issue must be in evcc.yaml i think.

i'm using the latest version of EVCC. i've also read https://github.com/evcc-io/evcc/issues/2880thread which discusses the same issue but couldn't find an answer

I've shut down the HA integration when trying to connect from EVCC, same result. I've changed the port to the port that home assistant is talking to (6053) and then I get an IO timeout

### Steps to reproduce

1. check if Slimmelezer+ is sending data to HA
2.add DSMR template to evcc.yaml
3.change IP number, comment out the port
4.start up evcc add-on from HA
...

### Configuration details

```yaml
sponsortoken: *************

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: slimmelezer
    type: template
    template: dsmr
    usage: grid
    host: 10.10.10.46 # IP-Adresse oder Hostname
#    port: 6053 # Port (Optional) 

chargers:
  - name: alfeneve
    type: template # Wallbe charger
    template: alfen
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 10.10.10.48 # Hostname
    port: 502 # Port 

site:
  title: VA35 # display name for UI
  meters:
    grid: slimmelezer # grid meter
    pv:
      - pv # list of pv inverters/ meters
  residualPower: 0 # additional household usage margin
  prioritySoc: 0 # give home battery priority up to this soc (empty to disable)
  bufferSoc: 0 # continue charging on battery above soc (0 to disable)
  bufferStartSoc: 0 # start charging on battery above soc (0 to disable)
  maxGridSupplyWhileBatteryCharging: 0 # ignore battery charging if AC consumption is above this value
  smartCostLimit: 0 # set cost limit for automatic charging in PV mode

loadpoints:
  - title: Oprit # display name for UI
    charger: alfeneve # charger
    meter: slimmelezer # charge meter
    mode: "off" # set default charge mode, use "off" to disable by default if charger is publicly available
    # vehicle: car1 # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 3 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 32 # maximum charge current (default 16A)
```

### Log details

```text
sing config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/06/20 20:48:57 evcc 0.118.1
[main  ] INFO 2023/06/20 20:48:57 using config file: /config/evcc.yaml
[main  ] INFO 2023/06/20 20:48:57 starting ui and api at :7070
[main  ] FATAL 2023/06/20 20:48:57 cannot create meter 'slimmelezer': cannot create meter 'template': cannot create meter 'dsmr': dial tcp 10.10.10.46:1502: connect: connection refused
[main  ] FATAL 2023/06/20 20:48:57 will attempt restart in: 5m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.1

andig, 2023-06-20:
> dial tcp 10.10.10.46:1502: connect: connection refused

Nothing we can do about that :/

closed 2023-06-20 by andig

andig, 2023-06-20:
I do not know your meter and without logfile can't say anything about the timeout.

bolisaf, 2023-06-20:
Hi @andig ,

I might be incorrect but I don't think you need to worry about what type of meter I have exactly. you are talking to my P1 reader, Slimmelezer+. the address for the data you need is always on the same address (in [this](https://github.com/evcc-io/evcc/pull/3284) post you say which one) since it's the DSMR format. and in [this](https://github.com/evcc-io/evcc/issues/2880) post it is confirmed that the template you created works with the Slimmelezer P1 reader.

as for the logs, I changed the config again to get evcc to talk to 6053, this is the logs that I get:

`Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/06/20 22:24:15 evcc 0.118.1
[main  ] INFO 2023/06/20 22:24:15 using config file: /config/evcc.yaml
[main  ] INFO 2023/06/20 22:24:15 starting ui and api at :7070
[main  ] FATAL 2023/06/20 22:24:33 cannot create meter 'slimmelezer': cannot create meter 'template': cannot create meter 'dsmr': i/o timeout
[main  ] FATAL 2023/06/20 22:24:33 will attempt restart in: 5m0s`

I'm running it as an HA add-on and because the container shuts down right after running into this error, it's hard to sh into it and run evcc in debug mode. any tips there?

thanks again
L

StefanSchoof, 2023-06-20:
From the bug Formular:
> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.

So try if it works from you PC and turn debug level up.

andig, 2023-06-21:
> i/o timeout

Is just that- device doesn't send data in expected time. Default is 15s, maybe try higher value?

bolisaf, 2023-06-21:
> From the bug Formular:
> 
> > NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.
> 
> So try if it works from you PC and turn debug level up.

thanks for that, sorry i missed it

bolisaf, 2023-06-21:
> > i/o timeout
> 
> Is just that- device doesn't send data in expected time. Default is 15s, maybe try higher value?

okay, where can i set the timeout value? it isn't mentioned [here](https://docs.evcc.io/docs/devices/meters/#dsmr)

and also, can anyone please confirm what the port is supposed to be? 1502 of 6053?

bolisaf, 2023-07-12:
hi guys, I'm still stuck. is there anyone in here who has managed to get the slimmelezer+ to work with evcc?

bolisaf, 2023-07-17:
Dear EVCC community, can anyone help me out here? I'm at the end of my wits and I'll I don't see any other option than abondoning this project entirely if I can get this to work

premultiply, 2023-07-17:
Looks like you are using a template that is no designed for this type of product.

andig, 2023-07-17:
I wouldn't say that but nobody knows why that meter doesn't answer or even which port it should answer on. Nothing we could do.

bolisaf, 2023-07-17:
it looks like that, but [this](https://github.com/evcc-io/evcc/issues/2880) thread it says that the DSMR template was designed for Slimmelezer+

what can I do to use the date from the P1 slimmelezer+ into evcc if I can't use that template? I have the data in HA, can I publish them to mqtt and then read them again with evcc? is that the best option?

bolisaf, 2023-07-17:
> Maybe try other firmware? 🤷🏻‍♂️
> https://www.zuidwijk.com/using-esp-link/

can someone (andig?) just confirms what the DSMR template was tested on?

premultiply, 2023-07-17:
P1 Meter Ethernet

bolisaf, 2023-07-17:
> P1 Reader Ethernet
> https://www.zuidwijk.com/product/p1-reader-ethernet/

thanks for confirming that. I have the wireless version, slimmelezer+
https://www.zuidwijk.com/product/slimmelezer-plus/

@andig should it work for that one too?

premultiply, 2023-07-17:
No, it's something different.

bolisaf, 2023-07-17:
@andig can you add support for slimmelezer+ or do I need to find a different solution?

andig, 2023-07-17:
Probably if there is any api documentation.

bolisaf, 2023-07-17:
do you mean this?
https://esphome.io/web-api/

bolisaf, 2023-07-17:
there's also this

https://github.com/zuidwijk/dsmr/blob/main/slimmelezer.yaml

bolisaf, 2023-07-19:
@andig do you have enough to work with? would you like to connect to my slimmelezer+ remotely to test?

bolisaf, 2023-07-23:
Hi @andig this here seems to work, but I have 2 issues:
- the current values are always positive. doesn't matter if I'm producing or consuming
- I'm reading values correctly, but I'm getting a lot of IO timeouts. can I set the interval somehow? I think the slimmelezer just can't keep up. the average response time is 5 seconds.

`- name: slimmelezer
  type: custom
  power:
    source: http
    uri: http://slimmelezer.local/sensor/power_produced
    method: GET
    headers:
      - content-type: application/json
    jq: .value
    scale: 1000
    timeout: 15s
  energy:
    source: calc
    add:
    - source: http
      uri: http://slimmelezer.local/sensor/energy_produced_tariff_1
      method: GET
      headers:
      - content-type: application/json
      jq: .value
      scale: 1
      timeout: 15s
    - source: http
      uri: http://slimmelezer.local/sensor/energy_produced_tariff_2
      method: GET
      headers:
      - content-type: application/json
      jq: .value
      scale: 1
      timeout: 15s
  currents:
  - source: http
    uri: http://slimmelezer.local/sensor/current_phase_1
    method: GET
    headers:
    - content-type: application/json
    jq: .value
    scale: 1
    timeout: 15s
  - source: http
    uri: http://slimmelezer.local/sensor/current_phase_2
    method: GET
    headers:
    - content-type: application/json
    jq: .value
    scale: 1
    timeout: 15s
  - source: http
    uri: http://slimmelezer.local/sensor/current_phase_3
    method: GET
    headers:
    - content-type: application/json
    jq: .value
    scale: 1
    timeout: 15s`

andig, 2023-07-23:
> the current values are always positive. doesn't matter if I'm producing or consuming

What does the api return? Is there any discriminator?

> the average response time is 5 seconds.

That's horrible. Try `cache: 3s` to reduce the number of api requests.

bolisaf, 2023-07-23:
> > the current values are always positive. doesn't matter if I'm producing or consuming
> 
> What does the api return? Is there any discriminator?
> 
> > the average response time is 5 seconds.
> 
> That's horrible. Try `cache: 3s` to reduce the number of api requests.

I'm sorry.. what is a discriminator? here's a screenshot of the response in postman
[image]

the IO timeout is solved if I replace the mDNS name with the IP address.

bolisaf, 2023-07-23:
another question:
there are 2 registers regarding power: power_produced and power_consumed. power_produced gives you the excess power, so power generated by the pv installation minus what the house is consuming. but the evcc code I think expects one register for both. see the screen: it takes what the pv generates and then adds the power I'm producing to get the total usage of power, where it should substract it. I think I need to take power_consumed and subtract power_produced from it to make the logic work. is that correct?

[image]