evcc-io/evcc #8560: Easee Wallbox "invalid status. 400" after car disconnecting

josefzahner, 2023-06-21:
### Describe the bug

I'm getting constantly a message in EVCC with the following text "charger: invalid status: 400" after we disconnect our car (Tesla MYP) from the Easee Wallbox.

### Steps to reproduce

Disconnect the car from the easee wallbox.

### Configuration details

```yaml
chargers:
- name: wallbox4
  type: template
  template: easee
  user: xxx
  password: xxx
  charger: EHxxxxx
```

### Log details

```text
Jun 21 07:17:55 evcc evcc[1227]: [lp-1  ] INFO 2023/06/21 07:17:55 start charging ->
Jun 21 08:12:25 evcc evcc[1227]: [lp-1  ] INFO 2023/06/21 08:12:25 stop charging <-
Jun 21 08:12:25 evcc evcc[1227]: [lp-1  ] INFO 2023/06/21 08:12:25 car disconnected
Jun 21 08:12:25 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:12:25 charger: invalid status: 400
Jun 21 08:12:35 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:12:35 charger: invalid status: 400
Jun 21 08:12:45 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:12:45 charger: invalid status: 400
Jun 21 08:12:55 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:12:55 charger: invalid status: 400
Jun 21 08:13:05 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:13:05 charger: invalid status: 400
Jun 21 08:13:15 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:13:15 charger: invalid status: 400
Jun 21 08:13:25 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:13:25 charger: invalid status: 400
Jun 21 08:13:35 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:13:35 charger: invalid status: 400
Jun 21 08:13:45 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:13:45 charger: invalid status: 400
Jun 21 08:13:55 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:13:55 charger: invalid status: 400
Jun 21 08:14:05 evcc evcc[1227]: [lp-1  ] ERROR 2023/06/21 08:14:05 charger: invalid status: 400
```

### What type of operating system are you running?

Linux

### Version

0.118.1

ulrichSchreiner, 2023-06-21:
hi,

same here with 0.118.1

the following error appears every minute after the car was disconnected

~~~
 [easee ] TRACE 2023/06/21 11:10:12 POST https://api.easee.com/api/chargers/XXXX/commands/resume_charging
 [easee ] TRACE 2023/06/21 11:10:12
 POST /api/chargers/XXXX/commands/resume_charging HTTP/1.1
 Host: api.easee.com
 User-Agent: Go-http-client/1.1
 Content-Length: 0
 Authorization: Bearer XXXX
 Content-Type: application/json
 Accept-Encoding: gzip

 HTTP/2.0 400 Bad Request
 Content-Length: 176
 Access-Control-Allow-Origin: *
 Cache-Control: no-cache,no-store
 Content-Type: application/problem+json; charset=utf-8
 Date: Wed, 21 Jun 2023 09:10:12 GMT
 Expires: -1
 Pragma: no-cache
 X-Amz-Apigw-Id: G3KyRFhnjoEFvCw=
 X-Amzn-Remapped-Connection: keep-alive
 X-Amzn-Remapped-Content-Length: 176
 X-Amzn-Remapped-Date: Wed, 21 Jun 2023 09:10:12 GMT
 X-Amzn-Remapped-Server: Kestrel
 X-Amzn-Requestid: 49c0fa34-8ce9-4444-89cc-b745ff7e3761

 {"errorCode":410,"errorCodeName":"ChargerDisconnected","type":null,"title":"Charger disconnected","status":400,"detail":"[Empty in production]",">
~~~

ulrichSchreiner, 2023-06-21:
more info: the car was disconnected at 09:52:57 and this event was logged in evcc:

~~~
[lp-1  ] INFO 2023/06/21 09:52:57 car disconnected
~~~

but as you can see by my previous comment, evcc endlessly tried to `resume_charging` until the car was plugged in about 1.5h later.

jheinitz, 2023-06-21:
Moin!

Die Meldung bekomme ich auch. Allerdings gleich nach dem Anschließen des Skoda Enyaqs. Hier ein Auszug aus dem Log mit meinem selbst erstellten Docker Image auf Basis des Commit `b2b4483b`:
```
[site  ] DEBUG 2023/06/21 15:00:45 ----
[lp-1  ] DEBUG 2023/06/21 15:00:45 charge power: 0W
[site  ] DEBUG 2023/06/21 15:00:45 co2 tariff
[site  ] DEBUG 2023/06/21 15:00:45 pv power: 6363W
[site  ] DEBUG 2023/06/21 15:00:45 battery soc: 100%
[site  ] DEBUG 2023/06/21 15:00:45 battery power: 0W
[site  ] DEBUG 2023/06/21 15:00:45 grid power: -4345W
[site  ] DEBUG 2023/06/21 15:00:45 grid currents: [-8.52 -8.48 -8.23]A
[site  ] DEBUG 2023/06/21 15:00:45 site power: -4095W
[lp-1  ] DEBUG 2023/06/21 15:00:45 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/21 15:00:45 charge total import: 4568.782kWh
[lp-1  ] DEBUG 2023/06/21 15:00:45 charger status: A
[easee ] DEBUG 2023/06/21 15:00:50 current mismatch, expected 6.0, got 32.0
[site  ] DEBUG 2023/06/21 15:01:05 ----
[lp-1  ] DEBUG 2023/06/21 15:01:05 charge power: 509W
[site  ] DEBUG 2023/06/21 15:01:05 co2 tariff
[site  ] DEBUG 2023/06/21 15:01:05 pv power: 5275W
[site  ] DEBUG 2023/06/21 15:01:05 battery soc: 100%
[site  ] DEBUG 2023/06/21 15:01:05 battery power: 0W
[site  ] DEBUG 2023/06/21 15:01:05 grid power: -3475W
[site  ] DEBUG 2023/06/21 15:01:05 grid currents: [-6.84 -6.79 -6.52]A
[site  ] DEBUG 2023/06/21 15:01:05 site power: -3225W
[lp-1  ] DEBUG 2023/06/21 15:01:05 charge currents: [15.6 15.6 15.6]A
[lp-1  ] DEBUG 2023/06/21 15:01:05 charge total import: 4568.782kWh
[lp-1  ] DEBUG 2023/06/21 15:01:05 charger status: C
[lp-1  ] INFO 2023/06/21 15:01:05 car connected
[lp-1  ] DEBUG 2023/06/21 15:01:05 vehicle api refresh
[lp-1  ] DEBUG 2023/06/21 15:01:05 pv timer elapse
[lp-1  ] DEBUG 2023/06/21 15:01:05 pv timer inactive
[lp-1  ] DEBUG 2023/06/21 15:01:05 charger: guard elapse
[lp-1  ] DEBUG 2023/06/21 15:01:05 guard timer inactive
[lp-1  ] DEBUG 2023/06/21 15:01:05 charge total import: 4568.782kWh
[lp-1  ] INFO 2023/06/21 15:01:05 start charging ->
[lp-1  ] DEBUG 2023/06/21 15:01:05 wake-up timer: stop
[lp-1  ] DEBUG 2023/06/21 15:01:05 charger vehicle id: 1Dxxxxxxx00001
[lp-1  ] WARN 2023/06/21 15:01:05 charger out of sync: expected disabled, got enabled
[lp-1  ] ERROR 2023/06/21 15:01:05 charger: invalid status: 400
[site  ] DEBUG 2023/06/21 15:01:05 telemetry: charge: Δ3/3Wh @ 509W
[site  ] ERROR 2023/06/21 15:01:06 telemetry: upload failed: unexpected status: 403 (Forbidden)
[site  ] DEBUG 2023/06/21 15:01:25 ----
[lp-1  ] DEBUG 2023/06/21 15:01:25 charge power: 11106W
[site  ] DEBUG 2023/06/21 15:01:25 co2 tariff
[site  ] DEBUG 2023/06/21 15:01:25 pv power: 6203W
[site  ] DEBUG 2023/06/21 15:01:25 battery soc: 100%
[site  ] DEBUG 2023/06/21 15:01:25 battery power: 2367W
[site  ] DEBUG 2023/06/21 15:01:25 grid power: 2802W
[site  ] DEBUG 2023/06/21 15:01:25 grid currents: [4.63 4.39 4.73]A
[site  ] DEBUG 2023/06/21 15:01:25 site power: 5419W
[lp-1  ] DEBUG 2023/06/21 15:01:25 charge currents: [15.6 15.6 15.6]A
[lp-1  ] DEBUG 2023/06/21 15:01:25 detected active phases: 3p
[lp-1  ] DEBUG 2023/06/21 15:01:25 charge total import: 4568.782kWh
[lp-1  ] DEBUG 2023/06/21 15:01:25 charger status: C
[lp-1  ] WARN 2023/06/21 15:01:25 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2023/06/21 15:01:26 available power 5687W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/06/21 15:01:26 start phase scale3p timer
[lp-1  ] DEBUG 2023/06/21 15:01:26 phase scale3p in 1m0s
[lp-1  ] DEBUG 2023/06/21 15:01:26 pv charge current: 0A = 0A + -7.85A (5419W @ 3p)
[lp-1  ] DEBUG 2023/06/21 15:01:29 max charge current: 6A
[lp-1  ] DEBUG 2023/06/21 15:01:32 charger enable
[lp-1  ] DEBUG 2023/06/21 15:01:32 wake-up timer: start
[site  ] DEBUG 2023/06/21 15:01:32 telemetry: charge: Δ64/84Wh @ 11106W
[site  ] ERROR 2023/06/21 15:01:32 telemetry: upload failed: unexpected status: 403 (Forbidden)
```

Das Laden klappt soweit. Allerdings scheint die Sonne noch so sehr, dass kein Phasenwechsel stattfindet.

Gruß Jens

jheinitz, 2023-06-21:
So, meine Frau hat das Auto abgestöpselt und nun bekomme ich auch in jedem Interval die Meldung:
```
[lp-1  ] ERROR 2023/06/21 15:48:25 charger: invalid status: 400
```
Nach dem Restart des **evcc** Containers kommt die Meldung nicht mehr. Soviel von mir dazu.

Gruß Jens

GrimmiMeloni, 2023-06-21:
Hab gerade mal geschaut, seit gestern nur genau 1 Vorkommnis. Bei mir beim Anstöpseln. Danach Ruhe.
Hat außer mir noch jemand die Response logs aktiviert? Zumindest in meinem Fall sieht unser Request nicht gut aus. Da fehlt das Token. Sieht das bei den anderen auch so aus?

```
POST /api/chargers/EXXXXX/commands/pause_charging HTTP/1.1^M
Content-Length: 0^M
Content-Type: application/json^M
Accept-Encoding: gzip

HTTP/2.0 400 Bad Request^M
Content-Length: 176^M
Access-Control-Allow-Origin: *^M
Cache-Control: no-cache,no-store^M
Content-Type: application/problem+json; charset=utf-8^M
Date: Tue, 20 Jun 2023 06:21:36 GMT^M
Expires: -1^M
Pragma: no-cache^M
X-Amz-Apigw-Id: GzfJjFkWjoEFTLA=^M
X-Amzn-Remapped-Connection: keep-alive^M
X-Amzn-Remapped-Content-Length: 176^M
X-Amzn-Remapped-Date: Tue, 20 Jun 2023 06:21:36 GMT^M
X-Amzn-Remapped-Server: Kestrel^M
X-Amzn-Requestid: 8c400e73-a8ca-4f0d-828c-3383b75870cf^M
^M
{"errorCode":410,"errorCodeName":"ChargerDisconnected","type":null,"title":"Charger disconnected","status":400,"detail":"[Empty in production]","instance":null,"extensions":{}}
```

@andig hattest Du nicht die Tage nochmal einen keep alive eingebaut? Ist der Change schon drin?

jheinitz, 2023-06-21:
Moin.

Das ist drin. Ist der PR #8516.

Gruß Jens

GrimmiMeloni, 2023-06-21:
> Da fehlt das Token.

Ich muß mich korrigieren. Habe mich vorhin mit dem Scrollback Buffer meines Terminals beim kopieren des Log Eintrags selbst ausgetrickst. Der Token ist bei mir da. Ich schaue nochmal genauer.

GrimmiMeloni, 2023-06-21:
Ich gehe mittlerweile ziemlich sicher von einem Problem bei Easee aus.
Bei mir im Log folgt auf den gescheiterten 400er direkt im nächsten Interval ein erfolgreicher 202er. 
Ich habe mir die Request traces von beiden angeschaut - sie sind 100% identisch.

GrimmiMeloni, 2023-06-21:
Ein weiteres Indiz, dann mach ich hier erstmal schluß - während meiner Ladevorgänge gestern und heute (20.6 und 21.6.) gab es eine Hand voll Timeouts beim setzen des Currents. Auffällig ist, das diese alle zeitlich nah an den Fehlern der Logs von allen anderen hier liegen.

```
lp-1  ] ERROR 2023/06/20 08:21:36 charger: invalid status: 400

[lp-1  ] ERROR 2023/06/20 10:08:45 max charge current 16A: timeout
[lp-1  ] ERROR 2023/06/20 10:08:55 max charge current 18A: timeout

[lp-1  ] ERROR 2023/06/20 10:23:35 max charge current 23A: timeout
[lp-1  ] ERROR 2023/06/20 10:29:35 max charge current 22A: timeout

[lp-1  ] ERROR 2023/06/20 11:01:35 max charge current 12A: timeout
[lp-1  ] ERROR 2023/06/20 11:05:35 max charge current 16A: timeout
[lp-1  ] ERROR 2023/06/20 11:07:45 max charge current 26A: timeout

[lp-1  ] ERROR 2023/06/20 12:16:25 max charge current 19A: timeout

[lp-1  ] ERROR 2023/06/20 13:03:35 max charge current 16A: timeout
[lp-1  ] ERROR 2023/06/20 13:05:35 max charge current 21A: timeout

[lp-1  ] ERROR 2023/06/21 14:33:51 max charge current 6A: timeout

[lp-1  ] ERROR 2023/06/21 15:03:37 max charge current 21A: timeout
[lp-1  ] ERROR 2023/06/21 15:10:15 max charge current 14A: timeout

[lp-1  ] ERROR 2023/06/21 15:34:55 max charge current 9A: timeout
[lp-1  ] ERROR 2023/06/21 15:36:05 max charge current 13A: timeout
[lp-1  ] ERROR 2023/06/21 15:39:55 max charge current 25A: timeout
[lp-1  ] ERROR 2023/06/21 15:41:25 max charge current 26A: timeout
[lp-1  ] ERROR 2023/06/21 15:42:05 max charge current 26A: timeout
[lp-1  ] ERROR 2023/06/21 16:08:05 max charge current 25A: timeout
```

Insbesondere im Zeitraum heute Nachmittag ab 15:30 gab es wohl Probleme.
Wohlgemerkt, das sind nur die Timeouts. Ich habe für alle auch dann relativ schnell im nächsten Aufruf wieder einen erfolgreichen Call.

Ich schlage vor das hier mit Verweis auf Probleme bei Easee zu schließen.

ulrichSchreiner, 2023-06-22:
hi,

mir ist nicht klar, wo da ein Problem bei easee sein soll. Ich hab seit Wochen bei easee den Trace an, da war gestern garnix fehlerhaft. Man sieht auch in dem Log-Ausschnitt von mir: der Fehler ist, dass evcc im Minutentakt versucht den API call `resume_charging` aufzurufen, obwohl das Auto abgesteckt ist und easee antwortet da ganz sauber mit

~~~json
{ "errorCode":410,
  "errorCodeName":"ChargerDisconnected",
  "type":null,
  "title":"Charger disconnected",
  "status":400,
  "detail":"[Empty in production]",
  "instance":null,
  "extensions":{}}
~~~

verpackt in einem HTTP-400.

Und es ist reproduzierbar: Gestern sofort nach dem abstecken des Autos passiert. Exakt 2x (einmal vormittags, dann wieder am Spätnachmittag).

Das Ticket mit Verweis auf "Probleme bei easee" zu schliessen kann ich nicht nachvollziehen. Es gab bei mir keinen einzigen "fehlerhaften" Call zu easee. Die 400er Antworten von easee sind ja vollkommen rational ok, weil das Laden nicht fortgesetzt werden kann, wenn kein Auto angesteckt ist.

Wenn das so bleibt, kann ich mir gut vorstellen, dass easee da dann wieder anfängt IP-Adressen zu blocken. Denn wenn ein Client im Minutentakt Fehler produziert, ist es nicht so ungewöhnlich diesen dann irgendwann mal zu sperren.

jheinitz, 2023-06-22:
Moin!

Könnte man nicht einfach sehen, ob diese Fehler auch mit Version `0.117.4` kommen? Wenn nicht, dann denke ich auch, dass es nicht an **Easee**, sondern an **evcc** liegt. Bin heute nicht zu Hause, kann das also selbst gerade nicht ausprobieren.

Gruß Jens

ulrichSchreiner, 2023-06-22:
mit der 0.118.0 gabs den fehler, aber nicht in einer loop. d.h. am ende des ladens, wenn ich abgesteckt hab, kam ein `resume_charging` und dann ein `pause_charging`, die beide korrekterweise von easee mit einem 400er beantwortet wurden. im content des responses war (wie jetzt auch) der hinweis, dass kein auto angeschlossen ist.

im unterschied zu jetzt, war es das dann aber auch, sprich dann war ruhe. mit der neuen version kommt der fehler mit jedem intervall. ich hatte den gestern den fehlerinnerhalb von 90min 120x. am abend dann nur 4x, weil ich davon wusste und evcc durchgestartet hab.

GrimmiMeloni, 2023-06-22:
Ja ok, dann müssen wir uns das nochmal genau anschauen.

@ulrichSchreiner kannst Du nochmal das genaue vorgehen beim Abstecken beschreiben? Insbesondere wie Du den Ladevorgang stoppst (via Evcc, über die App vom Auto, durch Öffnen des Fahrzeugs). 
Ich bin etwas irritiert, daß evcc zunächst ein resume und dann ein pause sendet. Insbesondere der resume ist komisch. Da verstehe ich noch nicht wo der herkommt.
Am besten wäre ein vollständiges Trace Log - zumindest vom Zeitraum des Absteckens (+/- 1 min) , da ich das hier nicht nachgestellt bekomme.

GrimmiMeloni, 2023-06-22:
@andig please assign to me

andig, 2023-06-22:
Brauchen wir ein Rollback oder funktioniert es prinzipiell?

ulrichSchreiner, 2023-06-22:
also bei mir funktioniert es prinzipiell, ich weis nur nicht, ob es negative auswirkungen hat, wenn der `resume_charging` ggf. die ganze nacht und dann mehrere tage im minutentakt zu easee geht. die 120 aufrufe in 90min haben jedenfalls bei mir zu keiner sperrung der IP geführt :-)

GrimmiMeloni, 2023-06-22:
Mein Verständnis ist, es funktioniert, aber nach dem Abstecken kommt es in einer bestimmten Konstellation zu dem dauerhaften pause_charge call.
Der API call hat laut Dokumentation kein Rate Limit - ich erwarte hier erstmal kein Problem. Aber ich bleib jetzt auch am Ball. Das sollten wir hoffentlich heute oder morgen verstanden und gelöst haben. Ich würde den Rollback erstmal noch nicht sehen.

andig, 2023-06-22:
> Der API call hat laut Dokumentation kein Rate Limit - ich erwarte hier erstmal kein Problem

Der war gut :)

ulrichSchreiner, 2023-06-22:
@GrimmiMeloni nein, ein dauerhafter `resume_charging`

zu deiner frage wie ich beende: Einfach kabel am auto abstecken, fertig. Das es dann mal eine Race-Condition geben kann, weil evcc grad nen Call abgesetzt hat ist ok. Deshalb hab ich mir bei einem fehlerhaften `pause_charging` oder `resume_charging` nix gedacht. kann immer mal passieren, wenn man zwischendrin mal absteckt.

aber jetzt läuft der immer imm `resume_charging` obwohl im log auch steht, dass charger status == "A" ist. hier ein logauszug:

~~~
[site  ] DEBUG 2023/06/21 09:53:56 ----
[lp-1  ] ERROR 2023/06/21 09:53:42 charger: invalid status: 400
{"errorCode":410,"errorCodeName":"ChargerDisconnected","type":null,"title":"Charger disconnected","status":400,"detail":"[Empty in production]","instance":null,"extensions":{}}

X-Amzn-Requestid: 658795e7-583f-41c0-9626-2e7a61403a9c
X-Amzn-Remapped-Server: Kestrel
X-Amzn-Remapped-Date: Wed, 21 Jun 2023 07:53:42 GMT
X-Amzn-Remapped-Content-Length: 176
X-Amzn-Remapped-Connection: keep-alive
X-Amz-Apigw-Id: G2_lGFX-joEFqcA=
Pragma: no-cache
Expires: -1
Date: Wed, 21 Jun 2023 07:53:42 GMT
Content-Type: application/problem+json; charset=utf-8
Cache-Control: no-cache,no-store
Access-Control-Allow-Origin: *
Content-Length: 176
HTTP/2.0 400 Bad Request

Accept-Encoding: gzip
Content-Type: application/json
Authorization: Bearer xxxxx
Content-Length: 0
User-Agent: Go-http-client/1.1
Host: api.easee.com
POST /api/chargers/Exxxxxxx/commands/resume_charging HTTP/1.1
[easee ] TRACE 2023/06/21 09:53:42 
[easee ] TRACE 2023/06/21 09:53:42 POST https://api.easee.com/api/chargers/Exxxxxxx/commands/resume_charging
[lp-1  ] DEBUG 2023/06/21 09:53:42 charger status: A
[lp-1  ] DEBUG 2023/06/21 09:53:42 charge total import: 1606.980kWh
[lp-1  ] DEBUG 2023/06/21 09:53:42 charge currents: [0 0 0]A
[site  ] DEBUG 2023/06/21 09:53:42 site power: -4449W
[site  ] DEBUG 2023/06/21 09:53:42 grid power: -4449W
[site  ] DEBUG 2023/06/21 09:53:42 battery power: 0W
[site  ] DEBUG 2023/06/21 09:53:42 battery soc: -340282346638528859811704183484516925440%
[site  ] DEBUG 2023/06/21 09:53:42 pv power: 4986W
[site  ] DEBUG 2023/06/21 09:53:42 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:53:41 charge power: 0W
[site  ] DEBUG 2023/06/21 09:53:41 ----
[site  ] DEBUG 2023/06/21 09:53:27 site power: -4450W
[site  ] DEBUG 2023/06/21 09:53:27 grid power: -4450W
[site  ] DEBUG 2023/06/21 09:53:27 battery power: 0W
[site  ] DEBUG 2023/06/21 09:53:27 battery soc: -340282346638528859811704183484516925440%
[site  ] DEBUG 2023/06/21 09:53:27 pv power: 4992W
[site  ] DEBUG 2023/06/21 09:53:27 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:53:26 charge power: 0W
[site  ] DEBUG 2023/06/21 09:53:26 ----
[site  ] DEBUG 2023/06/21 09:53:12 site power: -4422W
[site  ] DEBUG 2023/06/21 09:53:12 grid power: -4422W
[site  ] DEBUG 2023/06/21 09:53:12 battery power: 0W
[site  ] DEBUG 2023/06/21 09:53:12 battery soc: -340282346638528859811704183484516925440%
[site  ] DEBUG 2023/06/21 09:53:12 pv power: 4985W
[site  ] DEBUG 2023/06/21 09:53:12 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:53:11 charge power: 0W
[site  ] DEBUG 2023/06/21 09:53:11 ----
[lp-1  ] ERROR 2023/06/21 09:52:58 charger: invalid status: 400
{"errorCode":410,"errorCodeName":"ChargerDisconnected","type":null,"title":"Charger disconnected","status":400,"detail":"[Empty in production]","instance":null,"extensions":{}}

X-Amzn-Requestid: db58791b-56e9-41bd-8bf3-f3d5276457cc
X-Amzn-Remapped-Server: Kestrel
X-Amzn-Remapped-Date: Wed, 21 Jun 2023 07:52:58 GMT
X-Amzn-Remapped-Content-Length: 176
X-Amzn-Remapped-Connection: keep-alive
[... 194 lines cut ...]
[lp-1  ] DEBUG 2023/06/21 09:51:27 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/06/21 09:51:27 vehicle soc: 72%
[lp-1  ] DEBUG 2023/06/21 09:51:27 soc estimated: 72.03% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/06/21 09:51:27 charger status: C
[lp-1  ] DEBUG 2023/06/21 09:51:27 charge total import: 1606.980kWh
[lp-1  ] DEBUG 2023/06/21 09:51:27 detected active phases: 3p
[lp-1  ] DEBUG 2023/06/21 09:51:27 charge currents: [15.5 15.5 15.6]A
[site  ] DEBUG 2023/06/21 09:51:27 site power: 6756W
[site  ] DEBUG 2023/06/21 09:51:27 grid power: 6756W
[site  ] DEBUG 2023/06/21 09:51:27 pv power: 4956W
[site  ] DEBUG 2023/06/21 09:51:27 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:51:26 charge power: 11167W
[site  ] DEBUG 2023/06/21 09:51:26 ----

~~~

den hab ich aus meinem loki, d.h. man muss da von unten nach oben lesen. man sieht auch, dass direkt nach dem abstecken diese log zeile kommt

~~~
[lp-1  ] INFO 2023/06/21 09:52:57 car disconnected
~~~

d.h. der loadpoint merkt, dass das auto abgesteckt wurde. aber dann gehts mit `resume_chargin` weiter und das jede minute bis um ca 11.30 das auto wieder angesteckt wurde. 

btw: die 1p3p umschaltung hat kurz vorher auch nicht so richtig geklappt, deshalb war die charge-power auch 11kW, aber das sollte hier mal nicht das Problem sein.

ulrichSchreiner, 2023-06-22:
und ahc ja: die "battery soc" bitte ignorieren, mein hausakku ist grad defekt :-)

jheinitz, 2023-06-22:
> btw: die 1p3p umschaltung hat kurz vorher auch nicht so richtig geklappt, deshalb war die charge-power auch 11kW, aber das sollte hier mal nicht das Problem sein.

Ich weiß, dass ihr keine *Cross Topic Postings* mögt, aber das finde ich schon interessant. Bei mir lief die Wallbox vorgestern auch plötzlich auf 11kW im PV Modus. Ich wollte das erstmal beobachten und dann ggf. ein Issue aufmachen. Zumindest bin ich ja aber dann wohl nicht der einzige.

Gruß Jens

GrimmiMeloni, 2023-06-22:
> ch weiß, dass ihr keine _Cross Topic Postings_ mögt, aber das finde ich schon interessant.

Ja, ist interessant. Würde Dich aber bitten eine neues Posting dafür zu öffnen und mich zu taggen.

andig, 2023-06-22:
> Frage: Den Fall zumindest loggen, oder einfach komplett raus?

Warum?

> Wir fragen jetzt explizit den HTTP Status Code ab, und bei allen anderen Codes als 200 oder 202 gibt es einen Fehler.
Daraufhin ändern wir den internen Status nicht, und deshalb entsteht die Endlosschleife.

Fehler -> Rückmeldung -> Neuer Versuch

Warum sollte der Fehler nur geloggt werden? Ist er kein Fehler?

GrimmiMeloni, 2023-06-22:
> Warum sollte der Fehler nur geloggt werden? Ist er kein Fehler?

So wie ich es sehe nein. Es ist im Endeffekt ein Aufruf der nichts bringt. Der Charger ist nicht verbunden, er kann weder starten noch stoppen. Easee gibt uns dafür aber einen HTTP 400 (Bad Request), was ich ziemlich unpassend finde. Am Request an sich ist nichts falsch. Im PayLoad haben sie als status eine 410 - so fern sich das auf HTTP bezieht wäre das ein Gone, was ich auch deutlich passender finde.

Anderer Vorschlag - wir schmeißen es nicht raus, sondern wir prüfen neben 200 und 202 auch auf 400 und werten zusätzlich das JSON auf den status code 410 aus. Damit haben wir zumindest für andere Fehler (500 o.ä.) noch den Error und korrektes Logging/Verhalten, können aber zumindest mit dieser Sonderlocke sauber umgehen.

jheinitz, 2023-06-22:
Aber muss denn in jedem Interval die Wallbox abgefragt werden, wenn gar kein Auto angeschlossen ist? Wie weiter oben schon gesagt wurde, bekommt der LP das ja mit und quitiert das mit `car disconnected`. Dann sollte vielleicht noch 1-2 Mal mit der Wallbox kommuniziert werden, aber dann müsste doch erstmal Ruhe sein bis wieder ein auto angeschlossen wird, oder?

andig, 2023-06-22:
Das sieht ein bisschen aus wie https://github.com/evcc-io/evcc/pull/8126

andig, 2023-06-22:
> So wie ich es sehe nein. Es ist im Endeffekt ein Aufruf der nichts bringt. Der Charger ist nicht verbunden, er kann weder starten noch stoppen. 

Wenn das so ist sollten wir den Aufruf nicht machen sondern einfach Erfolg zurück melden.

> Anderer Vorschlag - wir schmeißen es nicht raus, sondern wir prüfen neben 200 und 202 auch auf 400 und werten zusätzlich das JSON auf den status code 410 aus. Damit haben wir zumindest für andere [...]

Das wäre dann nicht nötig.

andig, 2023-06-22:
> Aber muss denn in jedem Interval die Wallbox abgefragt werden, wenn gar kein Auto angeschlossen ist? 

Kurz: ja. Lang: das passiert nicht weil wir die Info schon lokal haben.

GrimmiMeloni, 2023-06-22:
> Kurz: ja. Lang: das passiert nicht weil wir die Info bei Easee schon lokal haben.

Cool. Den Rückkanal zum Loadpoint sowie den state in c.opMode hatte ich noch gar nicht auf dem Schirm. Wieder was gelernt.

Wir berücksichtigen in Enable() aktuell diesen Wert nicht - und der Fix wäre in State A einfach gar nicht erst senden. 
Sounds good?

andig, 2023-06-22:
Im Prinzip ja. Ist klar was passiert, wenn wir kein disable mehr machen und dann ein neues Auto angesteckt wird?

GrimmiMeloni, 2023-06-22:
Verhalten zunächst erstmal undokumentiert. Ich erwarte das wir unverändertes Verhalten haben werden. Finden wir mal heraus was passiert. PR kommt gleich.

GrimmiMeloni, 2023-06-23:
Habe den PR gerade getestet. Abgesteckt. Kein Aufruf von Pause. Auch während der Guard Duration war Ruhe. Wieder angesteckt, verhalten wie immer. Aus meiner Sicht ist dieser Fix in Ordnung.

andig, 2023-06-23:
Warum taucht der Fehler (anscheinend) erst mit 118 auf, aber nicht in älteren Versionen? Das `disable` Verhalten hat sich m.E. nicht geändert?

GrimmiMeloni, 2023-06-24:
@andig doch, wir haben das Verhalten unbewusst verändert. Die Zeile die ich [hier](https://github.com/evcc-io/evcc/issues/8560#issuecomment-1602390828) erwähnt habe ist schuld. Durch diese werfen wir einen Error. 

Wenn Du den alten Code anschaust (https://github.com/evcc-io/evcc/blob/1ba4698b8b85ebb7913d5d440fa91bd6b42ca945/charger/easee.go#L395C16-L395C16) dann siehst Du das wir dort keine Fehlerbehandlung hatten (zumindest keine auf Basis von result codes). Die einzige Möglichkeit dort einen Error zu bekommen war aus c.Post(...). Da die go Library bei non-200 results aber nicht automatisch errors produziert, wurde also in dem hier diskutierten Szenario einfach weiter gemacht. Wir werden die 400er Antwort dann immer nur einmal im Log gehabt haben und gut war. Deshalb ist es mEn bisher auch nicht weiter aufgefallen.

andig, 2023-06-24:
Sehr gut, danke. Dann mergen wir mal!

PR #8583 (2023-06-24): easee: do not send pause/resume commands when disconnected
fix #8560
closed 2023-06-24 by andig