evcc-io/evcc #8564: When dsmr is not available log is flooded, backoff not working?

vinc-vo, 2023-06-21:
### Describe the bug

When starting evcc while the dsmr meter is not connected, EVCC starts polling the dsmr meter in an infinite loop, filling up the logs and generating 100% cpu usage. 
It seems that there is something wrong with the exponential backoff. (Maybe just for me? Could I be missing a parameter? I built from source, but did not edit anything.)

The debug output shows me starting EVCC for a couple of seconds with loglevel debug and no dsmr meter connected.

It feels like the same or a similar issue as was (not?) fixed by "DSMR: prevent busy loop on EOF (https://github.com/evcc-io/evcc/pull/5641)"

I'm using ser2net to convert the usb dsmr serial input, but I believe this issue could occur with a network dsmr adapter as well. Not sure, maybe it is the way ser2net behaves when the serial input is not found. 

### Steps to reproduce

1. Set up ser2net (does not matter that the serial device does not exist, that's the point)
2. Start EVCC
3. Log fills up very fast, cpu at 100%

ser2net.conf:
`#
# This is the configuration file for ser2net.  It has the following format:
#  <TCP port>:<state>:<timeout>:<device>:<options>
#     TCP port
#            Name   or  number of the TCP/IP port to accept con-
#            nections from for this device.  A port number may
#            be of the form [host,]port, such as 127.0.0.1,2000
#            or localhost,2000.  If this is specified, it will
#            only bind to the IP address specified. Otherwise
#            it will bind to all the ports on the machine.
#
#     state  Either raw or rawlp or telnet or off.  off disables
#            the  port  from  accepting  connections.  It can be
#            turned on later from the control port.  raw enables
#            the port and  transfers  all data as-is between the
#            port  and  the long.  rawlp  enables  the port  and
#            transfers  all input data to device, device is open
#            without  any  termios  setting.  It  allow  to  use
#            /dev/lpX  devices  and  printers connected to them.
#            telnet enables the port and runs the telnet  proto-
#            col  on the port to set up telnet parameters.  This
#            is most useful for using telnet.
#
#     timeout
#            The time (in seconds) before the port will be  dis-
#            connected  if  there  is no activity on it.  A zero
#            value disables this funciton.
#
#     device The  name  of  the  device   to  connect  to.  This
#            must be in the form of /dev/<device>.
#
#     options
#            Sets  operational  parameters  for the serial port.
#            Options 300, 1200, 2400, 4800, 9600, 19200, 38400,
#            57600, 115200 set the various baud rates.  EVEN,
#            ODD, NONE set the parity.  1STOPBIT, 2STOPBITS set
#            the number of stop bits.  7DATABITS, 8DATABITS set
#            the number of data bits.  [-]XONXOFF turns on (-
#            off) XON/XOFF support.  [-]RTSCTS turns on (- off)
#            hardware flow control, [-]LOCAL turns off (- on)
#            monitoring of the modem lines, and
#            [-]HANGUP_WHEN_DONE turns on (- off) lowering the
#            modem control lines when the connextion is done. 
#	     NOBREAK disables automatic setting of the break
#	     setting of the serial port.
#            The "remctl" option allow remote control (ala RFC
#            2217) of serial-port configuration.  A banner name
#            may also be specified, that banner will be printed
#            for the line.  If no banner is given, then no
#            banner is printed.
#
# or...

#  BANNER:<banner name>:banner
#    This will create a banner, if the banner name is given in the
#    options of a line, that banner will be printed.  This takes the
#    standard "C" \x characters (\r is carraige return, \n is newline,
#    etc.).  It also accepts \d, which prints the device name, \p,
#    which prints the TCP port number, and \s which prints the serial
#    parameters (eg 9600N81).  Banners can span lines if the last
#    character on a line is '\'.  Note that you *must* use \r\n to
#    start a new line.
#
# Note that the same device can be listed multiple times under different
# ports, this allows the same serial port to have both telnet and raw
# protocols.

# The original config file shipped with the upstream sources can be
# found in /usr/share/doc/ser2net/examples

BANNER:banner:\r\nser2net port \p device \d [\s] (Debian GNU/Linux)\r\n\r\n

#2000:telnet:600:/dev/ttyS0:9600 8DATABITS NONE 1STOPBIT banner
#2001:telnet:600:/dev/ttyS1:9600 8DATABITS NONE 1STOPBIT banner
#3000:telnet:600:/dev/ttyS0:19200 8DATABITS NONE 1STOPBIT banner
#3001:telnet:600:/dev/ttyS1:19200 8DATABITS NONE 1STOPBIT banner

2001:raw:600:/dev/tty-DSMR_1:115200 NONE 1STOPBIT 8DATABITS XONXOFF LOCAL -RTSCTS
`

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa

interval: 10s # control cycle interval

meters:
- type: template
  template: dsmr 
  usage: grid  
  host: localhost  
  port: 2001  
  energy: 1-0:1.7.0
  name: grid1
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.10.99
  port: 1502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv2

chargers:
- type: template
  template: abl 
  id: 1  
  device: /dev/tty-RS485_1  
  baudrate: 38400  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox3

loadpoints:
- title: Oprit
  charger: wallbox3
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: false
 # vehicle: ev1 # set default vehicle (disables vehicle detection)

vehicles:
- type: template
  template: bmw
  title: X3 xDrive 30e
  user: xxxxxxxx@xxxxxxxxxx.xx
  password: xxxxxxxxxxxxxxxxxxxx
  vin: xxxxxxxxxxxxxxxxx
  capacity: 11.15
  phases: 1
  cache: 15m
  name: ev1

site:
  title: Goorxxxxxxxxxt 5
  meters:
    grid: grid1
    pvs:
    - pv2

# push messages
messaging:
  events:
    start: # charge start event
      title: Laden begonnen
      msg: Laden gestart in "${mode}" mode
    stop: # charge stop event
      title: Laden gedaan
      msg: Laden gedaan ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Auto aangesloten
      msg: "Auto aangesloten met ${pvPower:%.1fk}kW PV"
    #disconnect: # vehicle connected event
    #  title: Auto losgekoppeld
    #  msg: Auto losgekoppeld na ${connectedDuration}
    #soc: # vehicle soc update event
    #  title: SoC geüpdatet
    #  msg: Batterij geladen tot ${vehicleSoC:%.0f}%
    guest: # vehicle could not be identified
      title: Ongekende auto
      msg: Ongekende auto aangesloten, gasten?
```

### Log details

```text
openhabian@openHABianDevice:~ $ sudo journalctl -fau evcc
-- Logs begin at Wed 2023-06-21 11:39:24 CEST. --
Jun 21 11:52:14 openHABianDevice systemd[1]: Started evcc.
Jun 21 11:52:15 openHABianDevice evcc[9282]: [main  ] INFO 2023/06/21 11:52:15 evcc 0.118.1 (b2b4483b)
Jun 21 11:52:15 openHABianDevice evcc[9282]: [main  ] INFO 2023/06/21 11:52:15 using config file: /etc/evcc.yaml
Jun 21 11:52:15 openHABianDevice evcc[9282]: [main  ] INFO 2023/06/21 11:52:15 starting ui and api at :7070
Jun 21 11:52:15 openHABianDevice evcc[9282]: [db    ] INFO 2023/06/21 11:52:15 using sqlite database: /root/.evcc/evcc.db
Jun 21 11:52:15 openHABianDevice evcc[9282]: [dsmr  ] ERROR 2023/06/21 11:52:15 peek: EOF
Jun 21 11:52:15 openHABianDevice evcc[9282]: [dsmr  ] ERROR 2023/06/21 11:52:15 peek: EOF

[a lot of the same lines, you get the point]

Jun 21 11:52:24 openHABianDevice evcc[9282]: [dsmr  ] ERROR 2023/06/21 11:52:24 peek: EOF
Jun 21 11:52:24 openHABianDevice evcc[9282]: [dsmr  ] ERROR 2023/06/21 11:52:24 peek: EOF
Jun 21 11:52:24 openHABianDevice evcc[9282]: [dsmr  ] ERROR 2023/06/21 11:52:24 peek: EOF
Jun 21 11:52:24 openHABianDevice evcc[9282]: [dsmr  ] ERROR 2023/06/21 11:52:24 peek: EOF
Jun 21 11:52:24 openHABianDevice systemd[1]: Stopping evcc...
Jun 21 11:52:24 openHABianDevice systemd[1]: evcc.service: Main process exited, code=killed, status=15/TERM
Jun 21 11:52:24 openHABianDevice systemd[1]: evcc.service: Succeeded.
Jun 21 11:52:24 openHABianDevice systemd[1]: Stopped evcc.
```
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.1 (b2b4483b)

andig, 2023-06-21:
That shouldn't be possible since we stop if no connection can be made:

	conn, err := m.connect()
	if err != nil {
		return nil, err
	}

Why does the connect work when there's no meter?

vinc-vo, 2023-06-21:
The ser2net accepts the incoming connection, but there is nothing to be read.

andig, 2023-06-25:
It think being able to read anything from the connection should reset the backoff, hence

		if b, err := reader.Peek(1); err == nil {
			backoff.Reset()

			if string(b) != "/" {
				log.DEBUG.Printf("ignoring garbage character: %c\n", b)
				_, _ = reader.ReadByte()
				continue
			}
		} else {
			handle("peek", err)
			continue
		}

Could you give that a try? Will send PR to merge after your ok.

vinc-vo, 2023-06-27:
I'm still getting a flood of 
`[dsmr  ] ERROR 2023/06/27 12:58:44 peek: EOF`

I think there should be a backoff period after every error.

`if b, err := reader.Peek(1); err == nil {
	backoff.Reset()

	if string(b) != "/" {
		log.DEBUG.Printf("ignoring garbage character: %c\n", b)
		_, _ = reader.ReadByte()
		continue
	}
} else {
	handle("peek", err)
	continue
}
`

should be something like this

`if b, err := reader.Peek(1); err == nil {
	backoff.Reset()

	if string(b) != "/" {
		log.DEBUG.Printf("ignoring garbage character: %c\n", b)
		_, _ = reader.ReadByte()
		continue
	}
} else {
	handle("peek", err)
                     time.Sleep(backoff.NextBackOff().Truncate(time.Second))
	continue
}
`

andig, 2023-06-29:
Added your suggestion. Does that work now?

closed 2023-06-30 by andig