evcc-io/evcc #8572: Ladeleistung wechselt von 3.7kW auf 11kW im PV Modus

jheinitz, 2023-06-22:
### Describe the bug

Moin!

Vor zwei Tagen habe ich im PV Modus das Auto geladen. Der Ladevorgang verlief auch zuerst ohne Problem. Dann aber merkte ich, dass die Ladeleistung 11kW betrug und ein großer teil der energie aus der Hausbatterie kam. Im Log konnte ich keinen Fehler entdecken und wollte abwarten, ob das wieder auftritt. Heute habe ich dann in #8560 gelesen, dass ich nicht der einzige mit dem Problem bin. Leider habe ich kein Log. In meiner Grafana Historie kann ich sehen, dass die Leistung direkt von 3.7 kW auf 11 kW wechselte. Dies ist nur mit einem Wechsel von 1 auf 3 Phasen möglich. Es sieht so aus, dass dabei aber mit jeweils 16A weitergeladen wird.

### Steps to reproduce

1. connect car
2. start charging in pv mode
3. wait for phase switch
4. might not occur every time when phases are switched
...

### Configuration details

```yaml
- Skoda Enyaq
- Easee allbox
```

### Log details

```text
Log from that particular event no longer available
```

### What type of operating system are you running?

Docker container

### Version

0.118.1

jheinitz, 2023-06-22:
Moin!

cc /@GrimmiMeloni

Hier der Issue dazu.

Jens

ulrichSchreiner, 2023-06-22:
hi,

hier ein log (bitte von unten nach oben lesen)

auf 3p umgeschaltet um
~~~
[lp-1  ] DEBUG 2023/06/21 09:41:42 phase scale3p timer elapsed
[lp-1  ] DEBUG 2023/06/21 09:41:42 phase scale3p in 0s
[lp-1  ] DEBUG 2023/06/21 09:41:42 available power 4160W > 4140W min 3p threshold
~~~

durch war das ganze dann

~~~
[lp-1  ] DEBUG 2023/06/21 09:42:56 charge power: 11167W
~~~

und dann blieb er leider auf den 11kW. Kurz darauf wurde das Auto eh abgesteckt, d.h. es tat nicht weh. Am nachmittag ist die Umschaltung ein paarmal gut gegangen, aber einmal war es wieder so (er blieb bei 11kW). Was bei mir hilft: in der Easse-Android App den Ladevorgang durch drücken auf "Pause"  kurz pausieren. Dann kriegt evcc eine menge "ProductUpdate's" und dann fängt es sich wieder (evcc regelt wieder). 

Aber bis dahin gibt es die aus Benutzersicht schräge Situation, dass evcc in der Oberfläche einen Überschuß von bspw. 6kW anzeigt und gleichzeitig aber mit 11kW lädt und keine Anstaltenn macht es zu ändern.

~~~
[lp-1  ] DEBUG 2023/06/21 09:43:57 pv charge current: 0A = 6A + -10.3A (7083W @ 3p)
[lp-1  ] DEBUG 2023/06/21 09:43:57 phase scale1p in 1m32s
[lp-1  ] DEBUG 2023/06/21 09:43:57 available power 4084W < 4140W min 3p threshold
[lp-1  ] DEBUG 2023/06/21 09:43:57 vehicle range: 383km
[lp-1  ] DEBUG 2023/06/21 09:43:57 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/06/21 09:43:57 vehicle soc: 69%
[lp-1  ] DEBUG 2023/06/21 09:43:57 soc estimated: 69.00% (vehicle: 69.00%)
[lp-1  ] DEBUG 2023/06/21 09:43:57 charger status: C
[lp-1  ] DEBUG 2023/06/21 09:43:57 charge total import: 1606.980kWh
[lp-1  ] DEBUG 2023/06/21 09:43:57 detected active phases: 3p
[lp-1  ] DEBUG 2023/06/21 09:43:57 charge currents: [15.5 15.5 15.6]A
[site  ] DEBUG 2023/06/21 09:43:57 site power: 7083W
[site  ] DEBUG 2023/06/21 09:43:57 grid power: 7083W
[site  ] DEBUG 2023/06/21 09:43:57 battery power: 0W
[site  ] DEBUG 2023/06/21 09:43:57 battery soc: -340282346638528859811704183484516925440%
[site  ] DEBUG 2023/06/21 09:43:57 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:43:56 charge power: 11167W
[site  ] DEBUG 2023/06/21 09:43:56 ----
[site  ] DEBUG 2023/06/21 09:43:42 site power: 7124W
[site  ] DEBUG 2023/06/21 09:43:42 grid power: 7124W
[site  ] DEBUG 2023/06/21 09:43:42 battery power: 0W
[site  ] DEBUG 2023/06/21 09:43:42 battery soc: -340282346638528859811704183484516925440%
[site  ] DEBUG 2023/06/21 09:43:42 pv power: 4809W
[site  ] DEBUG 2023/06/21 09:43:42 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:43:41 charge power: 11167W
[site  ] DEBUG 2023/06/21 09:43:41 ----
[site  ] DEBUG 2023/06/21 09:43:27 site power: 7141W
[site  ] DEBUG 2023/06/21 09:43:27 grid power: 7141W
[site  ] DEBUG 2023/06/21 09:43:27 battery power: 0W
[site  ] DEBUG 2023/06/21 09:43:27 battery soc: -340282346638528859811704183484516925440%
[site  ] DEBUG 2023/06/21 09:43:27 pv power: 4806W
[site  ] DEBUG 2023/06/21 09:43:27 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:43:26 charge power: 11167W
[site  ] DEBUG 2023/06/21 09:43:26 ----
[lp-1  ] DEBUG 2023/06/21 09:43:12 pv charge current: 0A = 6A + -10.3A (7094W @ 3p)
[lp-1  ] DEBUG 2023/06/21 09:43:12 phase scale1p in 2m17s
[lp-1  ] DEBUG 2023/06/21 09:43:12 available power 4073W < 4140W min 3p threshold
[lp-1  ] DEBUG 2023/06/21 09:43:12 vehicle range: 383km
[lp-1  ] DEBUG 2023/06/21 09:43:12 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/06/21 09:43:12 vehicle soc: 69%
[lp-1  ] DEBUG 2023/06/21 09:43:12 soc estimated: 69.00% (vehicle: 69.00%)
[lp-1  ] DEBUG 2023/06/21 09:43:12 wake-up timer: stop
[lp-1  ] INFO 2023/06/21 09:43:12 start charging ->
[lp-1  ] DEBUG 2023/06/21 09:43:12 charger status: C
[lp-1  ] DEBUG 2023/06/21 09:43:12 charge total import: 1606.980kWh
[lp-1  ] DEBUG 2023/06/21 09:43:12 charge currents: [15.5 15.5 15.6]A
[site  ] DEBUG 2023/06/21 09:43:12 site power: 7094W
[site  ] DEBUG 2023/06/21 09:43:12 grid power: 7094W
[... 360 lines cut ...]
[lp-1  ] DEBUG 2023/06/21 09:40:57 phase scale3p in 15s
[lp-1  ] DEBUG 2023/06/21 09:40:57 available power 4161W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/06/21 09:40:57 vehicle range: 377km
[lp-1  ] DEBUG 2023/06/21 09:40:57 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/06/21 09:40:57 vehicle soc: 70%
[lp-1  ] DEBUG 2023/06/21 09:40:57 soc estimated: 69.51% (vehicle: 68.00%)
[lp-1  ] DEBUG 2023/06/21 09:40:57 charger status: C
[lp-1  ] DEBUG 2023/06/21 09:40:57 charge total import: 1606.980kWh
[lp-1  ] DEBUG 2023/06/21 09:40:57 detected active phases: 1p
[lp-1  ] DEBUG 2023/06/21 09:40:57 charge currents: [15.6 0.024 0.011]A
[site  ] DEBUG 2023/06/21 09:40:57 site power: -352W
[site  ] DEBUG 2023/06/21 09:40:57 grid power: -352W
[site  ] DEBUG 2023/06/21 09:40:57 battery power: 0W
[site  ] DEBUG 2023/06/21 09:40:57 battery soc: -340282346638528859811704183484516925440%
[site  ] DEBUG 2023/06/21 09:40:57 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:40:56 charge power: 3809W
[site  ] DEBUG 2023/06/21 09:40:56 ----
[site  ] DEBUG 2023/06/21 09:40:42 site power: -328W
[site  ] DEBUG 2023/06/21 09:40:42 grid power: -328W
[site  ] DEBUG 2023/06/21 09:40:42 pv power: 4732W
[site  ] DEBUG 2023/06/21 09:40:42 static grid tariff
[lp-1  ] DEBUG 2023/06/21 09:40:41 charge power: 3809W
[site  ] DEBUG 2023/06/21 09:40:41 ----

~~~

GrimmiMeloni, 2023-06-22:
Danke für das Log, @ulrichSchreiner
Puh, das ist jetzt echt ätzend. Man kann es ziemlich genau sehen.
Selbst mit Abwarten auf die Antwort macht Easee hier Quatsch. Sie sind trotz der CommandResponse für den Resume Charge immer noch nicht intern im Tritt. Der Charger steht noch auf 6A. Dann setzen wir explizit 6A was mit der leeren Antwort quittiert wird (weil das noch laut Easee der aktuelle Stand ist), und erst DANACH kommt das Product Update mit den 32A - was ja der bekannte Seiteneffekt von resume_charge ist.
Das explizite Setzen von 6A nach resume hatte ich extra eingebaut um diesen Fall möglichst zu verhindern, aber offensichtlich ist das zu kurz gedacht. Da muß ich erstmal drüber nachdenken was hier möglich ist.

GrimmiMeloni, 2023-06-22:
@andig please accept as bug and assign to me

ulrichSchreiner, 2023-06-22:
Seid ihr eigentlich sicher, dass die Reihenfolge der eingehenden ProductUpdate's auch immer der zeitl. entspricht? Jeder dieser update's kommt mit einem Key/Value, das abgegriffen wird, es kommt aber auch ein Timestamp mit, wann dieser konkrete Wert gesetzt wurde. Kann es sein, dass sich da asynchrone Abläufe überholen, aber der entspr. Timestamp ok wäre? 

Ich kenne sowas von Trackern die Daten senden; wenn es da mal einen Hänger gibt, puffern die das und senden dann die Daten-Updates aber nicht zeitlich wie es passiert ist sondern anders rum, so dass der Empfänger als erstes den aktuellen Status bekommt.

Ich hab mir mal eine Version gebaut, die den TS mitloggt, aber mein Auto ist grad gut voll, d.h. das wird länger dauern, bis ich mal wieder auflade.

GrimmiMeloni, 2023-06-22:
> Seid ihr eigentlich sicher, dass die Reihenfolge der eingehenden ProductUpdate's auch immer der zeitl. entspricht?

Wenn mehrere Calls gleichzeitig laufen ist die Antwort ganz klar nein. Ich habe auch schon beobachtet, daß es dann Random zurückkommt. Deshalb habe ich den Change gebaut das wir jetzt auf das Command Update warten, und somit alle Aufrufe gegen die API nur noch serialisiert erfolgen. Ferner habe ich bisher auch immer die Reihenfolge so gesehen.
- Command senden
- Product Update kommt
- Command Response kommt

Somit erschien es schlüssig auf die Command Response zu warten. Ich glaube auch das dies weiterhin korrekt ist. Das Problem ist die Sonderlocke bei `pause` und `resume`. Beide habe einen (bekannten) Seiteneffekt das sie Dynamic Charger Current verändern. So wie sich das in Deinem Log darstellt, geschieht das aber auch wieder asynchron. Also der Empfang des Command Updates für `resume` sagt nur aus, daß der Resume durch ist. Über die Veränderung von Dynamic Charger Current sagt dies aber offensichtlich nichts aus. Mir sieht es danach aus, daß Easee aber selbst die Settings parallel schreibt und somit kommt dann dieses Product Update auch nochmal anders rein.

Vermutlich wird es ausreichen im Fall von `resume` noch zusätzlich auf ein Product Update mit den 32A zu warten. Dann sind wir hoffentlich wirklich sicher das Easee mit seinen Veränderungen durch ist, und können unsere 6A senden.

ulrichSchreiner, 2023-06-22:
wenn das asyn-verhalten von easee wirklich verkehrt ist (also auch die timestamps bei den product-updates nix bringen), kannst ja überlegen, alle paar minuten einen

~~~
client.Send("SubscribeWithCurrentState", c.charger, true);
~~~
abzusetzen. soviel ich weis, bläst der dann den kompletten charger-status über ProductUpdate raus. so eine art "hardcore"-Reconcile. Ist bei kubernetes auch so, man lauscht auf events des controllers/operators und regelmässig rekonziliert man trotzdem den ganzen zustand um sicherzugehen, dass es nicht doch mal nen netzhänger oder sowas gab :smiley:

andig, 2023-06-22:
…um dann geblacklisted zu werden und zwischendurch nie zu wissen ob der Zustand nicht eh schon inkonsistent ist.

ulrichSchreiner, 2023-06-22:
da die signalr verbindung eine stehende websocket-verbindung ist, mach ich jede wette, dass da drüber nix geblacklistet wird. 

aber es ist müssig drüber zu spekulieren. s'einfachste wäre ein lokale api, aber haben sie halt nicht. d

andig, 2023-06-22:
Stimmt 🙄

GrimmiMeloni, 2023-06-22:
Hab mal eine Mögliche Lösung in #8588 zusammen gebaut. Ist ungetestet, aber vielleicht wollt Ihr einen ersten Blick drauf werfen.

GrimmiMeloni, 2023-06-24:
Ich habe #8588 heute durch mehrfaches Stoppen und Starten des Ladevorgangs getestet. Es sieht hier erstmal alles OK aus, daher habe ich den Change zum Review freigegeben. 
Das Problem beim Testen ist, daß diese Race Condition nur ab und zu aufgetreten ist. Es ist nicht möglich 100%ig zu sagen, daß es die Lösung ist. Zumindest macht es meiner Ansicht nach aber nichts kaputt.

Ich habe build https://github.com/GrimmiMeloni/evcc/commit/07b7f86d63f253064144d6f0eb8b154d699d5830 ins docker repo gepushed. ([grimmimeloni/evcc:nightly](https://hub.docker.com/r/grimmimeloni/evcc/tags)) 

@ulrichSchreiner & @jheinitz habt Ihr Lust den auch mal zu testen? Der build enthält den aktuellen Master (inkl. #8560 & #8605 ) sowie den Change hier.

ulrichSchreiner, 2023-06-24:
ich werde es bei mir einspielen, allerdings wird das noch länger dauern bis ich das testen kann. das auto ist noch 80% voll und aktuell ist eigentlich nur eine grössere radltour geplant, d.h. viel auto gefahren wird in den kommenden tagen nicht.

jheinitz, 2023-06-24:
Moin!

Danke für Deine ganze Mühe @GrimmiMeloni . Ich kann das heute leider nicht testen. Aber morgen Nachmittag kann ich das mal machen. Gebe dann Feedback. 

Danke und Gruß Jens

andig, 2023-06-24:
@GrimmiMeloni könntest Du auch hier bitte nochmal erklären, warum das Problem neuerdings auftaucht?

GrimmiMeloni, 2023-06-25:
Gerne. 
Das Problem ist nicht neu. Es ist ein Duplicate von meinem ursprünglichen Issue https://github.com/evcc-io/evcc/discussions/7366. Im Kern ist dies weiterhin die Race Condition zwischen dem automatischen maxA bei Resume und der Regelung durch evcc. 
Der Grund für den zusätzlichen Fix hier ist folgender. Ich hatte in #8307 ursprünglich erwartet das es ausreicht auf die Antwort vom Resume zu warten. Das Beispiel von @jheinitz hat gezeigt, daß der von Easee intern ausgelöste Wechsel zu maxA aber nochmal entkoppelt läuft. Resume kommt zurück, aber der maxA wird erst später aktiv. 
Die Problematik als API user ist, daß der Call für maxA komplett intern abläuft. Wir haben weder eine Tick ID, noch sehen wir eine Command Response. Wir können daher nur auf das Product Update warten. Deshalb habe ich jetzt beim Resume noch den zusätzlichen Wait auf Setzen von maxA eingebaut.

jheinitz, 2023-06-25:
Moin!

So, ich bin wieder zu Hause und konnte testen. Ich habe bei mir eingestellt, dass automatisch im PV-Modus geladen wird, sobald das Auto angesteckt wird. (mode: pv am loadpoint). Nun habe ich das gemacht und der Ladevorgang hat auch begonnen, allerdings mit 11kW und hat ca. 3kW vom Netz gezogen. Das hat sich auch nicht eingependelt. Ist das noch ein Sonderfall, der von dem Fix nicht abgedeckt wird? 

Ich habe dann über das UI das Laden auf "Aus" gestellt und nach kurzer Zeit erneut auf "PV". Nun läuft das und es wird je nach Sonne geregelt. 

Während der Situation mit dem Netzbezug kann ich im Log sehen, dass dort 11kW chargePower ausgewiesen werden, aber nur 2 Phasen auf knapp 16A und die dritte auf ca. 0A angezeigt werden. In der Easee App sind aber alle 3 Phasen bei knapp 16A. Das passt nicht so ganz zusammen. siehe Log.

[evcc_test.log](https://github.com/evcc-io/evcc/files/11859806/evcc_test.log)

Gruß Jens

jheinitz, 2023-06-25:
Und jetzt sagt die Easee App, dass mit 1,4kW geladen wird, aber evcc sagt **Ladebereit, warte auf Fahrzeug**.

Ich muss sagen, dass ich bis einschließlich 0.117.4 keine solchen Probleme hatte. Hmm 

Und nun?

GrimmiMeloni, 2023-06-25:
@jheinitz ich habe gestern auch durch Zufall so eine Art "einpendeln" beobachtet. Ich konnte es auf das zu kurze Interval von 10s zurückführen. Hintergrund hier ist, daß die im Regelkreis benutzten "ist"-Werte ebenfalls asynchron reinkommen.

Das bei Dir eine Phase läuft aber komplett nicht erkannt wird, könnte dadurch erklärt werden, daß da ggf. eine Nachricht ausgeblieben ist. Bei Dir läuft der Easee Teil nicht auf trace, daher kann man das aus dem Log allerdings nicht erkennen.

jheinitz, 2023-06-25:
Okay, verstehe. Mein Intervall ist allerdings schon 20s.

GrimmiMeloni, 2023-06-25:
Zur Sicherheit: Du benutzt jetzt nicht meinen oben verlinkten build, und hast auch nicht die WIP changes aus #8588 mit drin? Dann sollten wir hier erstmal Pause machen.

jheinitz, 2023-06-25:
Nein, ich habe heute aus dem master mein docker image gebaut. Mit dem Commit 9cd614a0. Und das zeigt evcc --version auch an.

GrimmiMeloni, 2023-06-25:
ok, dann fehlt Dir zumindest der hier diskutierte Fix noch. Der ist meiner Ansicht nach essentiell um das Problem wirklich in den Griff zu bekommen. Lass uns warten bis der im Master/Nightly ist (oder teste meinen build wenn Du möchtest). Dann schauen wir nochmal.

jheinitz, 2023-06-25:
Aber Dein nightly hat doch den gleichen Commit Hash. Verstehe das jetzt nicht was da noch fehlt bzw. wo der Unterschied ist.

GrimmiMeloni, 2023-06-25:
Hmmm. Mein Nightly ist Hash [fe1c6941](https://github.com/evcc-io/evcc/commit/fe1c6941).
Sicher das Du wirklich meinen und nicht den offiziellen evcc nightly genommen hast?

jheinitz, 2023-06-25:
Ja, bin mir sicher. Habe das Image, also Deins, gerade gezogen. Hast Du das letzte nicht gepusht oder einen Fehler beim Push gehabt?

ulrichSchreiner, 2023-06-25:
hm .... das docker-image meldet aber 

`[main  ] INFO 2023/06/25 17:47:02 evcc 9cd614a0 (9cd614a0)`

~~~
❯ docker run -it --rm grimmimeloni/evcc:nightly
Unable to find image 'grimmimeloni/evcc:nightly' locally
nightly: Pulling from grimmimeloni/evcc
31e352740f53: Pull complete
42ff8ef31f99: Pull complete
1a919f7d54db: Pull complete
4105becda998: Pull complete
59f07eaba7e3: Pull complete
38139eeee33f: Pull complete
Digest: sha256:5be2e1dbc03b9ea1fed26be7400ee700267432c22a69472a062549d3a0192dd1
Status: Downloaded newer image for grimmimeloni/evcc:nightly
[main  ] INFO 2023/06/25 17:47:02 evcc 9cd614a0 (9cd614a0)
~~~

jheinitz, 2023-06-25:
Ja genau. Den habe ich auch!

GrimmiMeloni, 2023-06-25:
Danke für den Hinweis, Jens.
Du hattest in der Tat recht. Ich habe versehentlich master und nicht meinen nightly branch gebaut. 🤦 
(Da die anderen zwei Changes bereits in Master sind, ist es nicht sofort aufgefallen.)

Ich habe meine GitHub Actions korrigiert, und nochmal einen Build mit den letzten Changes bereit gestellt.
docker pull liefert für mein Image jetzt https://github.com/GrimmiMeloni/evcc/commit/cc7b9850bbb5f2d4527fea19921b0db6c6bb77bf .
Sorry für die Verwirrung.

jheinitz, 2023-06-26:
Moin!
Dann lag ich ja doch nicht so verkehrt. Nach einem `docker-compose pull` bekomme ich jetzt auch den *richtigen* Hash:
```
jens@server2:/opt/evcc$ docker run --rm -it grimmimeloni/evcc:nightly evcc --version
evcc version cc7b9850 (cc7b9850)
jens@server2:/opt/evcc$
```

Sonne scheint auch, jetzt fehlt nur noch das Auto. Das kommt aber erst gegen 17:00 Uhr zurück ..... 

Werde dann berichten. Bis später!

MaxV85, 2023-06-26:
Was ich an der Stelle mal grundsätzlich wissen wollen würde - und das soll bitte nicht als irgendeine Kritik oder so gesehen werden -, wie macht das beispielsweise clever PV? Hat da jemand nen Vergleich? Dort kann man ja auch die Easee Box verwenden...

jheinitz, 2023-06-26:
> Sonne scheint auch, jetzt fehlt nur noch das Auto. Das kommt aber erst gegen 17:00 Uhr zurück .....

Auto ist wieder da, dafür keine Sonne mehr. Muss euch auf morgen vertrösten.... Sorry

jheinitz, 2023-06-27:
Moin!

so, Auto ist angesteckt und hat mit 3.8kW einphasig gestartet mit dem Laden. Dann nach Ablauf des Timers auf 3phasig gewechselt und mit 11kW geladen und aus dem Netz gezogen. Dann eine Reihe Fehlermeldungen:
```
charger enable: timeout
```
und
```
charger out of sync: excpected disabled, got enabled
```
Und das immer im Wechsel. :-(

Das ist so für mich nicht zu gebrauchen. Sorry. Wenn's nach mir ginge, würde ich mir die Logik aus **0.117.4** zurück wünschen. 

Das Logfile kann ich leider erst heute abend hier anhängen.

Viele Grüße

Jens

GrimmiMeloni, 2023-06-27:
Blöd. :( 
Danke trotzdem fürs Testen, Jens.
Ich schaue mir Dein Log heute Abend dann mal an.

GrimmiMeloni, 2023-06-27:
Den Fehler bzgl. enable konnte ich heute nicht nachstellen.
Dafür habe ich aber auch wieder maxA gesehen wo keines sein sollte.
So langsam wird es mit der Easee API echt bunt.

Ich habe hier ein Log vorliegen, in dem Easee von Ihrer Seite ein Dynamic Charger Update auf 32A pushed, wir daraufhin im Anschluß einen API Call absetzen um auf 6A zu korrigieren, und dieser wird dann mit einer leeren Antwort quittiert - womit Easee ausdrückt "sind doch schon 6A". Sie widersprechen sich selbst.

Wie wir mit einer dermaßen inkonsistenten API umgehen sollen ist mir gerade etwas unklar.

GrimmiMeloni, 2023-06-27:
@andig was hältst Du davon ein `Loadpoint.setLimit()` einen check einzubauen das `lp.chargeCurrent` mit `lp.ChargeCurrents` übereinstimmt, und bei Abweichung nochmal `Charger.maxCurrent()` aufzurufen? Es ist die einzige Stelle im System die ich sehe, die zumindest die Chance hat zu erkennen, wenn der Charger das setzen des Limits ignoriert hat.

jheinitz, 2023-06-27:
Moin! Aber wieso funktioniert das denn so viel besser in 0.117.4? Oder kommt mir das nur so vor?

PS: Aufgrund einer Weichenstörung komme ich später nach Hause....(nerv)...

GrimmiMeloni, 2023-06-27:
> Moin! Aber wieso funktioniert das denn so viel besser in 0.117.4? Oder kommt mir das nur so vor?

Das Problem an sich habe ich bereits in 0.115 beobachtet und gemeldet. Die Changes in 0.118.1 machen das Fehlverhalten nur sichtbar(er).

jheinitz, 2023-06-27:
> Ich schaue mir Dein Log heute Abend dann mal an.

Ich habe leider schlechte Nachrichten. Ich *Dussel* habe ein `docker-compose up -d` gemacht **bevor** ich das Log gesichert habe. Tut mir leid. Mal sehen, kann dann leider erst Freitag einen neuen Test machen.

Sorry und viele Grüße

Jens

GrimmiMeloni, 2023-06-28:
Nach dem ich gestern zunächst mal das Handtuch geworfen hatte, habe ich heute morgen nochmal einen Ladevorgang beobachtet, und zumindest einen Fehler in der Logik gefunden.
Den habe ich gefixt, und ein neues Image gebaut das ich im Laufe des Vormittags hier fleißig testen konnte. Während ich im alten Build das Problem relativ schnell reproduzieren konnte, habe ich heute schon ca. 15 pause/resume cycle, bzw. Phasenwechsel durch. Alle waren bisher in Ordnung. Ich bin vorsichtig optimistisch das dies die Lösung ist.

Es wäre toll wenn Ihr auch diesen Build nochmal testen könntet. 
Container image wie gehabt: `grimmimeloni/evcc:nightly` - build / commit ID `254fd022`

jheinitz, 2023-06-28:
Moin!
Ich konnte doch einen schnellen Test mit Deinem `nightly` machen. Leider wieder das gleiche, aber diesmal mit Log.
Ich hoffe, dass du da etwas mit anfangen kannst.

Viele Grüße
Jens
[evcc_28-Jun-2023.log](https://github.com/evcc-io/evcc/files/11894629/evcc_28-Jun-2023.log)

GrimmiMeloni, 2023-06-28:
Danke Jens.
Ich sehe schonmal ein bisschen mehr. Kurze Frage: Das Log sieht ähnlich aus wie meins - ich nehme an, Du betreibst die Easee "offen" ohne Authentifizierung?

Was mir auffällt, ist das bei Dir der disable Aufruf immer wieder fehlschlägt. Das erklärt zumindest in Deinem Fall, warum nicht von den 11kW runtergeregelt wird. evcc kommt gar nicht so weit, und bricht jedes Intervall beim Charger Sync bereits ab. Das bedeutet auch, daß Dein Szenario ein anderes Problem ist.

Gibt es die Möglichkeit mir ein trace log inkl. `--log-headers` von der Situation zur Verfügung zu stellen? Ich müßte mal sehen was da auf der API gesprochen wird um das weiter einzukreisen. Für mich ist so erstmal nicht ersichtlich an welchem Aufruf es scheitert (pause_charge vs. /settings).

GrimmiMeloni, 2023-06-28:
Ja, trace von Easee reicht. Bitte aber auch auf der Commandline --log-headers einschalten. Da Du compose verwendest, kannst Du in Deinem Compose File dafür folgendes zum evcc container hinzufügen:

```
    command: ["evcc", "--log-headers"]
``` 

Bitte schau das Du es sauber sanitzied wenn Du es hier postest, denn das Log enthält deine Tokens. Alternativ kannst du es mir auch direkt senden an `m_hess<AT>gmx.net`. Weiterhin Danke fürs Mithelfen.

ulrichSchreiner, 2023-06-29:
hi,

mal als zwischenstand von mir, weil ich dann gleich weg muss. ich hab mir heut morgen den nightly-branch zu mir reingemerged (20398ab08ce0f31284e78661c5d3949332744b77) und dann angesteckt. bisher läuft es; ein paarmal wurde 1p-3p bzw. auch wieder von 3p-1p gemacht, alles ohne probleme. da jetzt genug sonnen-power da ist geh ich davon aus, dass es keinen wechsel mehr geben wird.

das einzige was mir auffällt: bei einem 1p-3p wechsel geht die wallbox kurz auf 11kW, wird dann aber gleich runtergeregelt. früher war das nicht (immer) so. wenn das runteregeln stabil funktioniert stört mich das nicht, dass es da für ein paar sekunden mal 11kW sind. wenn es aber genau zu dem zeitpunkt dann irgendwo hakt, dann ist es halt doof. 

aber wie gesagt: seit anderthalb stunden läuft die version problemlos

andig, 2023-06-29:
Mach gerne mal ein Ticket bei Easee auf, dass sie sich dieses "Maximalleistungsverhalten" sparen sollen. Das macht die Steuerung schwieriger und hat schlimmstenfalls Nebeneffekte...

jheinitz, 2023-06-29:
Moin! 
Ich habe auch die Version [20398ab](https://github.com/evcc-io/evcc/commit/20398ab08ce0f31284e78661c5d3949332744b77) gezogen. Jetzt warte ich nur auf das Auto.....
Ich habe gesehen, dass im Branch [issue8572](https://github.com/evcc-io/evcc/tree/issue8572) noch ein Commit ([6e9a0d2](https://github.com/evcc-io/evcc/commit/6e9a0d281adcbc315a30dfaaa0819debafbaac71)) für einen *Observation Channel* enthalten ist. @andig, was macht das und macht es Sinn, dass @GrimmiMeloni den auch noch in sein Nightly aufnimmt?

ulrichSchreiner, 2023-06-29:
@andig ich bin mir nicht sicher, ob easee an der stelle der richtige adressat ist. denn die umschaltung von 1-3p hat bei mir jedenfalls kein pause/resume gemacht, sondern 

erstmal den

~~~
{"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":16,"dynamicCircuitCurrentP3":16}
~~~

und da die box ja bisher 1p bei voller leistung war, sind das jetzt ja drei phasen volle leistung also 11kW. und dann nach ca. 30sec wurde das abgesetzt

~~~
{"dynamicChargerCurrent":6}
~~~

und jetzt war ich auf 4.3kW. 

ein pause/resume ist da nirgends im log.

so gesehen war das von easee nicht so verkehrt auf 11kW zu schalten

andig, 2023-06-29:
> ich bin mir nicht sicher

Aber ich. Die Easee schaltet immer auf Maximum :(

ulrichSchreiner, 2023-06-29:
> Aber ich. Die Easee schaltet immer auf Maximum :(

das mag ohne zweifel so sein :-)

aber an der stelle ist es halt nicht verkehrt, weil via API das ja auch so hingesendet wurde, volle kanne, 3 phasen. ich bin mir relativ sicher, das früher vor dem umschalten immer erst ein "pause" passierte. dann die umschaltung und dann ein "resume".

andig, 2023-06-29:
Ich habe verstanden: weil es an dieser Stelle ausnahmsweise passt kein Ticket bei Easee. Fein, lass uns die Diskussion beenden.

GrimmiMeloni, 2023-06-29:
> macht es Sinn, dass @GrimmiMeloni den auch noch in sein Nightly aufnimmt?

Ja @jheinitz, das macht Sinn. Wenn, dann wollen wir auch den möglichst letzten Stand nehmen. Wir haben auch nochmal eine Menge debug und trace Meldungen mit eingebaut, damit wir besser verstehen was da genau passiert. Es läuft gerade ein build mit id [5e14315](https://github.com/GrimmiMeloni/evcc/commit/5e14315515c7c3297d2ed569391ae2de9f03e599), der ist ein paar Minuten fertig.

> mal als zwischenstand von mir, weil ich dann gleich weg muss. ich hab mir heut morgen den nightly-branch zu mir reingemerged ([20398ab](https://github.com/evcc-io/evcc/commit/20398ab08ce0f31284e78661c5d3949332744b77)) und dann angesteckt. bisher läuft es; ein paarmal wurde 1p-3p bzw. auch wieder von 3p-1p gemacht, alles ohne probleme. da jetzt genug sonnen-power da ist geh ich davon aus, dass es keinen wechsel mehr geben wird.
> 
> das einzige was mir auffällt: bei einem 1p-3p wechsel geht die wallbox kurz auf 11kW, wird dann aber gleich runtergeregelt. früher war das nicht (immer) so. wenn das runteregeln stabil funktioniert stört mich das nicht, dass es da für ein paar sekunden mal 11kW sind. wenn es aber genau zu dem zeitpunkt dann irgendwo hakt, dann ist es halt doof.
> 
> aber wie gesagt: seit anderthalb stunden läuft die version problemlos

Cool. Bitte ruhig weiter beobachten. Ich hab das ganze gestern ausgiebig "gequält" und es sah weitestgehend gut aus. Ich hatte am späten Nachittag/Abend nochmal einen Fall wo es nicht passte. Da bin ich aber unsicher woran es genau lag. Könnte ein anderes Fehlerbild gewesen sein.

jheinitz, 2023-06-29:
Okay, Deinen Build habe ich gezogen und aktiviert. Was jetzt hier im Norden fehlt, ist die Sonne ...

andig, 2023-06-29:
Auch das ist am ehesten ein Bug bei Easee und sollte gefixed werden. Dafür müsste es aber jemand melden…

GrimmiMeloni, 2023-06-29:
Ja, allerdings ist die SignalR Schnittstelle wohl intern. Ich hatte dazu mal beim Support nach einer Referenz gefragt. Die Antwort dazu war, daß dieses Interface undokumentiert - weil intern - sei.

Ich habe gerade nochmal gezielt diverse (ältere) Logs durchgeschaut, und es gibt definitiv zumindest einen zeitlichen Zusammenhang zwischen negotiate und pause/resume. Ob das jetzt aber wirklich die Effekte erklärt weiß ich nicht, aber es ist zumindest ein Hinweis.

andig, 2023-06-29:
Es ist dokumentiert oder war es zumindest. Von intern war da keine Rede. Die biegen es sich hin wie sie lustig sind…

ulrichSchreiner, 2023-06-29:
Nur damit das nicht falsch rüberkommt: Aus meiner Sicht hat evcc heut genau so funktioniert wie es sollte. Die 1p3p Umschaltung mit dem kurzen Peak ist imho vollkommen OK.  Wer hier mehr will, soll selbst erstmal ein async API ansteuern, welches ja selbst nochmal async die Wallbox ansteuert. Also meine Anmerkung diesbgzl. jetzt nicht als Kritik sehen; es waren einfach nur Beobachtungen!

Und das Negotiation von der SignalR Verbindung hat mich jetzt nicht wirklich gestört. Im Gegenteil, wenn evcc damit die aktuellen Einstellungen von easee kriegt ... so what? War halt auch nur eine Beobachtung.

andig, 2023-06-29:
> War halt auch nur eine Beobachtung.

...und dürfte dennoch ein Bug sein wenn die immer bei der Phasenumschaltung weg ist. 

> Im Gegenteil, wenn evcc damit die aktuellen Einstellungen von easee kriegt

Das brauchts nicht, weil den haben wir ja dank der Verbindung...

jheinitz, 2023-06-30:
Moin!

So, ich habe jetzt einen Trace. Das Verhalten war wieder das gleiche. Es ging auf 11kW und hat sich auch nicht eingeregelt wie bei @ulrichSchreiner . Ich habe das Tracefile bereinigt und gezippt. Der Trace beginnt gestern nachmittag, aber das Auto habe ich heute um 10:05 Uhr angeschlossen und dann fängt der Ladevorgang automatisch an. 

Was mich wundert ist, dass es ein Timeout für `charger enable` gibt, obwohl der Ladevorgang aktiv ist. Ich habe die Wallbox so eingestellt, dass der Ladevorgang ohne RFID Karte automatisch im PV Modus beginnt.

Habe dann um 10:11 Uhr abgebrochen und bin auf 0.117.4 zurück. Da klappt das dann mit dem Laden ohne Fehler und timeouts. 

@GrimmiMeloni Vielleicht siehst Du ja etwas im Trace, was uns weiterbringt. 

[evcc_trace_30-Jun-2023_anonym.log.gz](https://github.com/evcc-io/evcc/files/11915196/evcc_trace_30-Jun-2023_anonym.log.gz)

Viele Grüße

Jens

GrimmiMeloni, 2023-06-30:
Danke für Deine Mühe Jens.
Das Problem ist jetzt klar. Wir warten beim Resume auf 16A, Easee setzt aber 32A.
Frage zu Deiner Elektroinstallation. Wie ist die Easee bei Dir abgesichert?

jheinitz, 2023-06-30:
Moin!
Im Sicherungskasten ist eine ABB S203-C32 Sicherung verbaut. Zusätzlich noch ein ABB F204 A FI Schalter. siehe Bild. Die 32A brauche ich eigentlich nicht, denn mein Skoda Enyaq kann maximal 16A. Aber wo ist denn der Unterschied zu 0.117.4?

Danke Dir auch für Deine Geduld @GrimmiMeloni .

Ciao Jens
[image]

[17 more replies]

PR #8588 (2023-06-30): Easee: wait for async current update on resume
fix #8572
closed 2023-06-30 by andig