evcc-io/evcc #8592: Umschaltung auf % Anzeige trotz unbekanntem SoC

bit-driver, 2023-06-23:
### Describe the bug

Ich habe neben der Wallbox eine Schaltbare Steckdose (Tasmota) eingerichtet.
Funktion ist gegeben und auch die Datenübertragung funktioniert.

Wenn der Ladevorgang nicht aktiv ist, kann ich wie gewünscht Ladeziele (z.B. +50Wh) einstellen.

Sobald von der Steckdose die erste geladene Energiemenge übertragen wird springt die Anzeige um auf einstellbaren SoC und die Schwelle wird auf 100% gesetzt.

[image]

### Steps to reproduce

1. Ladevorgang Starten
2. Erste geladene Energie der Session wird übertragen
3. Anzeige springt um
...

### Configuration details

```yaml
meters:
  - name: tasmota_charger
    type: template
    template: tasmota
    usage: charge
    host: 192.168.2.10

chargers:
  - name: steckdose
    type: template
    template: tasmota
    host: 192.168.2.10 # IP-Adresse oder Hostname
    user: # Standard-User ist admin (Optional)
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (Optional)
    channel: 1 # Nummer des Schaltkanals (1-8), bei Geräten mit mehr als einem Schalter
    standbypower: 1 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (Optional) 

vehicles:
  - name: pedelec
    type: template
    template: offline
    icon: bike
    title: Pedelec # display name for UI
    capacity: 0.5 # byttery capacity (kWh)

loadpoints:
  - title: Steckdose
    charger: steckdose
    meter: tasmota_charger
    vehicle: pedelec
    mode: off
    guardduration: 1m
    phases: 1
    minCurrent: 1
```

### Log details

```text
[main  ] INFO 2023/06/23 12:55:56 evcc 0.118.1
[main  ] INFO 2023/06/23 12:55:56 using config file: /etc/evcc.yaml
[main  ] INFO 2023/06/23 12:55:56 starting ui and api at :7070
[site  ] WARN 2023/06/23 12:56:01 deprecated: use 'pv' instead of 'pvs'
[site  ] WARN 2023/06/23 12:56:01 vehicle 'Pedelec' does not support automatic detection
```

### What type of operating system are you running?

Docker container

### Version

v0.118.1

Headroom42, 2023-06-26:
Und blöderweise kann man auch nicht auf eine 0.117.4 zurück gehen, da kommt bei einer nicht veränderten evcc.yaml nun folgende Fehlermeldung. Auch getestet mit allen anderen 0.117.x
Warum verstehen die vorherigen Versionen vor der 0.118.0 auf einmal nicht mehr die evcc.yaml richtig?

[image]

naltatis, 2023-06-27:
Ursache dafür ist die Änderung hier: https://github.com/evcc-io/evcc/pull/8101

@andig In diesem Fall wird `loadpoint.x.vehicleSoc` hochgezählt. Vermutlich durch den Estimator. Das müssten wir für Fahrzeuge ohne Soc unterbinden.

PR #8663 (2023-06-27): Use estimator only for vehicles with soc
fixes #8592
closed 2023-06-27 by naltatis