evcc-io/evcc #8598: kia / bluelink Showin car as offline

cremmering, 2023-06-23:
### Describe the bug

Kia is configured according to documentation. Login in portal and also access via iobroker works.
In EVCC it shows up as offline and the following error messages appears:
vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: Post \"https://prd.eu-ccapi.kia.com:8080/api/v1/spa/notifications/register\": dial tcp: lookup prd.eu-ccapi.kia.com: i/o timeout

### Steps to reproduce

1. Configure car in yaml
2. restart evcc
3.
...

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: kia
    title: # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: W... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
```

### Log details

```text
vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: Post \"https://prd.eu-ccapi.kia.com:8080/api/v1/spa/notifications/register\": dial tcp: lookup prd.eu-ccapi.kia.com: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

0.118.1

andig, 2023-06-23:
> i/o timeout

Da kannste nix machen- Server antwortet einfach nicht...

closed 2023-06-23 by andig

LustigerBosniak, 2023-07-14:
Bei Euch nach wie vor so? Bei mir besteht das gleiche Problem immer noch.